Seeed-Studio/seeed-linux-dtoverlays #6: Question regarding the can hat and the jetson nano

JakeSays, 2020-09-23:
I have a couple of questions regarding the `jtsn-mcp25xx` branch.

1) is it to the point that I can do some tests with it?
2) does the mcp driver have a dependency on the spi-tegra114p driver?

I was able to build and install the mcp25xx driver but I don't get canx network cards.

turmary, 2020-09-24:
Hi JakeSays,
The spi-tegra114p driver only required on Jetson Nano platform cause spi-tegra114 is kernel builtin driver.

Please follow below instructions to install the driver, then refer [wiki](https://wiki.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi/) for rest opertion.
```bash
	# 1. Build dtbo & driver
	cd seeed-linux-dtoverlays
	git checkout origin/jtsn-mcp25xxfd -b jtsn-mcp25xxfd
	export CUSTOM_MOD_LIST="CAN-HAT"; make all_jetsonnano
	# 2. Install driver
	sudo -E make install_jetsonnano
	# 3. Install dtbo
	sudo cp overlays/jetsonnano/2xMCP2518FD-spi0.dtbo /boot
	sudo /opt/nvidia/jetson-io/config-by-hardware.py -n "Seeed 2xMCP2518FD"
```

JakeSays, 2020-09-24:
@turmary 
Hey thank you for your reply! I followed your directions and got a lot farther - I can see the driver attempting to initialize in dmesg.

Here is the dmesg output:

> [    9.530545] CAN device driver interface
[    9.638414] mcp25xxfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD m:20.00MHz r:18.50MHz e:0.00MHz) successfully initialized.
[    9.674629] mcp25xxfd spi0.1 (unnamed net_device) (uninitialized): Failed to detect MCP2518FD (osc=0x00000000).

It looks like it found can0 fine, but failed on can1. The can hats I have are https://www.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi-p-4072.html. They are brand new and have dual MCP2518FD's. Is this the right product?

turmary, 2020-09-24:
I am sure your shiled is dual MCP2517FD version which use SPI0 & SPI1, so only SPI0/CAN0 work when using this driver.  
New version SHIELD with dual MCP2518FD only use SPI0 with double chip selects, CAN0 and CAN1 are both work.

JakeSays, 2020-09-24:
I'm definitely using the MCP2518FD.

Here is  a pic. It's hard to see but you can read the part #'s.

[image]

turmary, 2020-09-24:
Hi JakeSays,
I know your shield using dual MCP2518FD now,
Cause MCP2517FD are EOL and New version shield only using single SPI bus( with double chip-selects) are not in sell.

Result is you can use only CAN0 on Jetson Nano and CAN0 & CAN1 on RPi.

JakeSays, 2020-09-24:
@turmary ah I understand now. So it is working as expected. Excellent!

I only need CAN0 so I'm good. Out of curiosity are you planning on adding support for CAN1 on the nano, or is it even possible?

And thank you for your help. I have been trying to get this controller to work on a nano for quite a while now!

turmary, 2020-09-24:
For the shield on your hand, CAN1 is can't working on Nano without wire firing.

yunlistener, 2020-09-27:
@turmary hi, could you please let me know where i can buy a new one with 2518chip on it?

JakeSays, 2020-09-27:
@yunlistener I get mine from amazon

closed 2020-10-27 by Pillar1989

baroselfdriving, 2021-03-06:
In the nano production module don't work the config-by-hardware.py command. Then is not possible to install it

JakeSays, 2021-03-06:
@baroselfdriving This issue is closed. you should open a new issue.