Seeed-Studio/seeed-linux-dtoverlays #8: NPi i.MX6ULL Dev Board - Make Errors

cdutko, 2020-10-16:
After cloning the git repo and running the make command in the correct directory, I am met with a lot of errors, that ultimately do not allow me to complete the make execute.

Output can be seen here:

```
root@npi:~/seeed-linux-dtverlays# make
  DTC     overlays/jetsonnano/2xMCP2518FD-spi0.dtbo
  DTC     overlays/jetsonnano/jetson-seeed-2mic-voicecard.dtbo
  DTC     overlays/jetsonnano/jetson-seeed-2mic-wm8960.dtbo
  Building modules, stage 2.
  MODPOST 3 modules
  Building modules, stage 2.
  MODPOST 2 modules
  Building modules, stage 2.
  MODPOST 1 modules
  Building modules, stage 2.
  MODPOST 1 modules
  Building modules, stage 2.
  MODPOST 1 modules
  Building modules, stage 2.
  MODPOST 1 modules
  Building modules, stage 2.
  MODPOST 1 modules
  Building modules, stage 2.
  MODPOST 1 modules
  Building modules, stage 2.
  MODPOST 1 modules
  Building modules, stage 2.
  MODPOST 1 modules
  CC [M]  /root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.o
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:60:34: warning: ‘struct snd_soc_codec’ declared inside parameter list will not be visible outside of this definition or declaration
 static int wm8960_set_pll(struct snd_soc_codec *codec,
                                  ^~~~~~~~~~~~~
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:134:31: warning: ‘struct snd_soc_codec’ declared inside parameter list will not be visible outside of this definition or declaration
  int (*set_bias_level)(struct snd_soc_codec *,
                               ^~~~~~~~~~~~~
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:179:37: warning: ‘struct snd_soc_codec’ declared inside parameter list will not be visible outside of this definition or declaration
 static int wm8960_set_deemph(struct snd_soc_codec *codec)
                                     ^~~~~~~~~~~~~
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c: In function ‘wm8960_set_deemph’:
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:181:31: error: implicit declaration of function ‘snd_soc_codec_get_drvdata’; did you mean ‘snd_soc_dai_get_drvdata’? [-Werror=implicit-function-declaration]
  struct wm8960_priv *wm8960 = snd_soc_codec_get_drvdata(codec);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~
                               snd_soc_dai_get_drvdata
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:181:31: warning: initialization of ‘struct wm8960_priv *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
In file included from ./include/linux/i2c.h:30,
                 from /root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:19:
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:200:15: error: dereferencing pointer to incomplete type ‘struct snd_soc_codec’
  dev_dbg(codec->dev, "Set deemphasis %d\n", val);
               ^~
./include/linux/device.h:1461:26: note: in definition of macro ‘dev_dbg’
   dev_printk(KERN_DEBUG, dev, dev_fmt(fmt), ##__VA_ARGS__); \
                          ^~~
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:202:9: error: implicit declaration of function ‘snd_soc_update_bits’; did you mean ‘snd_ac97_update_bits’? [-Werror=implicit-function-declaration]
  return snd_soc_update_bits(codec, WM8960_DACCTL1,
         ^~~~~~~~~~~~~~~~~~~
         snd_ac97_update_bits
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c: In function ‘wm8960_get_deemph’:
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:209:32: error: implicit declaration of function ‘snd_soc_kcontrol_codec’; did you mean ‘snd_soc_kcontrol_component’? [-Werror=implicit-function-declaration]
  struct snd_soc_codec *codec = snd_soc_kcontrol_codec(kcontrol);
                                ^~~~~~~~~~~~~~~~~~~~~~
                                snd_soc_kcontrol_component
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:209:32: warning: initialization of ‘struct snd_soc_codec *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:210:31: warning: initialization of ‘struct wm8960_priv *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  struct wm8960_priv *wm8960 = snd_soc_codec_get_drvdata(codec);
                               ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c: In function ‘wm8960_put_deemph’:
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:219:32: warning: initialization of ‘struct snd_soc_codec *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  struct snd_soc_codec *codec = snd_soc_kcontrol_codec(kcontrol);
                                ^~~~~~~~~~~~~~~~~~~~~~
[... 271 lines cut ...]
                      ^~~~
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:1309:22: note: (near initialization for ‘soc_codec_dev_wm8960’)
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c: In function ‘wm8960_i2c_probe’:
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:1393:8: error: implicit declaration of function ‘snd_soc_register_codec’; did you mean ‘snd_soc_register_dai’? [-Werror=implicit-function-declaration]
  ret = snd_soc_register_codec(&i2c->dev,
        ^~~~~~~~~~~~~~~~~~~~~~
        snd_soc_register_dai
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c: In function ‘wm8960_i2c_remove’:
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:1401:2: error: implicit declaration of function ‘snd_soc_unregister_codec’; did you mean ‘snd_soc_unregister_card’? [-Werror=implicit-function-declaration]
  snd_soc_unregister_codec(&client->dev);
  ^~~~~~~~~~~~~~~~~~~~~~~~
  snd_soc_unregister_card
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c: At top level:
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:1306:42: error: storage size of ‘soc_codec_dev_wm8960’ isn’t known
 static const struct snd_soc_codec_driver soc_codec_dev_wm8960 = {
                                          ^~~~~~~~~~~~~~~~~~~~
/root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.c:60:12: warning: ‘wm8960_set_pll’ used but never defined
 static int wm8960_set_pll(struct snd_soc_codec *codec,
            ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:304: /root/seeed-linux-dtverlays/modules/jtsn-wm8960/wm8960.o] Error 1
make[1]: *** [Makefile:1519: _module_/root/seeed-linux-dtverlays/modules/jtsn-wm8960] Error 2
make: *** [Makefile:151: all_jetsonnano] Error 2
root@npi:~/seeed-linux-dtverlays#
```

vibnwis, 2020-10-17:
Errors also, but different ones
  register_sysctl
/home/pi/seeed-linux-dtoverlays/modules/sht3x/sht3x.c: At top level:
/home/pi/seeed-linux-dtoverlays/modules/sht3x/sht3x.c:184:31: error: storage size of ‘file_oprts’ isn’t known
 static struct file_operations file_oprts = {
                               ^~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:266: /home/pi/seeed-linux-dtoverlays/modules/sht3x/sht3x.o] Error 1
make[1]: *** [Makefile:1709: /home/pi/seeed-linux-dtoverlays/modules/sht3x] Error 2
make: *** [Makefile:151: all_jetsonnano] Error 2

turmary, 2020-10-19:
hi all,
you may need specify a environment variable CUSTOM_MOD_LIST, for example:
```
export CUSTOM_MOD_LIST="CAN-HAT"
```
which contain names of driver module under foler modules/ you required.
Cause not all drivers supported on any one platform.

cdutko, 2020-10-19:
@turmary I will give that a try and post my results here.

closed 2020-10-19 by cdutko