Seeed-Studio/seeed-linux-dtoverlays #11: CANFD and Jetson Nano B01 support

mrspirytus, 2020-11-20:
**Is your feature request related to a problem? Please describe.**
I am thinking about purchasing few B01 [CANFD](https://www.seeedstudio.com/CAN-BUS-FD-HAT-for-Raspberry-Pi-p-4742.html#). Wiki page only describes rPI board but not Jetson Nano, which is mentioned at the above link. 

**Describe the solution you'd like**
Update install.sh script to work with Jetson Nano 

**Describe alternatives you've considered**
I did not try anything since I want to make sure that this works. I purchased something similar for a different vendor and it turns out it is not working. 

**Additional context**
I am looking at the above link
```
git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays
cd seeed-linux-dtoverlays//modules/CAN-HAT
sudo ./install.sh 
sudo reboot
``` 
Install script only check for raspberry pi. But also looking at https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/master/overlays/jetsonnano I see there is dst file for can0. 
Can install.sh be updated to support Nano? There are a lot of guys on the NVIDIA support page struggling to get various can modules to work. I just order one CANFD to try it before I order several more.
I can also try to update the install.sh script and do pull request. Just want to know what is known to work.

Pillar1989, 2020-11-21:
https://wiki.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi/#using-can-bus-shiled-with-jetson-nano @mrspirytus  If you still have problems using the Wiki, you can open the issues again.

closed 2020-11-21 by Pillar1989

JakeSays, 2020-11-21:
@Pillar1989 Hey that page talks about a version of the hat that supports both can buses on the nano, but I don't see where it says which version that is. There's  "version declare" link but that doesn't go anywhere.

mrspirytus, 2020-11-21:
@Pillar1989 I order on unit for testing before a order several more. While I am waiting for new board to ship (MCP2518FD)
I realized I have older version https://www.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi-p-4072.html .Does this board work with Nano? or I should wait for MCP2518FD?

JakeSays, 2020-11-21:
@mrspirytus if it's the MCP2517FD then yes it will work but you'll only have access to one of the two CAN buses

mrspirytus, 2020-11-21:
@JakeSays So I just looked little closer at my board and it is MCP2518FD but it does not have the RTC battery holder. I already did added RTC to my nano. So I should be fine and should have can0 and can1. 
I will try this tomorrow. One final question. Does this board uses two SPI on J41 nano header? I really need one CAN interface right now. If current configuration is us using two SPI, How can I disable second SPI for now. Also are the schematics available somewhere? 
Thank you

JakeSays, 2020-11-21:
@mrspirytus I am not sure exactly which version of the board supports two as the wiki is vague on that subject. Same with the SPI question - it can either use one spi bus with two peripherals, or two separate buses. You could look at the device tree file to know for sure, or perhaps @Pillar1989 could update the wiki with details.

JakeSays, 2020-11-21:
@mrspirytus I just tried the wiki instructions on my nano. No luck - only one CAN bus available.

mrspirytus, 2020-11-21:
@JakeSays which board version are you using? 17D or 18D?
I will test my board tomorrow that I 18D with out RTC battery. I will report my finding. Strange since I got this board from my rPI about two weeks ago. And the units I have does not match website. They only list 18D and 17D. My look like 17D but clearly 18D chips are used on it.

JakeSays, 2020-11-21:
@mrspirytus the 18FD w/o the RTC battery (which I assume means there is no RTC with it).

mrspirytus, 2020-11-22:
@JakeSays  I just did try it and I got can0 interface only. That is good enough for me.
@Pillar1989 should be get two can interfaces on Jetson? If only one is supported is there a reason why only one? And the newer board can support two?

Pillar1989, 2020-11-23:
[image]
Seeed currently has three versions, the first two of which are no longer available. Because it uses two  SPI controllers. But in the Jatson Nano there is only one  SPI controller, so there is only one  can0.
The last version, because the CAN bus for both channels uses the same SPI controller but uses different slice signals. So in Jatson Nano, there can be a 2-channel Can bus.

JakeSays, 2020-11-23:
@Pillar1989 Do you know if there's a hack or modification we could make to our 2518fd boards (#2 above) to make both buses work off of one spi channel?

Pillar1989, 2020-11-23:
@JakeSays  I recommend using our latest version of the hardware, We couldn't provide a way to break the hardware, and that didn't work as well as we expected.

mrspirytus, 2020-11-23:
@Pillar1989 as far a ikon no has two SPi on J41 connector (40 pin). So I am little confused by your statement about single SPI on Nano. Thank you