Seeed-Studio/seeed-linux-dtoverlays #12: When I use this on Jetson Nano, get an error ?

dennis08, 2020-12-15:
Hi,

On Jetson nano, we get a problem and after a few minutes Nano rebooted. 
dmesg log: (boot up)
mcp25xxfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD m:20.00MHz
 r:18.50MHz e:0.00MHz) successfully initialized.

kernel log: (sudo ip link set can0 up type can bitrate 1000000   dbitrate 8000000 restart-ms 1000 berr-reporting on fd on)
**mcp25xxfd spi0.0 can0: Controller failed to enter mode Mixed (CAN FD/CAN 2.0) Mode (0) and stays in Config
uration Mode (4).**

After a few minutes, Jetson nano rebooted, can you help to check about this bug on jetson nano?  Thanks so much.

Pillar1989, 2020-12-15:
Jetson nano reboot? We don't seem to have come across it. It seems to be an exception.

dennis08, 2020-12-15:
Yes, Jetson nano reboot, then the dtbo overlays works fine.

```
Steps:
#0
git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays
#1. Build dtbo & driver
cd seeed-linux-dtoverlays
git checkout origin/jtsn-mcp25xxfd -b jtsn-mcp25xxfd
export CUSTOM_MOD_LIST="CAN-HAT"; make all_jetsonnano
#2. Install driver
sudo -E make install_jetsonnano
#3. Install dtbo
sudo cp overlays/jetsonnano/2xMCP2518FD-spi0.dtbo /boot
sudo /opt/nvidia/jetson-io/config-by-hardware.py -n "Seeed 2xMCP2518FD"
```

```
nvidia@nano:~$ ifconfig -a
can0: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 112
```

```
dmesg log: (boot up)
mcp25xxfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD m:20.00MHz
r:18.50MHz e:0.00MHz) successfully initialized.
```
```
$sudo ip link set can0 up type can bitrate 1000000 dbitrate 8000000 restart-ms 1000 berr-reporting on fd on
We got a dmesg log and after a few minutes, Jetson nano rebooted as:

mcp25xxfd spi0.0 can0: Controller failed to enter mode Mixed (CAN FD/CAN 2.0) Mode (0) and stays in Configuration Mode (4).
```

closed 2020-12-19 by dennis08