Seeed-Studio/seeed-linux-dtoverlays #13: Raspberry Pi - driver and overlay available for seeed-can-fd-hat

menschel, 2020-12-31:
Hi,

there is now support for  seeed-can-fd-hat in regular raspberry pi kernel.

https://github.com/raspberrypi/linux/pull/4034

Please update your readme.md .

Thanks.

Pillar1989, 2021-01-05:
Thank you very much for the PR, we will revise the document as soon as possible, thanks.

closed 2021-01-05 by Pillar1989

marckleinebudde, 2021-01-05:
Hey @Pillar1989,

just a quick update, we've renamed the overlays, it's now `-v1` for the old mcp2517fd hat and `-v2` for the mcp2518fd.

```
	seeed-can-fd-hat-v1
￼	seeed-can-fd-hat-v2
```

Pillar1989, 2021-01-05:
@menschel  Thank you for the reminder, it is very much appreciated.

menschel, 2021-01-05:
Thank you but actually, most of the credit goes to @marckleinebudde .

marckleinebudde, 2021-01-05:
We all stand on the shoulders of giants. @menschel had the idea to create a single overlay for the board.

JakeSays, 2021-01-07:
> Hey @Pillar1989,
> 
> just a quick update, we've renamed the overlays, it's now `-v1` for the old mcp2517fd hat and `-v2` for the mcp2518fd.
> 
> ```
> 	seeed-can-fd-hat-v1
> ￼	seeed-can-fd-hat-v2
> ```

You missed one - the mcp2518fd with the rtc.

marckleinebudde, 2021-01-07:
Hey @JakeSays,

the `-v2` does include the rtc:

https://github.com/raspberrypi/linux/blob/abdf918004f2a4c203891e94328b8cf76fbae45f/arch/arm/boot/dts/overlays/seeed-can-fd-hat-v2-overlay.dts#L100-L116

```
	fragment@8 {
		target = <&i2cbus>;
		__overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			pcf85063@51 {
				compatible = "nxp,pcf85063";
				reg = <0x51>;
			};
		};
	};
	fragment@9 {
		target = <&i2c_arm>;
		i2cbus: __overlay__ {
			status = "okay";
		};
	};
```

menschel, 2021-01-07:
This list indicates there may have been a second batch of the first design populated with mcp2518fd instead.

https://wiki.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi/#versions-declare

And the first design is missing the RTC.

marckleinebudde, 2021-01-07:
Hmmm, ok. If they have the old board populated with a mcp2518fd instead (no RTC) and are using spi1-0 as on the mcp2517fd, then just use the `-v1` overlay. All overlays use `mcp251xfd` as compatible, which auto-detects the exact variant.

Let's wait for Seeed Studio to clarify.

JakeSays, 2021-01-07:
@marckleinebudde yeah there is unfortunately. I have two.

marckleinebudde, 2021-01-07:
Ok, the we have to clarify the readme.

marckleinebudde, 2021-01-07:
What about:

```
Name:   seeed-can-fd-hat-v1
Info:   Overlay for Seeed Studio CAN BUS FD HAT with two CAN FD channels
        (based on the mcp2517fd or mcp2518fd), without RFC.
        https://www.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi-p-4072.html
Load:   dtoverlay=seeed-can-fd-hat-v1
Params: <None>

Name:   seeed-can-fd-hat-v2
Info:   Overlay for Seeed Studio CAN BUS FD HAT with two CAN FD channels
        (based on the mcp2518fd) and an RTC.
        https://www.seeedstudio.com/CAN-BUS-FD-HAT-for-Raspberry-Pi-p-4742.html
Load:   dtoverlay=seeed-can-fd-hat-v2
Params: <None>
```

menschel, 2021-01-08:
I believe the battery holder is the "indicator" to distinguish v1 and v2.

The first design has spi0-0 and spi1-0 and no battery holder / RTC.

The second design has the battery holder / RTC and spi0-0 , spi0-1 combination.

So i would drop the (based on .*) entry in favor of a clause to look for the battery holder.

I also believe your
```
2-Channel CAN FD Master
HAT for Raspberry Pi
www.seeedstudio.com
v1.0 2/28/2020
```

is hardware identical to my 
```
CAN-BUS(FD) HAT for Raspberry Pi
www.seeedstudio.com
v1.0 6/20/2020
```

marckleinebudde, 2021-01-08:
That all makes sense :)

What about:
```
Name:   seeed-can-fd-hat-v1
Info:   Overlay for Seeed Studio CAN BUS FD HAT with two CAN FD channels without RTC. Use this overlay if your HAT has no battery holder.
        https://www.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi-p-4072.html
Load:   dtoverlay=seeed-can-fd-hat-v1
Params: <None>

Name:   seeed-can-fd-hat-v2
Info:   Overlay for Seeed Studio CAN BUS FD HAT with two CAN FD channels and an RTC. Use this overlay if your HAT has a battery holder.
        https://www.seeedstudio.com/CAN-BUS-FD-HAT-for-Raspberry-Pi-p-4742.html
Load:   dtoverlay=seeed-can-fd-hat-v2
Params: <None>
```

menschel, 2021-01-08:
Perfect, don't forget to limit the lines to 80 chars.
Thanks.

marckleinebudde, 2021-01-08:
Sure. I was to lazy to copy form here to editor and back here. :)

marckleinebudde, 2021-01-08:
rewrapped to 80 chars:

```
Name:   seeed-can-fd-hat-v1
Info:   Overlay for Seeed Studio CAN BUS FD HAT with two CAN FD
        channels without RTC. Use this overlay if your HAT has no
        battery holder.
        https://www.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi-p-4072.html
Load:   dtoverlay=seeed-can-fd-hat-v1
Params: <None>

Name:   seeed-can-fd-hat-v2
Info:   Overlay for Seeed Studio CAN BUS FD HAT with two CAN FD
        channels and an RTC. Use this overlay if your HAT has a
        battery holder.
        https://www.seeedstudio.com/CAN-BUS-FD-HAT-for-Raspberry-Pi-p-4742.html
Load:   dtoverlay=seeed-can-fd-hat-v2
Params: <None>
```

dav2dg, 2021-03-09:
I am trying to get the Seeed CAN HAT (with RTC) working.
Is there a convenient way to get seeed-can-fd-hat-v2.dtbo without doing an rpi-update? This seems like going a bit too far over the bleeding edge.

I used the Image tool and have the following rpi version:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux

but this does not have the dtbo in /boot/overlays which is where I understand they should be.

Assuming I can get a dtbo is all I need to do is update /boot/config.txt
and add...
dtoverlay=seeed-can-fd-hat-v2

Thanks
David

dav2dg, 2021-03-09:
I did the rpi-update to
Linux raspberrypi 5.10.20-v7+ #1404 SMP Thu Mar 4 19:40:23 GMT 2021 armv7l GNU/Linux

sudo reboot

then followed the tests on Seed. It all worked.

Thanks