Seeed-Studio/seeed-linux-dtoverlays #16: MCP251xFD CAN HAT driver support and documentation outdated

hartkopp, 2021-01-30:
The current script in https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/master/modules/CAN-HAT which refers to the 5.4 kernel has different issues:

1. The script states that the mcp25xxfd driver does not exist and therefore compiles and installs the mcp25xxfd driver
In fact in the LATEST 5.4 kernel from RasPi already contains the mainline mcp251xfd (not mcp25xxfd!) driver in the drivers/net/can/spi/mcp251xfd directory which is not detected by your script. But using this existing driver with the provided overlay does not work.

2. I used https://github.com/Hexxeh/rpi-update to update to the latest 5.10.11-v7l+ kernel with the latest mcp251xfd driver and the new combined overlays `seeed-can-fd-hat-v1` and `seeed-can-fd-hat-v2` from @marckleinebudde and @menschel which works like charm. Maybe these combined overlays also work for the latest 5.4 kernel out of the box - I didn't check that.

Please update your documentation / scripts to take care of the recent RasPi 5.4 kernel. The request for updating the documentation was already addressed by @menschel here: https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/13

Thanks!

marckleinebudde, 2021-01-30:
The combined overlay is also available on the rasp-5.4 kernels and should work out of the box as on v5.10.

hartkopp, 2021-01-30:
Oh, thanks! That would have been much easier than making the rpi-update to 5.10.11. This makes this Seeed-Studio documentation even more obsolete ;-)

marckleinebudde, 2021-01-30:
Yes, I don't know if the v5.4 in the latest raspberry pi os image has the driver and the overlays applied...But latest v5.4 git has. BTW: what are you doing with the mcp251xfd on a raspi?

menschel, 2021-01-31:
The latest Image from January 11th 2021 is past the date that we initiated the PR. It is pulled in by apt at least.

BTW: I'm using it as a HIL replacement as part of a DevOps pipeline with PyTest framework.

hartkopp, 2021-01-31:
And I just set up my first RasPi with the mcp251xfd for two reasons. The first is to support 42 Wolfsburg school to set up CAN bus teaching boards. And the other is for a potential CAN FD upgrade of an existing test & teaching platform for diagnosis testers.

thompcd, 2022-10-04:
Am I correct that this issue is now closed?

marckleinebudde, 2022-10-04:
ACK

menschel, 2022-10-04:
More or less. There are some obsolete steps in doc, as for cloning the repo which is no longer needed at all.

The Python Examples are no deprecated since socketcan python module is available on [pypi](https://pypi.org/project/socketcan/),

but anyways better late then never.

closed 2022-10-09 by bigbearishappy