Seeed-Studio/seeed-linux-dtoverlays #18: Can't detect mcp25xxfd spi0.1 can1

Callet91, 2021-03-03:
Hi there, 

I just installed the driver on the Jetson nano following the documentation [here](https://wiki.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi/#using-can-bus-shiled-with-jetson-nano). I have the latest "2-Channel CAN-BUS(FD) Shield for Raspberry Pi (MCP2518FD)" which should be support both can0 and can1 but I can't seem to detect can1. Any ideas of what to do? When I run: 
```
dmesg | grep spi && ifconfig -a 
```
I get the following output: 

```
[    0.470728] iommu: Adding device 7000d400.spi to group 7
[    0.471012] iommu: Adding device 7000d600.spi to group 8
[    0.471307] iommu: Adding device 70410000.spi to group 9
[    3.765799] tegra-qspi 70410000.spi: Prod settings list not found
[    3.773178] qspi_mtd spi32766.0: MX25U3235F (4096 Kbytes)
[    3.773186] qspi_mtd spi32766.0: mtd .name = spi32766.0, .size = 0x400000 (4MiB) .erasesize = 0x00001000 (4KiB) .numeraseregions = 0
[    8.096609] spi_tegra114p: loading out-of-tree module taints kernel.
[    9.500099] mcp25xxfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD m:20.00MHz r:18.50MHz e:0.00MHz) successfully initialized.
[    9.559049] mcp25xxfd spi0.1 (unnamed net_device) (uninitialized): Failed to detect MCP2518FD (osc=0x00000000).
can0: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 112  

dummy0: flags=130<BROADCAST,NOARP>  mtu 1500
        ether b6:6c:91:1e:3c:20  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.112  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::4b4e:be42:1a82:64c0  prefixlen 64  scopeid 0x20<link>
        ether 00:04:4b:ec:96:f8  txqueuelen 1000  (Ethernet)
        RX packets 6474  bytes 5717617 (5.7 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 3806  bytes 727322 (727.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 150  base 0xe000  

l4tbr0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether aa:71:21:60:f8:59  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 650  bytes 54536 (54.5 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 650  bytes 54536 (54.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

rndis0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether aa:71:21:60:f8:59  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether aa:71:21:60:f8:5b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

JakeSays, 2021-03-03:
@Callet91 does your shield have a real time clock on it?

Callet91, 2021-03-03:
@JakeSays no, it does not have the RTC-PCF85063 on board. The board I have is "2-channel CAN-BUS(FD) Shield for Raspberry  Pi <MCP2518FD> v.2.0 12/17/2019. However, the layout looks exactly like [this one](https://www.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi-p-4072.html).

JakeSays, 2021-03-03:
@Callet91 that version doesn't support both can buses. You need the one with the RTC.

Callet91, 2021-03-03:
Thank you @JakeSays and also thanks for the quick response. I close the issue then :)

closed 2021-03-03 by Callet91