Seeed-Studio/seeed-linux-dtoverlays #20: MCP2518FD Installation with the NANO Production module didn't work.

barovehicles, 2021-03-06:
When you use the installation procedure with the development board is all ok. But when you try to use with the production module the command: config-by-hardware.py don't work.
 
I'm using the carrier board B02 with the NANO Production module.

The problem is the Jetson-io library don't work with the production module.

How is possible to solve it?

JakeSays, 2021-03-06:
@barovehicles I am only aware of the A02 and B01 carrier boards.

JakeSays, 2021-03-11:
@barovehicles I just received a production module and now I see what you mean. It looks like it is in fact an issue with the jetson python scripts and not the driver.

barovehicles, 2021-03-31:
Yes. The problem is the python script in the Jetson. that doesn't work with the production module.
The problem is understand how to do the procedure manually.

is-qian, 2024-10-18:
Sorry for replying to you so late, is there any new progress on this issue?

closed 2024-11-04 by Lesords