Seeed-Studio/seeed-linux-dtoverlays #26: Trying to install reTerminal drivers on Ubuntu Mate 20.04, failed to build mipi_dsi

borongyuan, 2021-08-07:
DKMS make.log for mipi_dsi-0.1 for kernel 5.4.0-1041-raspi (aarch64)
2021年 08月 08日 星期日 02:01:23 CST
make -C /lib/modules/5.4.0-1041-raspi/build M=/var/lib/dkms/mipi_dsi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.0-1041-raspi'
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/touch_panel.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: In function ‘panel_get_modes’:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:242:9: error: too many arguments to function ‘funcs->get_modes’
  242 |   ret = funcs->get_modes(panel, connector);
      |         ^~~~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: At top level:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:255:15: error: initialization of ‘int (*)(struct drm_panel *)’ from incompatible pointer type ‘int (*)(struct drm_panel *, struct drm_connector *)’ [-Werror=incompatible-pointer-types]
  255 |  .get_modes = panel_get_modes,
      |               ^~~~~~~~~~~~~~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:255:15: note: (near initialization for ‘panel_funcs.get_modes’)
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: In function ‘i2c_md_probe’:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:348:2: error: too many arguments to function ‘drm_panel_init’
  348 |  drm_panel_init(&md->panel, dev, &panel_funcs, DRM_MODE_CONNECTOR_DSI);
      |  ^~~~~~~~~~~~~~
In file included from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi.h:35,
                 from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:11:
./include/drm/drm_panel.h:150:6: note: declared here
  150 | void drm_panel_init(struct drm_panel *panel);
      |      ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:271: /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:284:15: error: initialization of ‘int (*)(struct drm_panel *)’ from incompatible pointer type ‘int (*)(struct drm_panel *, struct drm_connector *)’ [-Werror=incompatible-pointer-types]
  284 |  .get_modes = ili9881d_get_modes,
      |               ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:284:15: note: (near initialization for ‘ili9881d_funcs.get_modes’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:269: /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o] Error 1
make[1]: *** [Makefile:1765: /var/lib/dkms/mipi_dsi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-1041-raspi'
make: *** [Makefile:15: all] Error 2

Pillar1989, 2021-08-09:
@borongyuan  Please use the latest Raspberry Pi Image. If you still have a problem you can join our Discord Channel, we can support you in real-time.
```
https://discord.gg/wSHHXADJ 
```

closed 2021-08-09 by Pillar1989

borongyuan, 2021-08-09:
查了下内核代码，看来至少要更新到5.10，Ubuntu 20.04里目前还是5.4。我们想要部署的东西并不支持raspbian。

Pillar1989, 2021-08-09:
https://ubuntu.com/download/raspberry-pi   如果要用Ubuntu，应该在这里下载

borongyuan, 2021-08-09:
我需要部署ROS Foxy，目前只能用20.04 LTS，否则需要编译安装。上面链接中只有21.04有Desktop，或者用20.04的Server版手动装桌面环境。因为我希望在树莓派上使用一个较为轻量的桌面，所以选择Ubuntu Mate，其色彩风格和reTerminal也比较搭。其实Ubuntu Mate 20.04官方是有[树莓派版](https://ubuntu-mate.org/download/arm64/)的，但没有被列入树莓派的第三方系统清单里。根据他们的[Changelog](https://ubuntu-mate.org/raspberry-pi/changelog/)，CM4是支持的。但我刷完第一次启动时，Ubiquity Wizard会崩溃，与之前[别人反馈的情况](https://ubuntu-mate.community/t/20-04-installer-crash-on-raspberry-pi-compute-module-4/24042)一致。检查了一下system-boot分区，发现并没有CM4的dtb。Ubuntu 20.04的[ReleaseNotes](https://wiki.ubuntu.com/FocalFossa/ReleaseNotes)说，CM4是在20.04.2中被支持的，而Ubuntu Mate的镜像似乎还停留在20.04.1。因此我换了一种方式来刷机，先刷20.04.2的Server版，然后把system-boot分区内容备份下来。再刷Ubuntu Mate 20.04，刷完将system-boot分区替换为前面备份的内容。之后就可以正常使用了。然后系统所有组件更新到最新后，内核还是5.4，因此只要能将内核更新到5.10，驱动应该就能装了。