Seeed-Studio/seeed-linux-dtoverlays #29: Build fail on Kali 2021.3-rpi4-nexmon-armhf

BlackDuck888, 2021-09-19:
**Describe the bug**
I tried to build the overlays on kali. I think the reTerminal would be a great application for kali. 

Building the modules is failing. 

```
DKMS make.log for mipi_dsi-0.1 for kernel 5.4.83-Re4son-v7l+ (armv7l)
Sun 19 Sep 2021 09:25:01 AM UTC
make -C /lib/modules/5.4.83-Re4son-v7l+/build M=/var/lib/dkms/mipi_dsi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-5.4.83-Re4son-v7l+'
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/touch_panel.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: In function ‘panel_get_modes’:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:250:9: error: too many arguments to function ‘funcs->get_modes’
  250 |   ret = funcs->get_modes(panel, connector);
      |         ^~~~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: At top level:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:263:15: error: initialization of ‘int (*)(struct drm_panel *)’ from incompatible pointer type ‘int (*)(struct drm_panel *, struct drm_connector *)’ [-Werror=incompatible-pointer-types]
  263 |  .get_modes = panel_get_modes,
      |               ^~~~~~~~~~~~~~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:263:15: note: (near initialization for ‘panel_funcs.get_modes’)
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: In function ‘i2c_md_probe’:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:356:2: error: too many arguments to function ‘drm_panel_init’
  356 |  drm_panel_init(&md->panel, dev, &panel_funcs, DRM_MODE_CONNECTOR_DSI);
      |  ^~~~~~~~~~~~~~
In file included from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi.h:35,
                 from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:11:
./include/drm/drm_panel.h:150:6: note: declared here
  150 | void drm_panel_init(struct drm_panel *panel);
      |      ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:294:15: error: initialization of ‘int (*)(struct drm_panel *)’ from incompatible pointer type ‘int (*)(struct drm_panel *, struct drm_connector *)’ [-Werror=incompatible-pointer-types]
  294 |  .get_modes = ili9881d_get_modes,
      |               ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:294:15: note: (near initialization for ‘ili9881d_funcs.get_modes’)
make[2]: *** [scripts/Makefile.build:266: /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:265: /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o] Error 1
make[1]: *** [Makefile:1732: /var/lib/dkms/mipi_dsi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.83-Re4son-v7l+'
make: *** [Makefile:15: all] Error 2
```

**To Reproduce**
Steps to reproduce the behavior:
1. Clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays.git
2. change `raspberrypi-kernel-headers raspberrypi-kernel` to `kalipi-kernel-headers kalipi-kernel` in /seeed-linux-dtoverlays/scripts/reTerminal.sh
3. run ./scripts/reTerminal.sh
4. See error

**Expected behavior**
I expect that the code is compiling like on raspbian OS

**Additional context**
For me is unclear why the same code on raspbian is working and on kali failing with compile errors.

What can I do to fix this? Is there a way to compile it elsewhere and copy it to Kali? If that is possible how to do this.

bigbearishappy, 2021-09-19:
Hello，it seems like a issue with kernel version.what is your kernel version?
it need to be 5.10 or higher.

BlackDuck888, 2021-09-20:
Hi, Kali is deploying 5.4.83-Re4son-v7l+ with nextmon firmware for the RPIs. Nextmon is only supporting up to 5.4 kernel.

I will try an other Kernel without nextmon, but for me unclear why are the compiler warnings `too many arguments to function` depending on the kernel and not on compiler or make enviroment.

Pillar1989, 2021-09-24:
@BlackDuck888  if you want to play Kali on reTerminal, you must make Linux kernel 5.10+. if you still have more questions about this, I recommend you read this thread.
https://github.com/raspberrypi/linux/issues/2521

closed 2021-09-24 by Pillar1989

BlackDuck888, 2021-09-24:
Thank you. I didn't get running Kernel 5.10+ with a Raspberry Image. All Kali images are build with nexmon kernel which is bound to 5.4

I will investigate into this to find a working solution for me. Thanks for the help.