Seeed-Studio/seeed-linux-dtoverlays #31: reTerminal CM4 bullseye 64 bit issue Screen fix vertical

winitboonkliang, 2021-11-11:
reTerminal CM4 model

install new raspberry pi os bullseye 64 bit 
lcd screen vertical only 
can't rotate to horizontol

please update screen driver for bullseye please

bigbearishappy, 2021-11-12:
Have you tried to use the xrandr to rotate the LCD display.
just like this the following:
DISPLAY=:0 xrandr --output DSI-1 --rotate left
DISPLAY=:0 xrandr --output DSI-1 --rotate right
DISPLAY=:0 xrandr --output DSI-1 --rotate normal
DISPLAY=:0 xrandr --output DSI-1 --rotate inverted

winitboonkliang, 2021-11-12:
> Have you tried to use the xrandr to rotate the LCD display? just like this the following: DISPLAY=:0 xrandr --output DSI-1 --rotate left DISPLAY=:0 xrandr --output DSI-1 --rotate right DISPLAY=:0 xrandr --output DSI-1 --rotate normal DISPLAY=:0 xrandr --output DSI-1 --rotate inverted

Thank your
lcd screen can rotate 

touchscreen pointer not correct 
how to calibrate touchscreen pointer?

bigbearishappy, 2021-11-12:
you can have a try with this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/f9721406b1f251f518b6e93b5c07b9e0acc6cc98) 's comment.

winitboonkliang, 2021-11-12:
Button F1 F2 F3 and other no response when press

bigbearishappy, 2021-11-12:
> > Have you tried to use the xrandr to rotate the LCD display? just like this the following: DISPLAY=:0 xrandr --output DSI-1 --rotate left DISPLAY=:0 xrandr --output DSI-1 --rotate right DISPLAY=:0 xrandr --output DSI-1 --rotate normal DISPLAY=:0 xrandr --output DSI-1 --rotate inverted
> 
> Thank you lcd screen can rotate
> 
> how to fix screen rotate on boot? add rotate config.txt
> 
> sudo nano /boot/config.txt DISPLAY=:0 xrandr --output DSI-1 --rotate right

if you want to fix screen rotate on boot. you can run the script to

bigbearishappy, 2021-11-12:
you can run the script reTerminal.sh to fix the screen rotate on boot.
you can check the gpio_keys function again after the script finished and rebooted.

bigbearishappy, 2021-11-15:
Closing the issue.You can reopen it if you have any problem.

matsujirushi, 2022-02-21:
FYI
[How to fix rotation of display for reTerminal + bullseye. (Japanese)](https://lab.seeed.co.jp/entry/2022/01/20/120000)

lakshanthad, 2022-02-24:
Hello @matsujirushi,

1. This fix can be used first:
https://lab.seeed.co.jp/entry/2022/01/20/120000

2. After that, you need to delete the line "dtoverlay=reTerminal-bridge" inside "/boot/config.txt" for LEDs and Buzzer to work 
3. Finally, you need to replace "Seeed_Python_ReTerminalQt5Examples/imports/armv7l/QtQuick" with the file below:
https://files.seeedstudio.com/wiki/ReTerminal/Qt/QtQuick.tar.gz

We will find better solutions for the above fixes and let you know.

matsujirushi, 2022-02-28:
@lakshanthad ,
Thank you for your support.

Raspberry Pi OS (32-bit) bullseye on reTerminal is working. (LCD, Touch Panel, Buttons, LEDs, RTC, Light Sensor, Accelerometer, Buzzer)

I have not yet been able to run Seeed_Python_ReTerminalQt5Examples.
Could you tell me how to install Qt5 for bullseye?

**Setup:**

```
cd ~
git clone --depth 1 https://github.com/Seeed-Studio/Seeed_Python_ReTerminalQt5Examples

wget https://files.seeedstudio.com/wiki/ReTerminal/Qt/QtQuick.tar.gz
rm -rf ~/Seeed_Python_ReTerminalQt5Examples/imports/armv7l/QtQuick
tar xvf ~/QtQuick.tar.gz -C ~/Seeed_Python_ReTerminalQt5Examples/imports/armv7l

cp -v ~/Seeed_Python_ReTerminalQt5Examples/src/r2.desktop ~/Desktop/

sudo apt install at-spi2-core onboard qml-module-qtquick-shapes python3-pyside2.qtqml python3-pyside2.qtcore python3-pyside2.qtnetwork python3-pyside2.qtgui python3-pyside2.qtwidgets python3-pyside2.qtquick qtquickcontrols2-5-dev qml-module-qtcharts qml-module-qtquick-controls qml-module-qtquick-controls2 python3-evdev python3-psutil
```

**Run Qt applicaiton:**

```
pi@raspberrypi:~ $ cd ~/Seeed_Python_ReTerminalQt5Examples/src
pi@raspberrypi:~/Seeed_Python_ReTerminalQt5Examples/src $ sudo DISPLAY=:0.0 python3 main.py
this app use pyside2
No protocol specified
qt.qpa.xcb: could not connect to display :0.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

Aborted
pi@raspberrypi:~/Seeed_Python_ReTerminalQt5Examples/src $
```

lakshanthad, 2022-02-28:
Hi @matsujirushi,

Could you double-click on the application "r2.desktop" and try?

matsujirushi, 2022-03-01:
Hi @lakshanthad ,

> Could you double-click on the application "r2.desktop" using reTerminal LCD and try again?

It worked fine. 🤩

[image]
[image]
[image]

lakshanthad, 2022-03-01:
Great! Any further questions? Shall I close the issue?

matsujirushi, 2022-03-01:
You can close this issue. 
Thank you so much.

closed 2022-03-01 by lakshanthad