Seeed-Studio/seeed-linux-dtoverlays #32: Installation Bug with ReSpeaker 2-Mics Pi HAT / WM8960 on Jetson Nano

ClemensKubach, 2022-02-06:
I tried to install the driver from the jtsn-wm8960 module to my Jetson Nano B0. I followed the described steps of the [README](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/tree/master/modules/jtsn-wm8960#readme). In the process I received the following error when executing `sudo /opt/nvidia/jetson-io/config-by-hardware.py -n "Seeed Voice Card 2MIC"`:
> $ sudo /opt/nvidia/jetson-io/config-by-hardware.py -n "Seeed Voice Card 2MIC"
Traceback (most recent call last):
  File "/opt/nvidia/jetson-io/config-by-hardware.py", line 125, in <module>
    main()
  File "/opt/nvidia/jetson-io/config-by-hardware.py", line 117, in main
    dtbo = configure_jetson(jetson, header, hw)
  File "/opt/nvidia/jetson-io/config-by-hardware.py", line 55, in configure_jetson
    % (hw, header))
NameError: No configuration found for Seeed Voice Card 2MIC on Jetson 40pin Header!

(I had to skip the command `sudo cp overlays/jetsonnano/jetson-seeed-2mic-wm8960.dtbo /boot`, because the file "jetson-seeed-2mic-voicecard.dtbo" is already created there. If I am not skipping this step of copying, the following error occurs instead:
> Multiple DT overlays for 'Seeed Voice Card 2MIC' found!
Please remove duplicate(s)
 /boot/jetson-seeed-2mic-wm8960.dtbo
/boot/jetson-seeed-2mic-voicecard.dtbo

)

To solve this issue, I added the following line `jetson-header-name = "Jetson 40pin Header";` to [jetson-seeed-2mic-wm8960.dts](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/3eb576a7baa015c80d505cbfea2910291e6fada5/overlays/jetsonnano/jetson-seeed-2mic-wm8960.dts). The property "jetson-header-name" seems to be a requirement following the [nvidia documentation](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/hw_setup_jetson_io.html#wwpID0E05B0HA) and was missing in the compiled .dtbo file.
Now, the "Seeed Voice Card 2MIC" Device is listed when executing `sudo /opt/nvidia/jetson-io/config-by-hardware.py -l`. This wasnt the case before and seems to be a fix for the first problem. 

However, there occurs another error, when trying to set the pin config of the Jetson 40pin Header to "Seeed Voice Card 2MIC" with `config-by-hardware.py`:
>Traceback (most recent call last):
  File "/opt/nvidia/jetson-io/config-by-hardware.py", line 125, in <module>
    main()
  File "/opt/nvidia/jetson-io/config-by-hardware.py", line 117, in main
    dtbo = configure_jetson(jetson, header, hw)
  File "/opt/nvidia/jetson-io/config-by-hardware.py", line 57, in configure_jetson
    jetson.hw_addon_load(hw)
  File "/opt/nvidia/jetson-io/Jetson/board.py", line 295, in hw_addon_load
    node)
RuntimeError: Failed to get pin number for node /fragment@4/__overlay__/header-40pin-pinmux/pin12/!

Any idea on how to fix that and achieve setting the pin config correctly?

KunalAivero, 2022-02-18:
Hi, I have hit the exact error while using Jetpack 4.6 (i.e. L4T 32.6.1 release). Loading dtbo works ok with Jetpack 4.5 and older releases of Jetpack. Can some one please look in to and fix the issue.

mingzhangqun, 2022-02-23:
Hi, this issue has been resolved. 
It's because in tegra210-P3448-0000-p3449-0000-hdr40.dts pinmux nodes' name are started with hdr40 prefix.
https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/b18755e0cbbfc312b7035f968bfd473bf5f06562

mingzhangqun, 2022-02-23:
Resolved.

closed 2022-02-23 by mingzhangqun

ClemensKubach, 2022-03-07:
I'm still getting the same error using Jetpack 4.6 (flushed on sdcard): `RuntimeError: Failed to get pin number for node /fragment@4/__overlay__/header-40pin-pinmux/hdr40-pin12/!` when running `sudo /opt/nvidia/jetson-io/config-by-hardware.py -n "Seeed Voice Card 2MIC"`. It works for all other configurations listed with `config-by-hardware.py -l`:

> 1. Adafruit SPH0645LM4H
> 2. FE-PI Audio V1 and Z V2
> 3. MCP251x CAN Controller
> 4. ReSpeaker 4 Mic Array
> 5. Seeed 2xMCP2518FD
> 6. Seeed Voice Card 2MIC
> 7. reComputer sdmmc

mingzhangqun, 2022-03-08:
@ClemensKubach Hi, Can you show me this:
```bash
$cat /proc/device-tree/compatible
nvidia,p3449-0000-b00+p3448-0002-b00nvidia,jetson-nanonvidia,tegra210
```

ClemensKubach, 2022-03-08:
Hi @mingzhangqun , I am getting something different:

```shell
$cat /proc/device-tree/compatible
nvidia,p3449-0000-a02+p3448-0000-a02nvidia,jetson-nanonvidia,tegra210
```
I double checked the board version and found that I had mixed up the version in my initial comment. I have a 4GB A0(2) board.

mingzhangqun, 2022-03-08:
could you give me the full print:
```bash
sudo /opt/nvidia/jetson-io/config-by-hardware.py -l
```

ClemensKubach, 2022-03-08:
Thanks for taking a care of the issue. I receive the following:
```shell
$ sudo /opt/nvidia/jetson-io/config-by-hardware.py -l
Configurations for the following hardware modules are available:
1. Adafruit SPH0645LM4H
2. FE-PI Audio V1 and Z V2
3. MCP251x CAN Controller
4. ReSpeaker 4 Mic Array
5. Seeed 2xMCP2518FD
6. Seeed Voice Card 2MIC
7. reComputer sdmmc
```
The [dts.txt file](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/8205996/dts.txt) is attached and the command produced also the following:
```shell
$ dtc -I fs /proc/device-tree > ~/dts.txt
<stdout>: Warning (unit_address_vs_reg): Node /reserved-memory/fb0_carveout has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /reserved-memory/ramoops_carveout has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /reserved-memory/fb1_carveout has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /reserved-memory/iram-carveout has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /ptm has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /ahub/admaif@0x702d0000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/amx@702d3100 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/afc@702d7300 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/i2s@702d1000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/iqc@702de000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/i2s@702d1300 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/dmic@702d4200 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/adx@702d3900 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/mvc@702da000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/sfc@702d2400 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/amx@702d3000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/afc@702d7200 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/afc@702d7500 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/i2s@702d1200 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/dmic@702d4100 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/amixer@702dbb00 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/sfc@702d2000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/ope@702d8400 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/adx@702d3800 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/iqc@702de200 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/mvc@702da200 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/sfc@702d2600 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/afc@702d7100 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/ope@702d8000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/afc@702d7400 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/i2s@702d1100 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/dmic@702d4000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/i2s@702d1400 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/sfc@702d2200 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/spkprot@702d8c00 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /ahub/afc@702d7000 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /clock has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /lens_imx219@RBPCV2 has a unit name, but no reg property
<stdout>: Warning (unit_address_vs_reg): Node /gpu has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /camera-pcl/dpd/csie has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /camera-pcl/dpd/csic has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /camera-pcl/dpd/csia has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /camera-pcl/dpd/csif has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /camera-pcl/dpd/csid has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /camera-pcl/dpd/csib has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /adma@702e2000 has a unit name, but no reg property
[... cut ...]
I solved this "No DTB found" problem with the [solution (from here)](https://forums.developer.nvidia.com/t/jetpack-4-3-l4t-r32-3-1-released/109271/22) runnning `$ sudo cp -v /boot/tegra210-p3448-0000-p3449-0000-[ab]0[012].dtb /boot/dtb/`.
Now there is 

```shell
$ ls /boot/dtb
kernel_tegra210-p3448-0000-p3449-0000-b00.dtb  tegra210-p3448-0000-p3449-0000-a02.dtb
tegra210-p3448-0000-p3449-0000-a00.dtb         tegra210-p3448-0000-p3449-0000-b00.dtb
tegra210-p3448-0000-p3449-0000-a01.dtb
```

After that, jetson-io is working and the above outputs can be reproduced.

mingzhangqun, 2022-03-10:
Hi, please confirm:
```
$ ls -l /boot/tegra210-p3448-0000-p3449-0000-a02-hdr40.dtbo
```
convert dtbo to dts:
```
$ fdtdump tegra210-p3448-0000-p3449-0000-a02-hdr40.dtbo > fdtdump tegra210-p3448-0000-p3449-0000-a02-hdr40.dts
```
check if the compatible is contained with your board:
```
$ cat fdtdump tegra210-p3448-0000-p3449-0000-a02-hdr40.dts
compatible = "nvidia,p3449-0000-b00+p3448-0000-b00", "nvidia,p3449-0000-a02+p3448-0000-a02";
```
if not contained, added it to the file(such as my emmc version):
```
$ cat fdtdump tegra210-p3448-0000-p3449-0000-a02-hdr40.dts
compatible = "nvidia,p3449-0000-b00+p3448-0000-b00", "nvidia,p3449-0000-b00+p3448-0002-b00", "nvidia,p3449-0000-a02+p3448-0000-a02";
```
then convert the dts to dtbo:
```
$ dtc -I dts -O dtb tegra210-p3448-0000-p3449-0000-a02-hdr40.dts > tegra210-p3448-0000-p3449-0000-a02-hdr40.dtbo
$ sudo cp tegra210-p3448-0000-p3449-0000-a02-hdr40.dtbo /boot/
$ sudo /opt/nvidia/jetson-io/config-by-hardware.py -l
Header 1 [default]: Jetson 40pin Header
  Available hardware modules:
  1. Seeed Voice Card 2MIC
$ sudo /opt/nvidia/jetson-io/config-by-hardware.py -n "Seeed Voice Card 2MIC"
Configuration saved to /boot/kernel_tegra210-p3448-0002-p3449-0000-b00-user-custom.dtb.
Reboot system to reconfigure.
```
[tegra210-p3448-0000-p3449-0000-a02-hdr40.zip](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/8221049/tegra210-p3448-0000-p3449-0000-a02-hdr40.zip)

ClemensKubach, 2022-03-10:
I can confirm that the mentioned compatible `compatible = "nvidia,p3449-0000-b00+p3448-0000-b00", "nvidia,p3449-0000-a02+p3448-0000-a02";` was in the `/boot/tegra210-p3448-0000-p3449-0000-a02-hdr40.dtbo` file.

One significant difference to your tegra dtbo was the prefix "hdr40" in "hdr40-pinNUM". I added it and also tested some other settings, but nothing solved the problem.

What is the meaning of 0002 in "nvidia,p3449-0000-b00+p3448-0002-b00" of your extended compatible?

mingzhangqun, 2022-03-11:
0002 means it's an EMMC version.

ClemensKubach, 2022-03-11:
Yes, there isn't any line about a header in my config-by-hardware.py.
I'll try reinstalling the OS with the SDK Manager in the next days and adding 0002-a02 as a compatible. I will report back then.

mingzhangqun, 2022-03-11:
okay, I think the mainly problem is the compatible.
You need to find a way to make tegra210-p3448-0000-p3449-0000-a02-hdr40.dtbo work.

ClemensKubach, 2022-03-11:
I finally got it working, somehow. I re-downloaded, installed and tested JetPack 4.5 and 4.6 again. The problem will probably have been a version/installation error. So with the image of JP 4.6 and your latest commit to the driver, I can also confirm that it works for the A02 board as well.

The respeaker works very well in analog mono mode, even directly with the default settings. Only in analog stereo input it is very noisy with the default settings, so you can hardly hear what is relevant.

The last thing I noticed is that 2 dtbos with `overlay-name = "Seeed Voice Card 2MIC";` are created with the procedure described in the README: jetson-seeed-2mic-voicecard.dtbo & jetson-seeed-2mic-wm8960.dtbo. So one must always be removed from /boot. They don't seem to be identical but both are working. Is there any relevant difference? Which one should be chosen or otherwise - maybe one should not be created in the make script or there might be something in the readme useful about this?

Thanks a lot for your support @mingzhangqun !

mingzhangqun, 2022-03-12:
The jetson-seeed-2mic-wm8960.dtbo is included by jetson-seeed-2mic-voicecard.dtbo, voicecard is used to enable an addtional spi port. And they have the share the compatible string.