Seeed-Studio/seeed-linux-dtoverlays #33: cannnot get events of user programmable buttons

matsujirushi, 2022-02-26:
**Environment:**

* reTerminal
* Raspberry Pi OS 32-bit Bullseye (2022-01-28)

---

**Procedure:**

* https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/31#issuecomment-1049636159

1. This fix can be used first:  
https://lab.seeed.co.jp/entry/2022/01/20/120000
2. After that, you need to delete the line "dtoverlay=reTerminal-bridge" inside "/boot/config.txt" for LEDs and Buzzer to work

**What happened:**

I cannnot get events of user programmable buttons.
`gpio_keys` is not displayed.

```
pi@raspberrypi:~ $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      ST LIS3LV02DL Accelerometer
/dev/input/event1:      vc4
/dev/input/event2:      seeed-tp
/dev/input/event3:      vc4
Select the device event number [0-3]:
```

**Links:**

* [4 User Programmable Buttons](https://wiki.seeedstudio.com/reTerminal-hardware-interfaces-usage/#4-user-programmable-buttons)

lakshanthad, 2022-02-27:
Hi @matsujirushi,

Could you please share the output of "i2cdetect -y 1"?

Thank you.

matsujirushi, 2022-02-28:
Hi @lakshanthad ,
```
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- UU -- -- -- -- -- --
20: 20 -- -- -- -- -- -- -- -- UU -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberrypi:~ $
```

lakshanthad, 2022-02-28:
Hi @matsujirushi,

You need to do this:
https://wiki.seeedstudio.com/reTerminal-FAQ/#q9-why-does-buzzer-leds-and-buttons-do-not-work-after-installing-the-reterminal-drivers

matsujirushi, 2022-02-28:
Hi @lakshanthad ,
It's fixed.
Thank you.

**evtest:**

```
pi@raspberrypi:~ $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      gpio_keys
/dev/input/event1:      ST LIS3LV02DL Accelerometer
/dev/input/event2:      seeed-tp
/dev/input/event3:      vc4
/dev/input/event4:      vc4
Select the device event number [0-4]: 0
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x1 product 0x1 version 0x100
Input device name: "gpio_keys"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 142 (KEY_SLEEP)
Properties:
Testing ... (interrupt to exit)
Event: time 1646024516.690089, type 1 (EV_KEY), code 30 (KEY_A), value 1
Event: time 1646024516.690089, -------------- SYN_REPORT ------------
Event: time 1646024516.870041, type 1 (EV_KEY), code 30 (KEY_A), value 0
Event: time 1646024516.870041, -------------- SYN_REPORT ------------
Event: time 1646024518.680089, type 1 (EV_KEY), code 31 (KEY_S), value 1
Event: time 1646024518.680089, -------------- SYN_REPORT ------------
Event: time 1646024518.810108, type 1 (EV_KEY), code 31 (KEY_S), value 0
Event: time 1646024518.810108, -------------- SYN_REPORT ------------
Event: time 1646024519.070091, type 1 (EV_KEY), code 32 (KEY_D), value 1
Event: time 1646024519.070091, -------------- SYN_REPORT ------------
Event: time 1646024519.200125, type 1 (EV_KEY), code 32 (KEY_D), value 0
Event: time 1646024519.200125, -------------- SYN_REPORT ------------
^Cpi@raspberrypi:~ $
```

**i2cdetect:**

```
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- UU -- -- -- -- -- --
20: UU -- -- -- -- -- -- -- -- UU -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberrypi:~ $
```

closed 2022-02-28 by matsujirushi