Seeed-Studio/seeed-linux-dtoverlays #34: failed to import hw:0 use case configuration -2

matsujirushi, 2022-03-01:
Error messages are displayed when running reTerminal.sh.

* **error: failed to import hw:0 use case configuration -2**
* **error: failed to import hw:1 use case configuration -2**

Is it okay to ignore it?

**Environment:**

* reTerminal
* Raspberry Pi OS 32-bit Bullseye (2022-01-28)

**Procedure:**

```
git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
cd seeed-linux-dtoverlays
sudo ./scripts/reTerminal.sh
```

**Displayed message:**

```
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/seeed.plymouth' -> '/usr/share/plymouth/themes/seeed/seeed.plymouth'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/seeed.script' -> '/usr/share/plymouth/themes/seeed/seeed.script'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/splash.png' -> '/usr/share/plymouth/themes/seeed/splash.png'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/splash_v.png' -> '/usr/share/plymouth/themes/seeed/splash_v.png'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/10-disp.conf' -> '/usr/share/X11/xorg.conf.d/10-disp.conf'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/plymouthd.conf' -> '/etc/plymouth/plymouthd.conf'
No state is present for card vc4hdmi0
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi0
No state is present for card vc4hdmi1
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi1
------------------------------------------------------
Please reboot your device to apply all settings
Enjoy!
------------------------------------------------------
```

**Ref:**

* https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/31

matsujirushi, 2022-03-01:
```
pi@raspberrypi:~/seeed-linux-dtoverlays $ sudo alsactl -L restore
No state is present for card vc4hdmi0
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi0
No state is present for card vc4hdmi1
alsa-lib main.c:1014:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi1
pi@raspberrypi:~/seeed-linux-dtoverlays $
```

lakshanthad, 2022-03-01:
Hello @matsujirushi,

Looks like this error relates to audio drivers. This will not affect the general usage of reTerminal. 

However, I will try to connect a reSpeaker HAT and check whether it works to confirm about this. Please wait.

lakshanthad, 2022-03-02:
Hi @matsujirushi,

I connected a reSpeaker 2-Mic Pi HAT to reTerminal, and it works well. There is no audio issue. The reSpeaker is identified as hw:2 as you can see from the below outputs:

```sh
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM i2s-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

According to the error message above, hw:0/ hw:1 should correspond to the audio output via HDMI. I connected reTerminal to an HDMI display that had speakers, tried to play audio and it played well. So I think we can ignore this issue for now.

matsujirushi, 2022-03-03:
Hi @lakshanthad ,

> So I think we can ignore these errors for now.

I understood.
Thanks for check.

closed 2022-03-03 by matsujirushi