Seeed-Studio/seeed-linux-dtoverlays #35: Installation fails for CAN-HAT on RPI3

terilenard, 2022-03-10:
Configuration:
- RaspberryPi 3
- Kernel:  4.19.118-v8+
- RaspberryPi OS

While running _seeed-linux-dtoverlays/modules/CAN-HAT/install.sh_ the following error bellow. I tried the installation also on the latest kernel for the RaspberryPi OS. Still, the installation fails.

```
Error:
------------------------------
Deleting module version: completely from the DKMS 
Done.                                                                                                                                                                                                                Creating symlink /var/lib/dkms/mcp25xxfd/0.1/source -> /usr/src/mcp25xxfd-0.1

DKMS: add completed.                                                                                                                 
Error! Your kernel headers for kernel 4.19.36+ cannot be found.
                                                                      
Please install the linux-headers-4.19.36+ package,                                                                                   
or use the --kernelsourcedir option to tell DKMS where it's located                                                                  
Error! Your kernel headers for kernel 4.19.36-v7+ cannot be found.                                                                   
Please install the linux-headers-4.19.36-v7+ package,                                                                                
or use the --kernelsourcedir option to tell DKMS where it's located                                                    
Error! Your kernel headers for kernel 4.19.36-v7l+ cannot be found.                                                                  
Please install the linux-headers-4.19.36-v7l+ package,                                                                               
or use the --kernelsourcedir option to tell DKMS where it's located                                                                  
------------------------------------------------------                                                                               
Please reboot your raspberry pi to apply all settings                                                                                
Enjoy!
------------------------------------------------------  

```

terilenard, 2022-03-11:
The solution I found so far:

1. Use a armfh RaspberryPi OS, not arm64. The image I have installed has the kernel version 5.10.92-v7+.
2. After you flash the sd card, and run install.sh, the installation will fail, but will install the latest version of the kernel headers, which at this point for me is 5.10.103-v7+.
3. After, I have created the following symlinks to link the sources to the currently running kernel:

```
ln -s /usr/src/linux-headers-5.10.103+  /lib/modules/5.10.92+/build

ln -s /usr/src/linux-headers-5.10.103-v7+  /lib/modules/5.10.92-v7+/build

ln -s /usr/src/linux-headers-5.10.103-v7l+  /lib/modules/5.10.92-v7l+/build
```

After doing this, I have re-run install.sh and I had to modify the /boot/config.txt as such:

1. Added: `dtoverlay=seeed-can-fd-hat-v2 `
2. Commented: `dtoverlay=2xMCP2518FD-spi0 `

bigbearishappy, 2022-03-14:
Cool. Thank you for your contribution of solving this problem.

closed 2022-03-14 by bigbearishappy

terilenard, 2022-03-14:
@bigbearishappy Can we get a fix in the script so it's not necessary to do all those steps?

bigbearishappy, 2022-03-14:
> @bigbearishappy Can we get a fix in the script so it's not necessary to do all those steps?

OK, I will fix this and let you know.

bigbearishappy, 2022-03-28:
Working on it now~

bigbearishappy, 2022-03-30:
@terilenard Just fixed it. See more detail in this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/03baea4fb44881055b7f26e4584238b2510919b6)

bigbearishappy, 2022-04-21:
@terilenard 
If you are work with the latest kernel. It's no need to install the CAN-HAT driver  manually.
We have alread merge the code into the mainline of raspberrypi kernle.Here is the [LINK](https://github.com/raspberrypi/linux/pull/4072)