Seeed-Studio/seeed-linux-dtoverlays #36: Image Seeed ReTerminal CM4 with Ubuntu

oleblueyes, 2022-03-11:
I am having difficulty with overcoming the embedded OS on the reTerminal CM4 offered by Seeed Studios. I have tried everything I can find except cracking the screen off to access the rp unit itself and actually get access to the microSD port. Does anyone have any suggestions on what steps I may be able to take before I break this unit open?

Thank you all in advance.

bigbearishappy, 2022-03-14:
If you have problem with the using of OS offered by Seeed. You can try to install the ubuntu on reTerminal by following the steps of this [issue](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/28#issuecomment-939662366).
And if you are not familiar with the reTerminal or CM4. You can get all you want from the wiki of [Seeed Studio](https://wiki.seeedstudio.com/reTerminal/)

If you have any problem, please let us know.
Waiting for you reply.

lakshanthad, 2022-03-14:
Hello @oleblueyes,

If you want to install Ubuntu, I have uupdated the below wiki:
https://wiki.seeedstudio.com/reTerminal-FAQ/#q10-how-can-i-install-ubuntu-on-reterminal

Please refer to it and let me know if you have any problems.

lakshanthad, 2022-04-28:
Since there is no more activity, I will close this for now. Please open again if needed.

closed 2022-04-28 by lakshanthad