Seeed-Studio/seeed-linux-dtoverlays #37: mipi_dsi reTerminal build fails: internal error in set_address, at ../../gold/output.h:322

RaymiiOrg, 2022-03-16:
Building the Yocto image for the reTerminal fails on Debian 11 with Yocto Hardknott:
  
```
      |   LD [M]  /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/mipi_dsi.ko
      | aarch64-poky-linux-ld: internal error in set_address, at ../../gold/output.h:322

```

bigbearishappy, 2022-03-17:
Hello.
We don't support the yocto Hardknott. If you want to use the latest code of yocto. We recommend you to use the yocto honister.

RaymiiOrg, 2022-03-17:
Do you have any idea what causes the error, or more specifically, why the previous commit mentioned does work? The previous commit mentioned is also on hardknott, my guess would be that the issue isn't per se hardknott or honister, but in the source code of the module.

bigbearishappy, 2022-03-17:
Can you show me the whole error log of "bitbake seeed-linux-dtoverlays -c do_build".

RaymiiOrg, 2022-03-17:
Here is the full output:

```
remy@hostname:~/b2qt/build-raspberrypi4-64$ bitbake seeed-linux-dtoverlays -c do_build
WARNING: Host distribution "debian-11" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |################################################################################################| Time: 0:00:00
Loaded 4378 entries from dependency cache.
Parsing recipes: 100% |##############################################################################################| Time: 0:00:00
Parsing of 2813 .bb files complete (2810 cached, 3 parsed). 4377 targets, 611 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "b2qt"
DISTRO_VERSION       = "3.3.4"
TUNE_FEATURES        = "aarch64 armv8a crc crypto cortexa72"
TARGET_FPU           = ""
SDKMACHINE           = "x86_64"
meta                 
meta-poky            = "HEAD:c40ac16d79026169639f47be76a3f7b9d8b5178e"
meta-raspberrypi     = "HEAD:b4ec97e4eb8e36efd1f7e2f8ae020a9e55cfc239"
meta-oe              
meta-python          
meta-networking      
meta-initramfs       
meta-multimedia      = "HEAD:f72a73b42fa740130b388ba8555cdbefdee8d37d"
meta-python2         = "HEAD:810d6d842f103eb59f18b06426106462b15de7e2"
meta-boot2qt         
meta-boot2qt-distro  = "HEAD:e59a2e20697e0afc2a0b068835cd90e102dec589"
meta-mingw           = "HEAD:422b96cb2b6116442be1f40dfb5bd77447d1219e"
meta-qt6             = "HEAD:eb3719266fc03b96d5056980b135b371f31811f4"
meta-seeed-reterminal = "main:57d1b68d73e625fe6a4cb14372a0cb7c42bae9c5"
meta-raymii          = "master:84123f093be34a9a4d73de545132cffc3e210c19"

Initialising tasks: 100% |###########################################################################################| Time: 0:00:01
Sstate summary: Wanted 11 Local 6 Network 0 Missed 5 Current 221 (54% match, 97% complete)
NOTE: Executing Tasks
ERROR: seeed-linux-dtoverlays-1.0-r0 do_compile: oe_runmake failed
ERROR: seeed-linux-dtoverlays-1.0-r0 do_compile: ExecutionError('/home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/temp/run.do_compile.1275688', 1, None, None)
ERROR: Logfile of failure stored in: /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/temp/log.do_compile.1275688
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 ARCH=arm64 KBUILD=/home/remy/b2qt/build-raspberrypi4-64/tmp/work-shared/raspberrypi4-64/kernel-source O=/home/remy/b2qt/build-raspberrypi4-64/tmp/work-shared/raspberrypi4-64/kernel-build-artifacts CROSS_COMPILE=aarch64-poky-linux- all_rpi
|   DTC     overlays/rpi/lis331dlh-overlay.dtbo
|   DTC     overlays/rpi/tpm-spi-slx9670-overlay.dtbo
|   DTC     overlays/rpi/2xMCP2517FD-overlay.dtbo
|   DTC     overlays/rpi/2xMCP2518FD-spi0-overlay.dtbo
|   DTC     overlays/rpi/reTerminal-bridge-overlay.dtbo
|   DTC     overlays/rpi/grove-led-overlay.dtbo
|   DTC     overlays/rpi/reTerminal-overlay.dtbo
|   DTC     overlays/rpi/bma456-i2c1-overlay.dtbo
| overlays/rpi/2xMCP2518FD-spi0-overlay.dts:59.25-68.15: Warning (spi_bus_reg): /fragment@4/__overlay__/can@0: SPI bus unit address format error, expected "1"
| overlays/rpi/2xMCP2518FD-spi0-overlay.dts:70.25-78.15: Warning (spi_bus_reg): /fragment@4/__overlay__/can@1: SPI bus unit address format error, expected "0"
| overlays/rpi/2xMCP2517FD-overlay.dts:93.25-102.15: Warning (spi_bus_reg): /fragment@5/__overlay__/can@1: SPI bus unit address format error, expected "0"
| overlays/rpi/grove-led-overlay.dts:33.3-32: Warning (label_is_string): /__overrides__:label: property is not a string
| overlays/rpi/grove-led-overlay.dts:30.3-31.26: Warning (gpios_property): /__overrides__:gpio: property size (22) is invalid, expected multiple of 4
| overlays/rpi/reTerminal-overlay.dts:23.15-33.5: Warning (avoid_unnecessary_addr_size): /fragment@1/__overlay__: unnecessary #address-cells/#size-cells without "ranges" or child "reg" property
| make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
|   CC [M]  /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/mipi_dsi_drv.o
|   CC [M]  /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/touch_panel.o
|   CC [M]  /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/panel-ili9881d.o
|   LD [M]  /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/mipi_dsi.o
|   MODPOST /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/Module.symvers
|   CC [M]  /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/mipi_dsi.mod.o
|   LD [M]  /home/remy/b2qt/build-raspberrypi4-64/tmp/work/raspberrypi4_64-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/mipi_dsi/mipi_dsi.ko
[... cut ...]
ERROR: Task (/home/remy/b2qt/sources/meta-seeed-reterminal/recipes-kernel/seeed-linux-dtoverlays/seeed-linux-dtoverlays.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 780 tasks of which 776 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/remy/b2qt/sources/meta-seeed-reterminal/recipes-kernel/seeed-linux-dtoverlays/seeed-linux-dtoverlays.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

```

RaymiiOrg, 2022-03-17:
I've solved the issue by not using the `gold` linker. The upstream yocto distro I'm building enabled that, but when I explicitly remove it:

    DISTRO_FEATURES_remove = "ld-is-gold"

the package no longer gives an error when compiling.

closed 2022-03-17 by RaymiiOrg