Seeed-Studio/seeed-linux-dtoverlays #39: can-hat module does not build for kernel 5.15.32

quite, 2022-04-13:
I'm trying to build the CAN-HAT module on a Raspberry Pi 4 running 2022-04-04-raspios-bullseye-armhf-lite.img. This has kernel 5.15.32. Error output follows below. Apparently, things has changed in the CAN api. Build goes through on the 5.10.92 kernel of 2022-01-28-raspios-bullseye-armhf-lite.

```
CC [M]  /tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.o
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c: In function 'mcp25xxfd_handle_tefif_one':
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:1313:3: error: too few arguments to function 'can_rx_offload_get_echo_skb'
 1313 |   can_rx_offload_get_echo_skb(&priv->offload,
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /tmp/seeed-linux-dtoverlays/modules/CAN-HAT/spi_can_compatible.h:18,
                 from /tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd.h:15,
                 from /tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:26:
./include/linux/can/rx-offload.h:47:14: note: declared here
   47 | unsigned int can_rx_offload_get_echo_skb(struct can_rx_offload *offload,
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c: In function 'mcp25xxfd_hw_rx_obj_to_skb':
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:1482:14: error: implicit declaration of function 'can_dlc2len'; did you mean 'can_fd_dlc2len'? [-Werror=implicit-function-declaration]
 1482 |   cfd->len = can_dlc2len(get_canfd_dlc(dlc));
      |              ^~~~~~~~~~~
      |              can_fd_dlc2len
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:1482:26: error: implicit declaration of function 'get_canfd_dlc' [-Werror=implicit-function-declaration]
 1482 |   cfd->len = can_dlc2len(get_canfd_dlc(dlc));
      |                          ^~~~~~~~~~~~~
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:1487:14: error: implicit declaration of function 'get_can_dlc' [-Werror=implicit-function-declaration]
 1487 |   cfd->len = get_can_dlc(FIELD_GET(MCP25XXFD_OBJ_FLAGS_DLC,
      |              ^~~~~~~~~~~
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c: In function 'mcp25xxfd_tx_obj_from_skb':
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:2321:8: error: implicit declaration of function 'can_len2dlc'; did you mean 'can_fd_len2dlc'? [-Werror=implicit-function-declaration]
 2321 |  dlc = can_len2dlc(cfd->len);
      |        ^~~~~~~~~~~
      |        can_fd_len2dlc
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c: In function 'mcp25xxfd_start_xmit':
/tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:2423:2: error: too few arguments to function 'can_put_e
cho_skb'
 2423 |  can_put_echo_skb(skb, ndev, tx_head);
      |  ^~~~~~~~~~~~~~~~
In file included from ./include/linux/can/dev.h:23,
                 from /tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd.h:14,
                 from /tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.c:26:
./include/linux/can/skb.h:20:5: note: declared here
   20 | int can_put_echo_skb(struct sk_buff *skb, struct net_device *dev,
      |     ^~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:277: /tmp/seeed-linux-dtoverlays/modules/CAN-HAT/mcp25xxfd-core.o] Error 1
make[1]: *** [Makefile:1868: /tmp/seeed-linux-dtoverlays/modules/CAN-HAT] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7l+'
make: *** [Makefile:43: all] Error 2
```

bigbearishappy, 2022-04-13:
There is no need to compile the CAN-HAT kernel driver in bulleyes now.
It's already merged into the main line of raspberrypi kernel.
Here is the PR [link](https://github.com/raspberrypi/linux/pull/4072 )

quite, 2022-04-13:
Oh! Sorry for the noise.

closed 2022-04-13 by quite