Seeed-Studio/seeed-linux-dtoverlays #40: reTerminal LCD does not work after latest RPi Bullseye Update (Kernel 5.15)

lakshanthad, 2022-04-25:
**Describe the bug**
After flashing the reTerminal with the latest Raspberry Pi OS Bullseye found [here](https://www.raspberrypi.com/software/operating-systems/), the LCD does not work anymore.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash reTerminal with latest RPi OS Bullseye (Kernel 5.15) using RPi Imager
2. git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
3. cd seeed-linux-dtoverlays
4. sudo ./scripts/reTerminal.sh
5. Reboot

**Expected behavior**
LCD worked with Kernal 5.10 before

**Debug output**
The output of "dmesg | grep dsi" is as follows:

```sh
root@raspberrypi:~# dmesg | grep dsi
[    0.069716] platform fe700000.dsi: Fixing up cyclic dependency with fe101000.cprman
[    5.400918] i2c 1-0045: Fixing up cyclic dependency with fe700000.dsi
[    8.090686] i2c_mipi_dsi 1-0045: I2C read id: 0xc3
[    8.094604] [DSI]mipi_dsi_device:
[    8.315444] [DSI]mipi_dsi_probe:
[    8.899624] mipi_dsi fe700000.dsi.0: failed to attach dsi to host: -517
[    8.899791] [DSI]mipi_dsi_probe:
[    9.066846] vc4-drm gpu: bound fe700000.dsi (ops vc4_dsi_ops [vc4])
[   10.473819] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[   10.473836] vc4_dsi fe700000.dsi: instat: 0x00000000
[   10.473848] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[   17.273826] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[   17.273851] vc4_dsi fe700000.dsi: instat: 0x00000000
[   17.273871] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[   74.635294] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[   74.635319] vc4_dsi fe700000.dsi: instat: 0x00000000
[   74.635338] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
[   80.955308] vc4_dsi fe700000.dsi: transfer interrupt wait timeout
[   80.955334] vc4_dsi fe700000.dsi: instat: 0x00000000
[   80.955354] [drm:vc4_dsi_host_transfer [vc4]] *ERROR* DSI transfer failed, resetting: -110
```

However. the LCD driver shows as loaded at 0x45:

```sh
root@raspberrypi:~# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- UU -- -- -- -- -- --
20: UU -- -- -- -- -- -- -- -- UU -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- UU -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Please help to fix this. Thank you.

Best Regards,
Lakshantha

matsujirushi, 2022-04-28:
@lakshanthad 
Hi,
Currently, in testing.
https://github.com/matsujirushi/seeed-linux-dtoverlays/tree/reterminal/kernel5.15

1. reTerminal pre-install OS ... OK
2. Raspberry Pi OS 32-bit bullseye(2022/4/4) ... OK
3. Raspberry Pi OS 32-bit buster(2022/4/4) ... WIP

lakshanthad, 2022-04-28:
Hello @matsujirushi,

Wonderful! That is great news! Looking forward to 64-bit testing also!

Best Regards,
Lakshanthad

lakshanthad, 2022-04-28:
PR can be tracked here:
https://github.com/Seeed-Studio/seeed-linux-dtoverlays/pull/41

nineisk, 2022-04-28:
Hi,

LCD worked for me.

[https://github.com/nineisk/seeed-linux-dtoverlays](url)

```
pi@reterminal:~ $ dmesg | grep DSI
[    5.652919] [DSI]i2c_md_init:
[    5.653168] [DSI]i2c_md_probe:start
[    5.660579] [DSI]mipi_dsi_device:
[    5.684693] [DSI]i2c_md_probe:finished.
[    5.749896] [DSI]mipi_dsi_probe:
[    6.273713] [DSI]mipi_dsi_probe:
[    6.515626] [DSI]panel_prepare:
[    6.571830] [DSI]ili9881d_prepare:
[    6.699863] [DSI]panel_enable:
[   10.620155] [DSI]panel_disable:
[   10.623447] [DSI]panel_unprepare:
[   10.679652] [DSI]panel_prepare:
[   10.735693] [DSI]ili9881d_prepare:
[   10.863702] [DSI]panel_enable:
[  610.674913] [DSI]panel_disable:
[  610.678012] [DSI]panel_unprepare:
[ 1204.750979] [DSI]panel_prepare:
[ 1204.809643] [DSI]ili9881d_prepare:
[ 1204.937674] [DSI]panel_enable:
pi@reterminal:~ $ uname -a
Linux reterminal 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

matsujirushi, 2022-04-30:
Note:
https://forums.raspberrypi.com/viewtopic.php?p=1989017

billstanden, 2022-05-04:
> PR can be tracked here:
> #41

This does not appear to resolve my issue on an reTerminal still on the older screen firmware version.

lakshanthad, 2022-05-05:
LCD is working now according to this PR:
https://github.com/Seeed-Studio/seeed-linux-dtoverlays/pull/41

closed 2022-05-05 by lakshanthad

maziarzamani, 2022-10-02:
Hi there. I have similar problems with Bullseye. Got the reTerminal last week and tried to install Bullseye 64-bit. 

```
pi@raspberrypi:~ $ dmesg | grep mipi
[    6.576060] i2c_mipi_dsi 1-0045: I2C read id: 0xc3
[    6.577774] [DSI]mipi_dsi_device:
[    6.608337] [DSI]mipi_dsi_probe:
[    7.248800] mipi_dsi fe700000.dsi.0: failed to attach dsi to host: -517
[    7.266310] [DSI]mipi_dsi_probe:
```