Seeed-Studio/seeed-linux-dtoverlays #45: Enable monitor mode on the BCM2711 wifi chip

KyleTryon, 2022-06-13:
# Issue:
I am currently attempting to make a [pwnagotchi](https://pwnagotchi.ai/) using the reTerminal. This is a tool that makes use of [bettercap](https://www.bettercap.org/), a tool for monitoring wifi. For this to work, the wifi driver must have monitoring mode enabled. This will work if you custom flash the pwnagotchi image linked above. The issue comes when you attempt to install the drivers for the reTerminal screen, which will overwrite the BCM kernel module it seems. 

# Error

```
bettercap no interface matching 'mon0' found.
```

# Steps to reproduce

1. Flash the [pwnagotchi](https://pwnagotchi.ai/)
2. Open `http://<hostname>.local:8080` in the web browser
     a. You will actually first need to modify the pwnagotchi screen type in the config to `inkyphat`, reference this [tutorial](https://cyberarms.wordpress.com/2019/12/13/pwnagotchi-on-a-pi-4-using-any-display/)
     a. You should notice the pwnagotchi webui loads, and is operating correctly.
4. Notice that at this point, Bettercap is working, the Pwnagotchi webui is running and everything _except_ the screen is functional. You should currently see a powered off screen on the device.
5. SSH into the device
6. Flash the reTerminal drivers: https://wiki.seeedstudio.com/reTerminal/#install-reterminal-drivers-after-flashing-new-raspberry-pi-os-ubuntu-os-or-other-os
7. Bettercap will no longer run due to the error above. This is because monitoring mode has been removed from the firmware.

## More

I am not familiar with much relating the kernel, so I am not sure what in these drivers is effectively removing monitoring mode from the wifi chip, but I can see it is referenced as "compatible" in two overlays.dts files

* https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/449394c6e5c54d9ec75ec209e3429b7efa36294c/overlays/rpi/reTerminal-bridge-overlay.dts#L12
* https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/3d57227bd1c549d727360ec780c5db689a082272/overlays/rpi/reTerminal-overlay.dts#L12

lakshanthad, 2022-06-13:
Hello,

Please wait. I will reproduce this and get back to you.

Thank you.

KyleTryon, 2022-06-13:
> Hello,
> 
> Please wait. I will reproduce this and get back to you.
> 
> Thank you.

Highly appreciated, please let me know if I can offer any more information / assistance. 

I did want to add this which may help for others doing investigation

If you SSH into a pwnagchi image, you can see that there is a script called `bettercap-launcher` and it looks like this:

```
#!/usr/bin/env bash                                                                                                                                                          
source /usr/bin/pwnlib                                                                                                                                                       
                                                                                                                                                                             
# we need to decrypt something                                                                                                                                               
if is_crypted_mode; then                                                                                                                                                     
  while ! is_decrypted; do                                                                                                                                                   
    echo "Waiting for decryption..."                                                                                                                                         
    sleep 1                                                                                                                                                                  
  done                                                                                                                                                                       
fi                                                                                                                                                                           
                                                                                                                                                                             
# check if wifi driver is bugged                                                                                                                                             
if ! check_brcm; then                                                                                                                                                        
  if ! reload_brcm; then                                                                                                                                                     
    echo "Could not reload wifi driver. Reboot"                                                                                                                              
    reboot                                                                                                                                                                   
  fi                                                                                                                                                                         
  sleep 10                                                                                                                                                                   
fi                                                                                                                                                                           
                                                                                                                                                                             
# start mon0                                                                                                                                                                 
start_monitor_interface                                                                                                                                                      
                                                                                                                                                                             
if is_auto_mode_no_delete; then                                                                                                                                              
  /usr/bin/bettercap -no-colors -caplet pwnagotchi-auto -iface mon0                                                                                                          
else                                                                                                                                                                         
  /usr/bin/bettercap -no-colors -caplet pwnagotchi-manual -iface mon0                                                                                                        
fi                                  
 
```
 
Specifically it appears that the failure is probably the flag here `-iface mon0` which from what I can see wither means that monitoring mode is disabled, or something else has happened networking wise. The issue only appears after installing the drivers which I am installing to get the screen back on.

lakshanthad, 2022-06-14:
Hello @KyleTryon,

Thank you for the detailed information. When we developed the overlays, it was initially intended to work with Raspberry Pi OS and Ubuntu OS. Because this is what we officially support on the reTerminal and this is what we have tested extensively.

Therefore, It will take some time to track down the root cause of the issue you have mentioned and fix it after installing the [pwnagotchi](https://pwnagotchi.ai/) image and using the `reTerminal.sh` because this is a whole new custom image that we are not familiar with in the past. Can I know the kernel version information when running this image on your reTerminal?

Thank you.

KyleTryon, 2022-06-14:
Thanks @lakshanthad,

Here's what I have done and some information about the current state of the machine.

1. I have just now flashed the pwnagotchi image here: https://github.com/evilsocket/pwnagotchi/releases
2. SSH'd to the Pi
3. I have modified the config file `/etc/pwnagotchi/config.toml` just with some minimal information to ensure the WebUI will run
4. Loaded `http://pwnagotchi.local:8080` in my browser to confirm that everything is working. I have checked additional logs as well and can confirm, that everything I am looking for works, expect for the screen at this point.

Here is some information about the system at this time.

1. `uname -a`

```
pi@pwnagotchi:~ $ uname -a
Linux pwnagotchi 4.19.127-Re4son-v7l+ #1 SMP Sun Feb 21 18:36:03 AEDT 2021 armv7l GNU/Linux
```

2. `cat /proc/version`

```
pi@pwnagotchi:~ $ cat /proc/version
Linux version 4.19.127-Re4son-v7l+ (root@kali-pi-dev) (gcc version 8.2.0 (Debian 8.2.0-6)) #1 SMP Sun Feb 21 18:36:03 AEDT 2021
```

3. `cat /etc/os-release`

```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

4. `find /lib/modules/$(uname -r) -type f -name '*net*.ko'`

```
pi@pwnagotchi:~ $ find /lib/modules/$(uname -r) -type f -name '*net*.ko'
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/9p/9pnet.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/bridge/br_netfilter.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/l2tp/l2tp_netlink.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netrom/netrom.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nft_fib_inet.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nft_fwd_netdev.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nfnetlink_queue.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nfnetlink_osf.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nft_reject_inet.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nf_flow_table_inet.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nft_dup_netdev.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nft_fib_netdev.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nfnetlink.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nf_conntrack_netbios_ns.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nf_conntrack_netlink.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/ipset/ip_set_hash_net.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/ipset/ip_set_hash_netport.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/ipset/ip_set_hash_netiface.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/ipset/ip_set_hash_ipportnet.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nfnetlink_log.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nf_dup_netdev.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/netfilter/nfnetlink_acct.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/sched/sch_netem.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/net/ipv4/inet_diag.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/drivers/net/netconsole.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/drivers/net/usb/sierra_net.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/drivers/net/usb/net1080.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/drivers/leds/trigger/ledtrig-netdev.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/drivers/usb/storage/ums-onetouch.ko
/lib/modules/4.19.127-Re4son-v7l+/kernel/drivers/usb/serial/omninet.ko
```

5. `ip a` Ok, _this_ I think is maybe the most important. Here you can see there is an interface named `mon0` which we know must be missing/broken after installing the drivers as the original error states.

```
pi@pwnagotchi:~ $ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether e4:5f:01:2c:11:e5 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.XX/24 brd 192.168.1.255 scope global dynamic eth0
       valid_lft 83831sec preferred_lft 83831sec
    inet6 fe80::e65f:1ff:fe2c:11e5/64 scope link 
       valid_lft forever preferred_lft forever
3: usb0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether be:7b:70:5d:be:b8 brd ff:ff:ff:ff:ff:ff
    inet 10.0.0.2/24 brd 10.0.0.255 scope global usb0
       valid_lft forever preferred_lft forever
4: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether e4:5f:01:2c:11:e6 brd ff:ff:ff:ff:ff:ff
5: mon0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UNKNOWN group default qlen 1000
    link/ieee802.11/radiotap e4:5f:01:2c:11:e6 brd ff:ff:ff:ff:ff:ff
```

---

This is all _before_ running the `reTerminal.sh` file, and the pwnagotchi web app and bettercap application are currently working.

Let me know if there is anything else you would like to see before I move forward and run the reTerminal script again for comparison. I should have thought before to check the network interfaces, it must be this `mon0` interface that is being affected.

lakshanthad, 2022-06-16:
Hello,

Thank you very much for your detailed response.

I can see that your kernel is v4.19.127. The overlays are guaranteed to work with only kernel v5.10+ and unfortunately, it seems that we cannot further support your issue.

Hope you understand.

Thank you.

KyleTryon, 2022-06-17:
Thank you @lakshanthad, I will look into updating the kernel and re-patching the BCM firmware. Nexmon is the firmware patch that enables monitor mode, it appears that it does support v5+, it may just need to be applied. I will try this over the weekend.

https://github.com/seemoo-lab/nexmon#build-patches-for-bcm43430a1-on-the-rpi3zero-w-or-bcm434355c0-on-the-rpi3rpi4-or-bcm43436b0-on-the-rpi-zero-2w-using-raspbianraspberry-pi-os-recommended

is-qian, 2024-10-18:
If there is more information, please feel free to reopen this issue. Now I will close this issue first.

closed 2024-10-18 by is-qian