Seeed-Studio/seeed-linux-dtoverlays #50: Fail to build lan78xx ethernet driver on Raspberry Pi OS lite (64 bit)

xenOs76, 2022-08-09:
**Describe the bug**
On a fresh install of Raspberry Pi OS lite (64 bit) I'm following the following guide:
https://wiki.seeedstudio.com/Dual-Gigabit-Ethernet-Carrier-Board-for-Raspberry-Pi-CM4/

> Once you flash the Raspberry Pi OS onto the CM4 module, you will get the full Gigabit speed (1Gbps) only on the Ethernet port connected to the Gigabit Ethernet PHY of the CM4 module which is based on Broadcom BCM54210PE (right-side port). The port which is connected to the Microchip's LAN7800 USB 3.0 to GbE (Gigabit Ethernet Bridge) (left-side port), will not provide the full speed of 1Gbps, but rather a much reduced speed. This is because the lan78xx driver in the raspberry core is not up to date.

Building the driver fails with the following error:
```
root@seed:/home/pi/git/seeed-linux-dtoverlays# ./scripts/cm4_lan7800.sh
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
Installed: /usr/src/linux-headers-5.15.32-v8+
make: Entering directory '/usr/src/linux-headers-5.15.32-v8+'
  CC [M]  /home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan78xx.o
/home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c: In function 'lan78xx_skb_return':
/home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:3511:20: error: 'GRO_DROP' undeclared (first use in this function)
 3511 |  if (gro_result == GRO_DROP)
      |                    ^~~~~~~~
/home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:3511:20: note: each undeclared identifier is reported only once for each function it appears in
make[1]: *** [scripts/Makefile.build:277: /home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan78xx.o] Error 1
make: *** [Makefile:1868: /home/pi/git/seeed-linux-dtoverlays/modules/lan7800] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.32-v8+'
Build failed: lan7800
```

**To Reproduce**
On Raspberry Pi OS lite (64 bit):
```
> git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays.git
> cd seeed-linux-dtoverlays/
> sudo ./scripts/cm4_lan7800.sh
```

**Additional context**
```
root@seed:/home/pi/git/seeed-linux-dtoverlays# cat /etc/debian_version 
11.4
root@seed:/home/pi/git/seeed-linux-dtoverlays# uname -a 
Linux seed 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

clranalytics, 2022-09-11:
I installed the latest 32bit and 64bit lite version of Pi OS to Dual Gigabit Ethernet Carrier Board. I got the same error for both versions. It may indicate this fix for LAN7800 doesn't work under the latest Bullseye version of Pi OS. It does work for the Buster version of the Pi OS.

Paraphraser, 2022-09-13:
If you'd like to test the fix I've proposed ahead of it being reviewed and (hopefully) approved by the site's maintainer:

```
$ sudo apt install raspberrypi-kernel-headers
$ cd
$ git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays.git
$ cd seeed-linux-dtoverlays/
$ git checkout master
$ git pull origin master
$ git fetch origin pull/52/head:PR52
$ git checkout PR52
$ sudo ./scripts/cm4_lan7800.sh
$ sudo reboot
```

I'm just starting out with the Seeed Mini Router. I don't need OpenWRT so I installed 2022-04-04-raspios-bullseye-arm64.img.xz and did a fair number of things including several `apt update ; apt upgrade`, all before realising that the performance of the eth1 interface was well below what I expected. Eventually I found my way to [ethernet-ports-configuration](https://wiki.seeedstudio.com/Dual-Gigabit-Ethernet-Carrier-Board-for-Raspberry-Pi-CM4/#ethernet-ports-configuration).

The first problem I struck was that the kernel headers installed in `/usr/src` had a different version number from `uname -r` so nothing worked. I tried purging and re-installing. No change. Then I tried:

```
$ sudo apt update && sudo apt install --reinstall raspberrypi-bootloader raspberrypi-kernel
```

Still didn't fix the problem. I switched focus to trying to build on a separate RPi4 starting with `2022-09-06-raspios-bullseye-arm64.img.xz`. That got matching headers but hit the error with `GRO_DROP` which became the basis of #50. But, otherwise, it proved that it *should* work on a clean install.

I went back to the Seeed Mini Router with every intention of starting over but I decided to give it one more try. This time the headers matched, the driver built, installed, and everything is working as it should.

I have no idea whether something changed overnight that fixed the problem of the mismatched headers, or if it was just a matter of persistence.

* 4GB Raspberry Pi Compute Module 4 Rev 1.0 running Debian GNU/Linux 11 (bullseye) as full 64-bit OS

	```
	$ uname -a
	Linux sys-rtr 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux
	
	$ ls /usr/src
	linux-headers-5.15.61-v8+  sense-hat
	
	$ dmesg | grep lan
	[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=E4:5F:01:6D:F2:29 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=7585afc3-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cgroup_memory=1 cgroup_enable=memory initcall_blacklist=lan78xx_driver_init
	[    0.000000] blacklisting initcall lan78xx_driver_init
	[    1.281002] initcall lan78xx_driver_init blacklisted
	[    2.979252] lan7800: loading out-of-tree module taints kernel.
	[    2.987963] lan78xx.napi20201111 2-3:1.0 (unnamed net_device) (uninitialized): USB_CFG0 0x60768164
	[    3.228303] usbcore: registered new interface driver lan78xx.napi20201111
	```

closed 2022-09-14 by bigbearishappy

Paraphraser, 2022-09-14:
Just as a closing note for anyone who applied PR50 to their local clone of this repo, now that #50 has been approved you can "undo" the test setup like this:

```
$ cd ~/seeed-linux-dtoverlays/
$ git checkout master
$ git branch -D PR52
$ git pull origin master
```

After that the documentation should work "as is".

xenOs76, 2022-10-02:
Hi @Paraphraser 
sorry for the late reply.  
This is just to confirm that building the driver works for me:
```
❯ uname -a 
Linux seed 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

❯ cat /etc/debian_version 
11.5

pi@seed  in …/seeed-linux-dtoverlays on  master        
❯ sudo ./scripts/cm4_lan7800.sh
Installed: /usr/src/linux-headers-5.15.61-v8+
make: Entering directory '/usr/src/linux-headers-5.15.61-v8+'
  MODPOST /home/pi/git/seeed-linux-dtoverlays/modules/lan7800/Module.symvers
  CC [M]  /home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan7800.mod.o
  LD [M]  /home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan7800.ko
make: Leaving directory '/usr/src/linux-headers-5.15.61-v8+'
'/home/pi/git/seeed-linux-dtoverlays/modules/lan7800/lan7800.ko' -> '/lib/modules/5.15.61-v8+/extra/seeed/lan7800.ko'
------------------------------------------------------
Please reboot your device to apply all settings
Enjoy!
------------------------------------------------------
```
and, after a reboot:
```
❯ modinfo lan7800
filename:       /lib/modules/5.15.61-v8+/extra/seeed/lan7800.ko
license:        GPL
description:    LAN78XX USB 3.0 Gigabit Ethernet Devices
author:         WOOJUNG HUH <woojung.huh@microchip.com>
srcversion:     DCAFB53D1C71A96710E4C59
alias:          usb:v0424p7801d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0424p7850d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0424p7800d*dc*dsc*dp*ic*isc*ip*in*
depends:        
name:           lan7800
vermagic:       5.15.61-v8+ SMP preempt mod_unload modversions aarch64
parm:           msg_level:Override default message level (int)

❯ ls -la  /sys/class/net/eth1/device/driver
lrwxrwxrwx    0 root  2 Oct 09:01 2-3:1.0 -> ../../../../devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-3/2-3:1.0
.-w------- 4.1k root  2 Oct 09:01 bind
lrwxrwxrwx    0 root  2 Oct 09:01 module -> ../../../../module/lan7800
.rw-r--r-- 4.1k root  2 Oct 09:01 new_id
.rw-r--r-- 4.1k root  2 Oct 09:01 remove_id
.-w------- 4.1k root  2 Oct 08:52 uevent
.-w------- 4.1k root  2 Oct 09:01 unbind

❯ iperf3 -s 
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.1.101, port 33776
[  5] local 192.168.1.9 port 5201 connected to 192.168.1.101 port 33780
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   110 MBytes   921 Mbits/sec                  
[  5]   1.00-2.00   sec   112 MBytes   941 Mbits/sec                  
[  5]   2.00-3.00   sec   111 MBytes   930 Mbits/sec                  
[  5]   3.00-4.00   sec   112 MBytes   941 Mbits/sec                  
[  5]   4.00-5.00   sec   111 MBytes   929 Mbits/sec                  
[  5]   5.00-6.00   sec   112 MBytes   941 Mbits/sec                  
[  5]   6.00-7.00   sec   111 MBytes   927 Mbits/sec                  
[  5]   7.00-8.00   sec   112 MBytes   941 Mbits/sec                  
[  5]   8.00-9.00   sec   111 MBytes   930 Mbits/sec                  
[  5]   9.00-10.00  sec   112 MBytes   941 Mbits/sec                  
[  5]  10.00-10.00  sec   267 KBytes   844 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  1.09 GBytes   934 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
```
Thanks

PR #52 (2022-09-14): 20220913 GRO-DROP-fix - master branch
As reported in #50, trying to run `./scripts/cm4_lan7800.sh` results in a build failure because `GRO_DROP` is not declared.

Searching https://linux-commits-search.typesense.org for `GRO_DROP` reveals that it has been removed:

   https://github.com/torvalds/linux/commit/38f7b449256490c312e4e83101075201fb5f87d1

Because `GRO_DROP` can now never be returned, the "GRO packet dropped" debug message can never be produced. The simplest fix is to remove the check and implicitly ignore the content of `gro_result`.

After making that change:

```
$ sudo ./scripts/ [... cut]
commit 252ba5e3 (2022-09-14): 20220913 GRO-DROP-fix - master branch

As reported in #50, trying to run `./scripts/cm4_lan7800.sh` results in
a build failure because `GRO_DROP` is not declared.

Searching https://linux-commits-search.typesense.org for `GRO_DROP`
reveals that it has been removed:

   https://github.com/torvalds/linux/commit/38f7b449256490c312e4e83101075201fb5f87d1

Because `GRO_DROP` can now never be returned, the "GRO packet dropped"
debug message can never be produced. The simplest fix is to remove the
check and implicitly ignore the content of `gro_result`.

After making that change:

```
$ sudo ./scripts [... cut]