Seeed-Studio/seeed-linux-dtoverlays #53: Vertical flickering occurs during video playback

moonlight-aska, 2022-09-18:
**Describe the bug**
When reTerminal is used in a horizontal style and a fast-moving video is played by a video player, a line of image misalignment in the vertical direction is observed in the view.

**To Reproduce**
Steps to reproduce the behavior:
1. Use reTerminal in horizontal style.
2. For example, with the VLC Player app, you can play a video.
3. A line of vertical misalignment occurs in the image in the window.

**Expected behavior**
Normal video playback without vertical misalignment.

**Screenshots**
In the screenshots, the vertical line deviations could not be recordde.

**reTerminal (please complete the following information):**
 - OS: RaspberryPi OS(64bit) 2022-09-06
 - Video Player: VLC Media Player 3.0.17.4

**Additional context**
Similarly, the same symptoms occurred when using OpenCV and displaying with cv2.imshow().

bigbearishappy, 2022-09-22:
Sorry to hear this. After I check it. I find that it's a problem from the hardware design. We don't connect the TE pin from LCD to controller.So the controller can't do a good job on LCD's frame sync.

moonlight-aska, 2022-09-22:
Thank you for investigating the problem.
I understand. It's not a software bug.
We hope that the circuit pattern will be corrected as soon as possible.

By the way, if it can be easily modified with a jumper, please provide more detailed information (either a schematic or a picture).

closed 2023-08-23 by bigbearishappy