Seeed-Studio/seeed-linux-dtoverlays #54: Screen rotates automatically despite having it set to "right" orientation on reTerminal

jeremyherbert, 2022-09-30:
**Describe the bug**
Whenever I physically rotate the reTerminal, it rotates the display to match. But I don't want it to autorotate, and I can't find any way to disable it.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Raspbian 32 bit
2. Run the reTerminal setup script from this repository
3. Configure the display as per https://wiki.seeedstudio.com/reTerminal-FAQ/#q11-the-screen-orientation-is-incorrect-after-installing-raspberry-pi-os-bullseye
4. Reboot
5. Rotate the reTerminal and observe rotation

**Expected behavior**
The display should not rotate.

bigbearishappy, 2022-10-09:
The easiest way to achieve this is to remove the driver of accelerometer(rmmod lis331dlh_i2c).
And if you want to make it work after reboot, you can just remove the driver(lis331dlh-i2c.ko) for accelerometer in /lib/modules/5.15.32-v7l+/updates, then run depmod -a.

jeremyherbert, 2022-10-18:
when I remove this module, I still need to update the screen orientation as per that seeedstudio faq. is that expected?

bigbearishappy, 2022-10-31:
If the screen orientation is not working as you excepted, you should do this.

closed 2023-08-23 by bigbearishappy