Seeed-Studio/seeed-linux-dtoverlays #55: Problems installing on Raspbian 64-bit

maziarzamani, 2022-10-02:
**Describe the bug**
When reTerminal Drivers are installed and system is booted, is seems like Raspbian hits an infinite reboot condition. After 5-10 minutes it finally boots up and the system appears on the LCD screen. I am not sure if this I an LCD screen problem or something else, but it is not behaving as it should. When it boots, the touch screen is not working correctly either.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the guide here https://wiki.seeedstudio.com/reTerminal/2
2. Replace Raspbian OS32-bit with 64-bit. 
3. Flash.

**Expected behavior**
That the system boots properly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Raspbian 64-bit

**Additional context**
Add any other context about the problem here.

lakshanthad, 2022-10-08:
Hello @maziarzamani,

Could you please open `/boot/config.txt` file, delete the line **tp_rotate=1** and try again?

bigbearishappy, 2022-10-09:
Hello, where did you get the system image, raspberrypi offcial or seeed wiki?

maziarzamani, 2022-10-10:
> Hello @maziarzamani,
> 
> Could you please open `/boot/config.txt` file, delete the line **tp_rotate=1** and try again?

Hi @lakshanthad. Thanks for getting back to me! I just tried this workaround, and it fixed the touch display issue! Any idea why this is the case on 64-bit?

maziarzamani, 2022-10-10:
> Hello, where did you get the system image, raspberrypi offcial or seeed wiki?

Hi @bigbearishappy. I did not find any 64-bit image on the seeed wiki, so I used the official in the Raspberry Pi flasher tool.

maziarzamani, 2022-10-17:
> Hello @maziarzamani,
> 
> Could you please open `/boot/config.txt` file, delete the line **tp_rotate=1** and try again?

This fix solved the issue.

closed 2022-10-17 by maziarzamani