Seeed-Studio/seeed-linux-dtoverlays #58: reTerminal LCD does not work with RPi Kernel 6.1

JeremiahGillis, 2023-03-19:
**Describe the bug**
My reTerminal screen stopped worked after my Kernel was updated via RPi OS GUI interface. I tried download and run the reTerminal overlay script, but it aborts.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash reTerminal with latest RPi OS Bullseye (Kernel 5.15) using RPi Imager
2. Update through the GUI interface, or 
1, `sudo apt update`
2, `sudo apt full-upgrade`
3, `sudo apt install raspberrypi-kernel-headers`, `sudo reboot`
4. git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
5. cd seeed-linux-dtoverlays
6. sudo ./scripts/reTerminal.sh
7. Reboot

**Expected behavior**
LCD functioning, LCD worked with Kernal 5.15.x

**Debug Output**
The output of "dmesg | grep mipi" is as follows:
```
[    0.072037] platform fe700000.dsi: Fixed dependency cycle(s) with /soc/i2c@7e804000/mipi_dsi@45
```

LCD driver shows as not loaded at 0x45:
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- 19 -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- 29 -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- --
40: -- -- -- -- -- 45 -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

 ./scripts/reTerminal.sh output:
```
### Install required tool packages
Hit:1 http://deb.debian.org/debian bullseye InRelease
Hit:2 http://security.debian.org/debian-security bullseye-security InRelease
Hit:3 http://deb.debian.org/debian bullseye-updates InRelease
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (2.8.4-3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

### Will compile with the latest kernel...

### Uninstall previous dkms module

------------------------------
Deleting module version: 0.1
completely from the DKMS tree.
------------------------------
Done.

------------------------------
Deleting module version: 0.1
completely from the DKMS tree.
------------------------------
Done.

------------------------------
Deleting module version: 0.1
completely from the DKMS tree.
------------------------------
Done.

------------------------------
Deleting module version: 0.1
completely from the DKMS tree.
------------------------------
Done.
removed '/boot/overlays/reTerminal.dtbo'
removed '/boot/overlays/reTerminal-bridge.dtbo'
removed 'overlays/rpi/.reTerminal-bridge-overlay.dtbo.d.dtc.tmp'
removed 'overlays/rpi/.reTerminal-bridge-overlay.dtbo.d.pre.tmp'
removed 'overlays/rpi/.reTerminal-bridge-overlay.dtbo.dts.tmp'
removed 'overlays/rpi/.reTerminal-overlay.dtbo.d.dtc.tmp'
removed 'overlays/rpi/.reTerminal-overlay.dtbo.d.pre.tmp'
removed 'overlays/rpi/.reTerminal-overlay.dtbo.dts.tmp'
removed 'overlays/rpi/.reTerminal-bridge-overlay.dtbo.cmd'
removed 'overlays/rpi/.reTerminal-overlay.dtbo.cmd'
removed 'overlays/rpi/reTerminal-bridge-overlay.dtbo'
removed 'overlays/rpi/reTerminal-overlay.dtbo'

### Install required kernel package
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230317-1).
raspberrypi-kernel-headers is already the newest version (1:1.20230317-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
W: --force-yes is deprecated, use one of the options starting with --allow instead.
 !!! Your kernel version is

6.1.19-v8+
     Couldn't find *** corresponding *** kernel headers with apt-get.
     This may happen if you ran 'rpi-update'.
 Choose  *** y *** to install kernel-headers to version 6.1.19-v8+ and continue.
 Choose  *** N *** to exit without this driver support, by default.
Would you like to proceed? (y/N)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20230317-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

Creating symlink /var/lib/dkms/mipi_dsi/0.1/source ->
                 /usr/src/mipi_dsi-0.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=6.1.19-v8+ all KVERSION=6.1.19-v8+....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.19-v8+ (aarch64)
Consult /var/lib/dkms/mipi_dsi/0.1/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
```

Please advise when this can be addressed. I need to send my reTerminal to another county this week as it is part of a test fixture. Anyone connecting this to the internet will be prompted to update the OS through the GUI, which will break screen functionality.

bigbearishappy, 2023-03-20:
You can take a look into this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/76a37d4126da04ad38acc2a26fc12b368d32e64f).

nobodywantstoknow, 2023-03-20:
Hi @bigbearishappy, I tested the latest branch for the commit that you mentioned. The Pi was able to run the reTerminal.sh, but there is an error on detecting the card vc4hdmi0 and vc4hdmi1 as shown in the image below.

[image]

After reboot, I can't seem to ssh into the pi and the screen still showing blank screen.

JeremiahGillis, 2023-03-20:
@bigbearishappy Thanks for the link. This fixed my issues. I was using SSH with no display so I didn't get any HDMI issues like @nobodywantstoknow mentioned. 

I eliminated the other issues such as no sound card and buzzer / LEDs with `#dtoverlay=reTerminal-bridge` since I don't have one plugged in. The only remaining error I see in DMESG regarding your overlay is:

```
[    6.477149] mcp230xx 1-0038: error -EREMOTEIO: can't write IOCON 56
[    6.477389] mcp230xx: probe of 1-0038 failed with error -121
```

You should also note that my reTerminals have always required `dtoverlay=reTerminal,tp_rotate=0`. Otherwise the TP input is rotated incorrectly with your stock `dtoverlay=reTerminal,tp_rotate=1` setting. My reTerminals show 05/09/22 on the box UPCs. All documentation states `tp_rotate=1` which is wrong for mine.

hugokernel, 2023-03-20:
I just encountered the problem with a core 6 and the problem with the Touch Screen, thanks for sharing!

nobodywantstoknow, 2023-03-21:
I think the problem happened when I'm using the OS using raspberry pi imager instead of the image from seeedstudio. Reflashing the OS downloaded from the seeedstudio website did solve the problem.

twilfredo, 2023-03-26:
On that branch I'm seeing commits for a `reterminal Plus`, is that a new device?

twilfredo, 2023-03-26:
Also please make sure the  documentation is updated to inform that even if flash the seeed image, after you do  the following sequence:
```
1, sudo apt update
2, sudo apt full-upgrade
3, sudo apt install raspberrypi-kernel-headers, sudo reboot
```
The display will be blank, you will need to checkout this specific [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/76a37d4126da04ad38acc2a26fc12b368d32e64f). and reinstall the drivers for things to work.

twilfredo, 2023-03-26:
Also please make sure the  documentation is updated to inform that even if flash the seeed image, after you do  the following sequence:
```
1, sudo apt update
2, sudo apt full-upgrade
3, sudo apt install raspberrypi-kernel-headers, sudo reboot
```
The display will be blank, you will need to checkout this specific [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/76a37d4126da04ad38acc2a26fc12b368d32e64f). and reinstall the drivers for things to work. 

You also need to:

```
sudo vi /boot/config.txt
#dtoverlay=reTerminal-bridge
sudo reboot
```
Not sure why this isn't disabled by default?

bigbearishappy, 2023-05-08:
Now, we have merge the fixing code of kernel version issue to the master branch.You can get the detail in this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/9c293653e6b356244ec101610aeb5cdc22ee2f4e)
You should be able to install the driver to the latest kernel(6.1) with the following command:
```
./scripts/reTerminal.sh
```

And the problem with buzzer and leds on reterminal. you can take look into the [FAQS](https://wiki.seeedstudio.com/reTerminal-FAQ#q12-leds-and-buzzer-do-not-work-after-installing-reterminal-drivers) in seeed's wiki.

One more thing. As @twilfredo mentioned above. Branch plus is for our new device reterminal DM. You can get more information of it in this [link](https://wiki.seeedstudio.com/reterminal-dm)

If you have any further question. You can reopen this issue or create a new one.

closed 2023-05-08 by bigbearishappy