Seeed-Studio/seeed-linux-dtoverlays #60: reTerminal doesn't work with CM4 8/32GB

windoze, 2023-04-29:
**reTerminal doesn't work with CM4 8/32GB**
I replaced the 4/16GB CM4 module came with reTerminal with CM4 8/32GB, which I confirmed to be working properly on the official CM4 expansion board.
The installation of Raspberry Pi OS was successfully, and SSH over wifi worked without problem.
After installing the dtoverlay, reTerminal hang on `sudo reboot`, and cannot boot after the power cycle.

**To Reproduce**
Steps to reproduce the behavior:
1. Replace the CM4 module with CM4 8/32GB
2. Install the Raspberry Pi OS by following the official wiki.
3. Install dtoverlay by following the official wiki.
4. The machine hangs.

**Expected behavior**
The reTerminal boot and run without issue.

I searched the website and didn't find any info about the reTerminal/CM4 compatibility, is there anything special in the CM4 that comes with reTerminal?

windoze, 2023-05-08:
I cannot find the line `dtoverlay=reTerminal-bridge` in `/boot/config.txt` after running `sudo ./scripts/reTerminal.sh`,  I added `arm_64bit=0` and now it seems to be working fine.

Thanks for your help.

bigbearishappy, 2023-05-09:
OK, it's a issue with the official raspbian OS building.
Closing this issue~

closed 2023-05-09 by bigbearishappy