Seeed-Studio/seeed-linux-dtoverlays #61: [reTerminal] fatal error: asm/types.h: No such file or directory with RasPi OS 32bit 2023-05-03

matsujirushi, 2023-05-04:
A compilation error occurs when running reTerminal.sh.

**Environment:**
* Device: reTerminal 05/09/22
* OS image: Raspberry Pi OS (32-bit) 2023-05-03

**Procedure:**
1. Flash OS image `Raspberry Pi OS (32-bit) 2023-05-03` to CM4 eMMC.
2. Install the reTerminal driver.

```
$ git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
$ cd seeed-linux-dtoverlays
$ sudo ./scripts/reTerminal.sh
```

**Error message:**

```
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
```

<details>
<summary>Full messages</summary>

```
### Install required tool packages
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  dctrl-tools
Suggested packages:
  debtags menu
Recommended packages:
  linux-headers-686-pae | linux-headers-amd64 | linux-headers-generic | linux-headers
The following NEW packages will be installed:
  dctrl-tools dkms
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 172 kB of archives.
After this operation, 673 kB of additional disk space will be used.
Get:1 http://ftp.udx.icscoe.jp/Linux/raspbian/raspbian bullseye/main armhf dctrl-tools armhf 2.24-3 [94.2 kB]
Get:2 http://ftp.udx.icscoe.jp/Linux/raspbian/raspbian bullseye/main armhf dkms all 2.8.4-3 [78.2 kB]
Fetched 172 kB in 2s (114 kB/s)
Selecting previously unselected package dctrl-tools.
(Reading database ... 106529 files and directories currently installed.)
Preparing to unpack .../dctrl-tools_2.24-3_armhf.deb ...
Unpacking dctrl-tools (2.24-3) ...
Selecting previously unselected package dkms.
Preparing to unpack .../archives/dkms_2.8.4-3_all.deb ...
Unpacking dkms (2.8.4-3) ...
Setting up dctrl-tools (2.24-3) ...
Setting up dkms (2.8.4-3) ...
Processing triggers for man-db (2.9.4-2) ...

### Will compile with the latest kernel...

### Uninstall previous dkms module

### Install required kernel package
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230405-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
The following NEW packages will be installed:
  raspberrypi-kernel-headers
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 30.0 MB of archives.
After this operation, 193 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian bullseye/main armhf raspberrypi-kernel-headers armhf 1:1.20230405-1 [30.0 MB]
Fetched 30.0 MB in 5s (6,140 kB/s)
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 106628 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1%3a1.20230405-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1:1.20230405-1) ...
Setting up raspberrypi-kernel-headers (1:1.20230405-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 6.1.21+
dkms: running auto installation service for kernel 6.1.21+:.
run-parts: executing /etc/kernel/header_postinst.d/dkms 6.1.21-v7+
dkms: running auto installation service for kernel 6.1.21-v7+:.
run-parts: executing /etc/kernel/header_postinst.d/dkms 6.1.21-v7l+
dkms: running auto installation service for kernel 6.1.21-v7l+:.
run-parts: executing /etc/kernel/header_postinst.d/dkms 6.1.21-v8+
dkms: running auto installation service for kernel 6.1.21-v8+:.
W: --force-yes is deprecated, use one of the options starting with --allow instead.
KBUILD: /lib/modules/6.1.21-v7l+/build

Creating symlink /var/lib/dkms/mipi_dsi/0.1/source ->
                 /usr/src/mipi_dsi-0.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=6.1.21-v7l+ all KVERSION=6.1.21-v7l+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.21-v7l+ (aarch64)
Consult /var/lib/dkms/mipi_dsi/0.1/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
DKMS make.log for mipi_dsi-0.1 for kernel 6.1.21-v7l+ (aarch64)
Thu  4 May 17:20:21 JST 2023
make -C /lib/modules/6.1.21-v7l+/build M=/var/lib/dkms/mipi_dsi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v7l+'
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/touch_panel.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o
In file included from ./include/linux/types.h:6,
                 from ./include/linux/limits.h:6,
                 from ./include/linux/kernel.h:16,
                 from ./include/linux/interrupt.h:6,
                 from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi.h:15,
                 from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:11:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from ./include/linux/types.h:6,
                 from ./include/linux/limits.h:6,
                 from ./include/linux/kernel.h:16,
                 from ./include/linux/interrupt.h:6,
                 from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi.h:15,
                 from /var/lib/dkms/mipi_dsi/0.1/build/touch_panel.c:1:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/mipi_dsi/0.1/build/touch_panel.o] Error 1
In file included from ./include/linux/types.h:6,
                 from ./include/linux/limits.h:6,
                 from ./include/linux/kernel.h:16,
                 from ./include/linux/interrupt.h:6,
                 from /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi.h:15,
                 from /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.c:8:
./include/uapi/linux/types.h:5:10: fatal error: asm/types.h: No such file or directory
    5 | #include <asm/types.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o] Error 1
make[1]: *** [Makefile:2012: /var/lib/dkms/mipi_dsi/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v7l+'
make: *** [Makefile:15: all] Error 2
```
</details>

Pillar1989, 2023-05-04:
@matsujirushi  Thanks for your feedback, we'll deal with these issues as soon as possible, it would be nice to have PR available. @bigbearishappy , please close all reTerminal driver issues on the latest raspbian os as soon as possible.
#60

bigbearishappy, 2023-05-08:
> I think that the fact that the 64bit kernel is running has an effect.
> 
> > https://www.raspberrypi.com/documentation/computers/config_txt.html#arm_64bit
> > [raspberrypi/documentation@6747659](https://github.com/raspberrypi/documentation/commit/6747659b0e5e0fd8b967b994928c3a9c0b7634b2)
> > Defaults to 1 on Pi 4s (Pi 4B, Pi 400, CM4 and CM4S), and 0 on all other platforms. However, if the name given in an explicit kernel option matches one of the known kernels then arm_64bit will be set accordingly.

You are right. This is a issue of option `arm_64bit` in `/boot/config.txt`.
It seems the raspbian OS official developer forget to add `arm_64bit=0` to the 32bit image's config.txt

bigbearishappy, 2023-05-08:
You can fix this problem by adding `arm_64bit=0` to `/boot/config.txt`.

closed 2023-05-08 by bigbearishappy