Seeed-Studio/seeed-linux-dtoverlays #62: [reTerminal] Show banner and freeze with RasPi OS 32bit 2023-05-03

matsujirushi, 2023-05-04:
If reTerminal run `dtoverlay=reTerminal-bridge` without commenting it, it freezes when the banner is displayed.

**Environment:**

* Device: reTerminal 05/09/22
* OS image: Raspberry Pi OS (32-bit) 2023-05-03

**Procedure:**

1. Flash OS image Raspberry Pi OS (32-bit) 2023-05-03 to CM4 eMMC.
2. Change kernel architecture. (Add `arm_64bit=0` in **/boot/config.txt** and reboot.)
3. Install the reTerminal driver.
```
$ git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
$ cd seeed-linux-dtoverlays
$ sudo ./scripts/reTerminal.sh
$ sudo reboot
```

**Serial console:**

```
[    8.287084] 8<--- cut here ---
[    8.290199] Unable to handle kernel paging request at virtual address 000f4240
[    8.297538] [000f4240] *pgd=030d0003, *pmd=00000000
[    8.297556] Internal error: Oops: 206 [#1] SMP ARM
[    8.297562] Modules linked in: gpio_pca953x(+) bq24179_charger(O+) industrialio pinctrl_mcp23s08_i2c pinctrl_mcp23s08 gpio_pcf857x spidev rtc_pcf8563 snd_soc_wm8960 regmap_i2c 8021q garp stp llc brcmfmac vc4 brcmutil cfg80211 raspberrypi_hwmon snd_soc_hdmi_codec drm_display_helper cec rfkill snd_soc_simple_card drm_dma_helper snd_soc_simple_card_utils bcm2835_codec(C) rpivid_hevc(C) v3d gpu_sched bcm2835_v4l2(C) drm_kms_helper drm_shmem_helper v4l2_mem2mem bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc snd_soc_bcm2835_i2s snd_soc_core i2c_brcmstb videobuf2_memops videobuf2_v4l2 snd_compress videobuf2_common snd_pcm_dmaengine snd_bcm2835(C) videodev i2c_bcm2835 spi_bcm2835 snd_pcm vc_sm_cma(C) snd_timer snd mc syscopyarea sysfillrect sysimgblt fb_sys_fops gpio_keys nvmem_rmem uio_pdrv_genirq uio drm i2c_dev fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    8.370386] CAN device driver interface
[    8.388596] CPU: 3 PID: 139 Comm: systemd-journal Tainted: G         C O       6.1.21-v7l+ #1642
[    8.388606] Hardware name: BCM2711
[    8.388609] PC is at __of_find_property+0x3c/0x68
[    8.409623] LR is at of_find_property+0x3c/0x54
[    8.409636] pc : [<c09cb950>]    lr : [<c09cb9b8>]    psr: 20000093
[    8.409640] sp : f1075de0  ip : 00000000  fp : 00000001
[    8.409643] r10: c4000f68  r9 : 00400cc0  r8 : 00000000
[    8.409647] r7 : c0e8aa58  r6 : 00000000  r5 : c0e8aa58  r4 : 000f4240
[    8.409650] r3 : 0000b6d2  r2 : 00000000  r1 : c0e8aa58  r0 : efee6b0c
```

**Links:**

* [Q12: LEDs and Buzzer do not work after installing reTerminal drivers](https://wiki.seeedstudio.com/reTerminal-FAQ/#q12-leds-and-buzzer-do-not-work-after-installing-reterminal-drivers)

matsujirushi, 2023-05-09:
Hi @bigbearishappy ,
Even when I connected a reTerminal E10-1, it did not start.

**Serial console:**

```
[    8.093429] 8<--- cut here ---
[    8.096535] Unable to handle kernel paging request at virtual address 000f4240
[    8.103871] [000f4240] *pgd=030da003, *pmd=00000000
[    8.103886] Internal error: Oops: 206 [#1] SMP ARM
[    8.113695] Modules linked in: als_ltr30x(O+) industrialio_triggered_buffer kfifo_buf industrialio bq24179_charger(O+) pinctrl_mcp23s08_i2c snd_soc_wm8960 pinctrl_mcp23s08 regmap_i2c spidev 8021q garp stp llc brcmfmac brcmutil cfg80211 v3d gpu_sched rfkill drm_shmem_helper snd_soc_simple_card snd_soc_simple_card_utils vc4 bcm2835_v4l2(C) bcm2835_isp(C) raspberrypi_hwmon videobuf2_vmalloc bcm2835_codec(C) rpivid_hevc(C) snd_soc_hdmi_codec drm_display_helper bcm2835_mmal_vchiq(C) snd_soc_bcm2835_i2s cec v4l2_mem2mem videobuf2_dma_contig drm_dma_helper videobuf2_memops videobuf2_v4l2 drm_kms_helper videobuf2_common videodev snd_soc_core i2c_brcmstb i2c_bcm2835 spi_bcm2835 vc_sm_cma(C) snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm snd_timer snd mc syscopyarea sysfillrect sysimgblt fb_sys_fops gpio_keys nvmem_rmem uio_pdrv_genirq uio drm i2c_dev fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    8.196129] CPU: 1 PID: 138 Comm: systemd-journal Tainted: G         C O       6.1.21-v7l+ #1642
[    8.205041] Hardware name: BCM2711
[    8.208486] PC is at __of_find_property+0x3c/0x68
[    8.213262] LR is at of_find_property+0x3c/0x54
[    8.217856] pc : [<c09cb950>]    lr : [<c09cb9b8>]    psr: 20000093
[    8.224210] sp : f1069de0  ip : 00000000  fp : 00000001
[    8.229505] r10: c3b31478  r9 : 00400cc0  r8 : 00000000
[    8.234799] r7 : c0e8aa58  r6 : 00000000  r5 : c0e8aa58  r4 : 000f4240
[    8.241417] r3 : 0000a168  r2 : 00000000  r1 : c0e8aa58  r0 : efee6b0c
```

<details>
<summary>Serial console - full</summary>

```
[    8.093429] 8<--- cut here ---
[    8.096535] Unable to handle kernel paging request at virtual address 000f4240
[    8.103871] [000f4240] *pgd=030da003, *pmd=00000000
[    8.103886] Internal error: Oops: 206 [#1] SMP ARM
[    8.113695] Modules linked in: als_ltr30x(O+) industrialio_triggered_buffer kfifo_buf industrialio bq24179_charger(O+) pinctrl_mcp23s08_i2c snd_soc_wm8960 pinctrl_mcp23s08 regmap_i2c spidev 8021q garp stp llc brcmfmac brcmutil cfg80211 v3d gpu_sched rfkill drm_shmem_helper snd_soc_simple_card snd_soc_simple_card_utils vc4 bcm2835_v4l2(C) bcm2835_isp(C) raspberrypi_hwmon videobuf2_vmalloc bcm2835_codec(C) rpivid_hevc(C) snd_soc_hdmi_codec drm_display_helper bcm2835_mmal_vchiq(C) snd_soc_bcm2835_i2s cec v4l2_mem2mem videobuf2_dma_contig drm_dma_helper videobuf2_memops videobuf2_v4l2 drm_kms_helper videobuf2_common videodev snd_soc_core i2c_brcmstb i2c_bcm2835 spi_bcm2835 vc_sm_cma(C) snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm snd_timer snd mc syscopyarea sysfillrect sysimgblt fb_sys_fops gpio_keys nvmem_rmem uio_pdrv_genirq uio drm i2c_dev fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[    8.196129] CPU: 1 PID: 138 Comm: systemd-journal Tainted: G         C O       6.1.21-v7l+ #1642
[    8.205041] Hardware name: BCM2711
[    8.208486] PC is at __of_find_property+0x3c/0x68
[    8.213262] LR is at of_find_property+0x3c/0x54
[    8.217856] pc : [<c09cb950>]    lr : [<c09cb9b8>]    psr: 20000093
[    8.224210] sp : f1069de0  ip : 00000000  fp : 00000001
[    8.229505] r10: c3b31478  r9 : 00400cc0  r8 : 00000000
[    8.234799] r7 : c0e8aa58  r6 : 00000000  r5 : c0e8aa58  r4 : 000f4240
[    8.241417] r3 : 0000a168  r2 : 00000000  r1 : c0e8aa58  r0 : efee6b0c
[    8.248033] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[    8.255357] Control: 30c5383d  Table: 031b9940  DAC: 55555555
[    8.261181] Register r0 information: non-slab/vmalloc memory
[    8.266920] Register r1 information: non-slab/vmalloc memory
[    8.272658] Register r2 information: NULL pointer
[    8.277427] Register r3 information: non-paged memory
[    8.282546] Register r4 information: non-paged memory
[    8.287667] Register r5 information: non-slab/vmalloc memory
[    8.293404] Register r6 information: NULL pointer
[    8.298172] Register r7 information: non-slab/vmalloc memory
[    8.303909] Register r8 information: NULL pointer
[    8.308678] Register r9 information: non-paged memory
[    8.313798] Register r10 information: slab vm_area_struct start c3b31460 pointer offset 24
[    8.322189] Register r11 information: non-paged memory
[    8.327397] Register r12 information: NULL pointer
[    8.332253] Process systemd-journal (pid: 138, stack limit = 0x70166787)
[    8.339048] Stack: (0xf1069de0 to 0xf106a000)
[    8.343463] 9de0: c13eff20 efee6b0c 60000013 c09cb9b8 c1b4b820 c1b65000 c1b65000 c1b4b820
[    8.351756] 9e00: c2cf8000 c09cb9e0 c1b4b820 c09ced8c c1b4b820 c1b65000 c1205048 c1b4b820
[    8.360049] 9e20: c2cf8000 00400cc0 c3b31478 c0843a50 00004004 c0447490 c1b89e00 c1990680
[    8.368342] 9e40: c1b4b820 c2cf8000 c1b4b820 2ee21c51 c1990680 c1b4b820 c1b65000 c0840e30
[    8.376634] 9e60: c130c7c0 c0840d94 c2cf8000 c1b4b820 c33dad80 00400cc0 c3b31478 c083e6c0
[    8.384927] 9e80: c3b31460 c0c57a20 c2cf8000 c04faaec c3b31460 00001000 c04f95a0 f1069f20
[    8.393219] 9ea0: 7ffff000 c048b658 01c04000 00000000 c3b31488 f1069f08 f473c75f c045e190
[    8.401512] 9ec0: 64599479 01c9c37e 64599479 c1205048 c3f44cc0 c04f95a0 00001000 f1069f78
[... cut ...]
[  218.460204]  rcu_gp_kthread from kthread+0xe4/0x100
[  218.465154]  kthread from ret_from_fork+0x14/0x34
[  218.469926] Exception stack(0xf0851fb0 to 0xf0851ff8)
[  218.475046] 1fa0:                                     00000000 00000000 00000000 00000000
[  218.483340] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  218.491632] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  218.498338] rcu: Stack dump where RCU GP kthread last ran:
[  218.503899] Sending NMI from CPU 3 to CPUs 0:
```

</details>

bigbearishappy, 2024-10-15:
Cause the firmware of 2023-05-03 is too old. I think it's better idea for us to try it with the latest firmware from [here](https://github.com/Seeed-Studio/pi-gen-expand).

I will try the latest one and feedback to you~

bigbearishappy, 2024-10-22:
I tried the latest image I mentioned before and it works fine. 
Please have a try with the latest image~

closed 2024-11-11 by Lesords