Seeed-Studio/seeed-linux-dtoverlays #63: reTerminal DM Ubuntu support

kirwinrMK, 2023-05-04:
It seems the driver install method in `scripts/reTerminal.sh` don't support Ubuntu on the reTerminal DM. 
Is this a possibility to add?

bigbearishappy, 2023-05-08:
We don't have a plan to support ubuntu on reTerminal DM. But we have did relative work on reTerminal.
You can take a look into [this](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/28#issuecomment-939662366).
It may help you to achieve what you want.

closed 2023-05-08 by bigbearishappy

danwalmsley, 2024-09-17:
In case anyone needs it, I have ubuntu 23.10 working on re terminal DM using these mods and --keep-kernel option.

https://github.com/danwalmsley/seeed-linux-dtoverlays/commit/edb961382bbeea31ce343b3d22336ba25734dbb7