Seeed-Studio/seeed-linux-dtoverlays #64: LAN7800 script fails, interface runs at 100Mb/s

monokinetic, 2023-05-04:
Attempting to follow the Seeed wiki page [here](https://wiki.seeedstudio.com/Dual-Gigabit-Ethernet-Carrier-Board-for-Raspberry-Pi-CM4/#ethernet-ports-configuration) on a recent install of Raspberry Pi os:
`Linux raspberrypi 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l
`

fails for me like this:

`dejf@raspberrypi:~ $ cd Documents/seeed-linux-dtoverlays/
dejf@raspberrypi:~/Documents/seeed-linux-dtoverlays $ sudo ./scripts/cm4_lan7800.sh
Installed: /usr/src/linux-headers-6.1.21-v7l+
make: Entering directory '/usr/src/linux-headers-6.1.21-v7l+'
  CC [M]  /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.o
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c: In function 'lan78xx_init_mac_address':
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:1955:26: warning: passing argument 1 of 'ether_addr_copy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 1955 |  ether_addr_copy(dev->net->dev_addr, addr);
      |                  ~~~~~~~~^~~~~~~~~~
In file included from /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:8:
./include/linux/etherdevice.h:295:40: note: expected 'u8 *' {aka 'unsigned char *'} but argument is of type 'const unsigned char *'
  295 | static inline void ether_addr_copy(u8 *dst, const u8 *src)
      |                                    ~~~~^~~
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c: In function 'lan78xx_set_mac_addr':
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:2550:24: warning: passing argument 1 of 'ether_addr_copy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 2550 |  ether_addr_copy(netdev->dev_addr, addr->sa_data);
      |                  ~~~~~~^~~~~~~~~~
In file included from /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:8:
./include/linux/etherdevice.h:295:40: note: expected 'u8 *' {aka 'unsigned char *'} but argument is of type 'const unsigned char *'
  295 | static inline void ether_addr_copy(u8 *dst, const u8 *src)
      |                                    ~~~~^~~
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c: In function 'lan78xx_probe':
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:4312:2: error: implicit declaration of function 'netif_set_gso_max_size'; did you mean 'netif_set_tso_max_size'? [-Werror=implicit-function-declaration]
 4312 |  netif_set_gso_max_size(netdev, LAN78XX_TSO_SIZE(dev));
      |  ^~~~~~~~~~~~~~~~~~~~~~
      |  netif_set_tso_max_size
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:4314:2: error: too many arguments to function 'netif_napi_add'
 4314 |  netif_napi_add(netdev, &dev->napi, lan78xx_poll, LAN78XX_NAPI_WEIGHT);
      |  ^~~~~~~~~~~~~~
In file included from /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:7:
./include/linux/netdevice.h:2571:1: note: declared here
 2571 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,
      | ^~~~~~~~~~~~~~
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:4361:9: error: too many arguments to function 'usb_maxpacket'
 4361 |  maxp = usb_maxpacket(dev->udev, dev->pipe_intr, 0);
      |         ^~~~~~~~~~~~~
In file included from /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:10:
./include/linux/usb.h:1981:19: note: declared here
 1981 | static inline u16 usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~
/home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:4377:19: error: too many arguments to function 'usb_maxpacket'
 4377 |  dev->maxpacket = usb_maxpacket(dev->udev, dev->pipe_out, 1);
      |                   ^~~~~~~~~~~~~
In file included from /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.c:10:
./include/linux/usb.h:1981:19: note: declared here
 1981 | static inline u16 usb_maxpacket(struct usb_device *udev, int pipe)
      |                   ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:250: /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800/lan78xx.o] Error 1
make: *** [Makefile:2012: /home/dejf/Documents/seeed-linux-dtoverlays/modules/lan7800] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.21-v7l+'
Build failed: lan7800`

**To Reproduce**
Steps to reproduce the behavior:
1. Install latest Raspberry Pi OS from the rpi-installer on the supplied CM4
2. Follow the wiki page as linked and try the Ethernet section where the LAN7800 interface needs drivers to reach full speed.
3.  Also note to even get this far the config.txt file required:
`arm64=0`
4. Install the kernel headers.
4. See the dirs and command issued above as taken from Seeed wiki page.
5. The module fails to compile.

**Expected behavior**
To be able to run the LAN7800 interface at 1Gb/s as advertised.

**Additional context**
I'm attempting to debug my LAN7800 based port using OpenWRT, see [issue42](https://github.com/Seeed-Studio/seeed-linux-openwrt/issues/42). See also forum thread [here](https://forum.seeedstudio.com/t/rerouter-eth1-running-at-only-100mb-s/270011/4) where I'm trying to get this hardware to work in any operating system. So far it always fails.

Pillar1989, 2023-05-06:
#50

sjpark608, 2023-06-06:
I am also having the same issue here. The script fails with the same  `make` error.

LynnL4, 2024-10-21:
HI，As far as I know, the 6.x kernel has been upgraded with a lan78xx driver that supports 1000M, by default. 
https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/59

closed 2024-11-04 by Lesords