Seeed-Studio/seeed-linux-dtoverlays #65: [reTerminal] Support for 64bit kernel running on RasPi OS 32bit

matsujirushi, 2023-05-09:
Using RasPi OS 32bit 2023-05-03, Compute Module 4 (i.e. reTerminal) works with 64bit kernel.
(Previous OS images used a 32bit kernel.)

I think future 32-bit OS images may recommend a 64-bit kernel.

What is your plan?

1. 32bit OS image is not supported. 64bit OS image is recommended.
2. Use of 32bit kernel is recommended. (`arm_64bit=0`)
3. seeed-linux-dtoverlays supports 64bit kernel running on RasPi OS 32bit.

**References:**
* #61 
* [raspian 32bit - update && sudo apt upgrade on raspian 32bit the kernel automatically gets updated to arm64bit??](https://forums.raspberrypi.com/viewtopic.php?t=349291)

bigbearishappy, 2023-05-10:
Before we understand the benefit of support 64bit kernel with 32bit raspbian OS. We would recommend to use the 32bit kernel with 32bit raspbian OS.

matsujirushi, 2023-05-11:
I got.
Thanks for reply.

closed 2023-05-11 by matsujirushi