Seeed-Studio/seeed-linux-dtoverlays #67: apt-update causes the screen not to show up anymore after restart

ChinnyxD, 2023-05-25:
**Describe the bug**
I'm not that techsavvy with linux, so i hope anyone here can help me out. After updating packages the reterminal screen after restart shows only a black screen. I also tried `How can I install Ubuntu on reTerminal?`, but unfortunately the `./script2.1.sh` is broken.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash the 2022-06-20-Raspbian-reTerminal.img with Raspberry pi imager
2. Connect with SSH
3. sudo apt-get update & sudo apt-get upgrade
4. sudo reboot
5. Reterminal starts with black screen, SSH still works.

**Expected behavior**
The screen should still work after updating the packages.

**Logs**
Please see the logs:
[Reterminal logs.txt](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/11566791/Reterminal.logs.txt)

bigbearishappy, 2023-05-26:
Can you show me the output of the following two command?
```
getconf LONG_BIT
```
```
uname -a
```

bigbearishappy, 2023-10-07:
There are some ways which maybe help you to fix the problem:
1 if you OS is 32bit, you need to add arm_64bit=0 to config.txt and then reinstall the device driver
2 check if you have already install the linux-kernel-header to you OS
Feel free to reopen this issue if you have any further problem.

closed 2023-10-07 by bigbearishappy