Seeed-Studio/seeed-linux-dtoverlays #69: Compile error Reterminal DM ltr30x

donserdal, 2023-08-04:
Hello,

Running OS:
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

When following the guide on: https://wiki.seeedstudio.com/reterminal-dm-flash-OS/#install-reterminal-drivers-after-flashing-new-raspbian-os

I get the following:
```

~/git/seeed-linux-dtoverlays $ sudo ./scripts/reTerminal.sh --device reTerminal-plus

### Install required tool packages
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (2.8.4-3).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 82 not upgraded.

### Will compile with the latest kernel...

### Uninstall previous dkms module

------------------------------
Deleting module version: 0.1
completely from the DKMS tree.
------------------------------
Done.

### Install required kernel package
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
raspberrypi-kernel is already the newest version (1:1.20230405-1).
raspberrypi-kernel-headers is already the newest version (1:1.20230405-1).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 82 not upgraded.
W: --force-yes is deprecated, use one of the options starting with --allow instead.
KBUILD: /lib/modules/6.1.21-v7l+/build

Creating symlink /var/lib/dkms/ltr30x/0.1/source ->
                 /usr/src/ltr30x-0.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=6.1.21-v7l+ all KVERSION=6.1.21-v7l+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.1.21-v7l+ (aarch64)
Consult /var/lib/dkms/ltr30x/0.1/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
DKMS make.log for ltr30x-0.1 for kernel 6.1.21-v7l+ (aarch64)
Fri  4 Aug 13:32:59 CEST 2023
make -C /lib/modules/6.1.21-v7l+/build M=/var/lib/dkms/ltr30x/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v7l+'
  CC [M]  /var/lib/dkms/ltr30x/0.1/build/ltr30x.o
In file included from ./include/linux/build_bug.h:5,
                 from ./include/linux/container_of.h:5,
                 from ./include/linux/list.h:5,
                 from ./include/linux/module.h:12,
                 from /var/lib/dkms/ltr30x/0.1/build/ltr30x.c:12:
./include/linux/compiler.h:246:10: fatal error: asm/rwonce.h: No such file or directory
  246 | #include <asm/rwonce.h>
      |          ^~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:250: /var/lib/dkms/ltr30x/0.1/build/ltr30x.o] Error 1
make[1]: *** [Makefile:2012: /var/lib/dkms/ltr30x/0.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v7l+'
make: *** [Makefile:13: all] Error 2
```

donserdal, 2023-08-04:
32Bit only, I tried 64Bit and it works now!

bigbearishappy, 2023-08-08:
It seems like same with this [issue](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/61). Please make the change mentioned there and have a try again.

closed 2023-09-01 by donserdal