Seeed-Studio/seeed-linux-dtoverlays #73: reTerminal - My mouse cursor wont correspond to where I tap on the screen

jorgedevs, 2023-09-07:
I have an issue when I use the touchscreen in the reTerminal, wherever I tap or drag my finger on the screen, the mouse cursor pops-in/moves in a totally different location on the screen.

https://github.com/Seeed-Studio/seeed-linux-dtoverlays/assets/3408668/9ef4002d-9e79-434f-abdd-6b0805c57b77

I tried the [Q12: Troubleshooting Touch Screen Inaccuracy](https://wiki.seeedstudio.com/reTerminal-FAQ/#q12-troubleshooting-touch-screen-inaccuracy) in your FAQ section but still behaves the same.

I'm running on Debian v11.3

bigbearishappy, 2023-09-12:
Have you tried to add or remove tp_rotate=1 in /boot/config.txt？

jorgedevs, 2023-09-12:
Wohoo, I had to remove it and its now working perfectly. Thanks so much!

closed 2023-09-12 by jorgedevs