Seeed-Studio/seeed-linux-dtoverlays #75: ReTerminal DM Bookworm Support

kirwinrMK, 2023-10-24:
It appears the current overlays do not support Bookworm on the ReTerminal DM. 
The raspberry pi foundation recently updated the default Raspberry Pi OS to Debian Bookworm and the install script is no longer working. 

When I attempt to install, I don't notice any errors however the onboard screen fails to turn on after a reboot and the HDMI output that worked before running the install script no longer works.

slara, 2023-10-25:
I think I'm having the same problem. No onboard screen after installing Raspberry Pi OS Lite 64 bit. reTerminal.sh script seems to run withouth problems

tonihoohoo, 2023-10-25:
It appears that the same problem occurs with regular reterminal also. Also have the problem with DM too.

bigbearishappy, 2023-10-27:
Oh, it's sorry to hear that. Can you provide the full kernel log of your device?

mistial-dev, 2023-10-30:
Just ran the script on a reTerminal DM.  It had been working with Wi-Fi (no screen), but won't come up at all after running it.

```
git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
git is already the newest version (1:2.39.2-1.1).
0 upgraded, 0 newly installed, 0 to remove and 47 not upgraded.
Cloning into 'seeed-linux-dtoverlays'...
remote: Enumerating objects: 859, done.
remote: Counting objects: 100% (859/859), done.
remote: Compressing objects: 100% (707/707), done.
remote: Total 859 (delta 160), reused 686 (delta 122), pack-reused 0
Receiving objects: 100% (859/859), 2.15 MiB | 696.00 KiB/s, done.
Resolving deltas: 100% (160/160), done.
mistial@catminder:~ $ cd seeed-linux-dtoverlays/
mistial@catminder:~/seeed-linux-dtoverlays $ sudo ./scripts/reTerminal.sh --device reTerminal-plus

### Install required tool packages
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  menu
Recommended packages:
  linux-headers-generic | linux-headers-686-pae | linux-headers-amd64 | linux-headers
The following NEW packages will be installed:
  dkms
0 upgraded, 1 newly installed, 0 to remove and 47 not upgraded.
Need to get 48.7 kB of archives.
After this operation, 190 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 dkms all 3.0.10-8+deb12u1 [48.7 kB]
Fetched 48.7 kB in 0s (127 kB/s)
Selecting previously unselected package dkms.
(Reading database ... 125759 files and directories currently installed.)
Preparing to unpack .../dkms_3.0.10-8+deb12u1_all.deb ...
Unpacking dkms (3.0.10-8+deb12u1) ...
Setting up dkms (3.0.10-8+deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...

### Will compile with the latest kernel...

### Uninstall previous dkms module

### Install required kernel package
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  raspberrypi-kernel raspberrypi-kernel-headers
0 upgraded, 2 newly installed, 0 to remove and 47 not upgraded.
Need to get 37.3 MB of archives.
After this operation, 96.8 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 raspberrypi-kernel arm64 1:1.20230405-1 [27.0 MB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 raspberrypi-kernel-headers arm64 1:1.20230405-1 [10.2 MB]
Fetched 37.3 MB in 1min 13s (513 kB/s)                                                                    
Selecting previously unselected package raspberrypi-kernel.
(Reading database ... 125783 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel_1%3a1.20230405-1_arm64.deb ...
Adding 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack'
Adding 'diversion of /boot/kernel7.img to /usr/share/rpikernelhack/kernel7.img by rpikernelhack'
Adding 'diversion of /boot/kernel7l.img to /usr/share/rpikernelhack/kernel7l.img by rpikernelhack'
Adding 'diversion of /boot/kernel8.img to /usr/share/rpikernelhack/kernel8.img by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b-plus.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-plus.dtb by rpikernelhack'
[... 699 lines cut ...]
Signing module /var/lib/dkms/rtc-pcf8563w/0.1/build/rtc-pcf8563w.ko
Cleaning build area...

rtc-pcf8563w.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/6.1.21-v8+/updates/dkms/
depmod....
  DTC     overlays/rpi/reTerminal-plus-overlay.dtbo
overlays/rpi/reTerminal-plus-overlay.dts:32.20-43.6: Warning (interrupt_provider): /fragment@1/__overlay__/pcf@21: Missing #address-cells in interrupt provider
'overlays/rpi/reTerminal-plus-overlay.dtbo' -> '/boot/firmware/overlays/reTerminal-plus.dtbo'
'/home/mistial/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/' -> '/usr/share/plymouth/themes/seeed'
'/home/mistial/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/seeed.plymouth' -> '/usr/share/plymouth/themes/seeed/seeed.plymouth'
'/home/mistial/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/seeed.script' -> '/usr/share/plymouth/themes/seeed/seeed.script'
'/home/mistial/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/splash.png' -> '/usr/share/plymouth/themes/seeed/splash.png'
'/home/mistial/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/splash_v.png' -> '/usr/share/plymouth/themes/seeed/splash_v.png'
'/home/mistial/seeed-linux-dtoverlays/extras/reTerminal/resources/10-disp.conf' -> '/usr/share/X11/xorg.conf.d/10-disp.conf'
'/home/mistial/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/plymouthd.conf' -> '/etc/plymouth/plymouthd.conf'
------------------------------------------------------
Please reboot your device to apply all settings
Enjoy!
------------------------------------------------------
```

kirwinrMK, 2023-10-30:
@bigbearishappy 
> Oh, it's sorry to hear that. Can you provide the full kernel log of your device?

Here are my dmesg outputs for both initial install and after a reboot:

[log.txt](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/13205942/log.txt)
[log_reboot.txt](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/13205944/log_reboot.txt)

bigbearishappy, 2023-11-01:
Thank you for your reply.I did a quick fix of this problem.After you pull the latest code the install the drivers again.The driver will work fine.

kirwinrMK, 2023-11-02:
That did it, Thanks!

closed 2023-11-02 by kirwinrMK