Seeed-Studio/seeed-linux-dtoverlays #76: Broken display driver for reTerminal when compiling for Yocto dunfell with linux kernel 5.10

hecko, 2023-11-02:
**Describe the bug**
Driver seeed-linux-dtoverlays/1.0-r0/git/modules/ili9881d/ili9881d.c will not compile for Yocto dunfell.

```
ERROR: seeed-linux-dtoverlays-1.0-r0 do_compile: oe_runmake failed
ERROR: seeed-linux-dtoverlays-1.0-r0 do_compile: Execution of '/home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/temp/run.do_compile.2983622' failed with exit code 1
ERROR: Logfile of failure stored in: /home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/temp/log.do_compile.2983622
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 16 ARCH=arm64 KBUILD=/home/marcel/reterminal-yocto/build/tmp/work-shared/seeed-reterminal/kernel-source O=/home/marcel/reterminal-yocto/build/tmp/work-shared/seeed-reterminal/kernel-build-artifacts CROSS_COMPILE=aarch64-poky-linux- all_rpi
| make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
|   CC [M]  /home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/ili9881d/ili9881d.o
| /home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/ili9881d/ili9881d.c: In function 'ili9881d_dsi_probe':
| /home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/ili9881d/ili9881d.c:501:12: error: 'struct drm_panel' has no member named 'prepare_upstream_first'
|   501 |  ctx->panel.prepare_upstream_first = true;
|       |            ^
| make[3]: *** [/home/marcel/reterminal-yocto/build/tmp/work-shared/seeed-reterminal/kernel-source/scripts/Makefile.build:280: /home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/ili9881d/ili9881d.o] Error 1
| make[2]: *** [/home/marcel/reterminal-yocto/build/tmp/work-shared/seeed-reterminal/kernel-source/Makefile:1825: /home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/git/modules/ili9881d] Error 2
| make[1]: *** [Makefile:185: __sub-make] Error 2
| make: *** [Makefile:157: all_rpi] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/marcel/reterminal-yocto/build/tmp/work/seeed_reterminal-poky-linux/seeed-linux-dtoverlays/1.0-r0/temp/run.do_compile.2983622' failed with exit code 1
ERROR: Task (/home/marcel/reterminal-yocto/layers/meta-seeed-cm4/recipes-kernel/seeed-linux-dtoverlays/seeed-linux-dtoverlays.bb:do_compile) failed with exit code '1'
```

**To Reproduce**
1. Go to https://wiki.seeedstudio.com/reTerminal-Yocto/#compile-with-the-default-configuration-for-reterminal
2. For step 9 use branch kirkstone for git://git.yoctoproject.org/meta-raspberrypi (instead of master as that will not work at all)
3. Continue with the tutorial

**Expected behavior**
Compilation done for Yocto image

**Desktop (please complete the following information):**
 - Ubuntu 22.04 LTS

hecko, 2023-11-03:
I have backported two drivers to work with rpi kernel 5.10 (from meta-raspberrypi kirkstone) and in combination with yocto dunfell this compiles okay - here is the patch: https://github.com/hecko/seeed-linux-dtoverlays/tree/dunfell 

I recommend to create a new branch in this repository called "dunfell" that will apply this patch so it can be used for Yocto compilation for Dunfell. Subsequently branch master in the https://github.com/Seeed-Studio/meta-seeed-cm4/tree/dunfell repo (dunfell branch)
I can help with this if needed.

swrpug, 2023-11-04:
Confirmed success building image (Ubuntu 23.04)

- applied this patch: https://github.com/hecko/seeed-linux-dtoverlays/tree/dunfell
- used branch kirkstone for git://git.yoctoproject.org/meta-raspberrypi

Minor Typo: "For step 9 use branch kirkstone ..." should be "step 5"

However, after flashing the image, the DM screen shows nothing.
I attempted flashing a standard raspbian image and while I can ssh into the box, after installing the screen overlay, the screen does not turn on.

I would be nice to be able to reference an older image; however, the artifacts for the last successful run have expired and there are no new artifacts because the build is broken.  So, I have a $400 brick

hecko, 2023-11-04:
Try to install Ubuntu Server 22.04 LTS as found in RPI Imager, SSH into system and then install drivers as per https://wiki.seeedstudio.com/reTerminal/#install-reterminal-drivers-after-flashing-new-raspberry-pi-os-ubuntu-os-or-other-os

I did try this exact procedure yesterday and I can confirm that the display works after drivers installation as per the wiki page. 

In terms of Yocto - the display works for me, but not every time when I reboot. I need to power cycle reTerminal few times to boot the system correctly and also UART console does not works for me in Yocto, but works in Ubuntu.

swrpug, 2023-11-05:
Tried multiple times using 22.04 as per the wiki.  After flashing, unable to see reTerminal DM on the network via raspberrypi.local, so can't ssh to install display drivers.  Will need to take up with Seeed Studios, product is not living up to the hype.  Just to be clear, I'm working with an reTerminal DM.  The content of the page appears to be the same though.

hecko, 2023-11-05:
Have you checked dhcp server whether ip has been assigned to the reTermina after boot? Have you tried accesing console using uart on the 40pin header?

bigbearishappy, 2023-11-06:
Sorry to hear your probelm.We will try our best to fix these problem.
Let me summarize the problem list:
reTerminal:
1 display driver broken with dunfell and kernel 5.10
2 display not work fine everytime booted
3 UART not working
reTerminal DM:
1 display driver broken(even with raspbian OS)

If there are any omissions. Please add them.

For reTerminal, we will make the display driver to work with kernel 6.1.x and let you here after we finished it.
For reTerminal DM, we will do the work same as reTermianl. And the problem of display with raspbian OS. Have you checked this issue(https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/75)? Maybe it can help you.

hecko, 2023-11-06:
> Sorry to hear your probelm.We will try our best to fix these problem. Let me summarize the problem list: reTerminal: 1 display driver broken with dunfell and kernel 5.10 2 display not work fine everytime booted 3 UART
> 
> 
> For reTerminal, we will make the display driver to work with kernel 6.1.x and let you here after we finished it. For reTerminal DM, we will do the work same as reTermianl. And the problem of display with raspbian OS. Have you checked this issue(#75)? Maybe it can help you.

For Yocto on reTerminal:
* Display works with some minor patches to drivers - I dont thisnk this needs much work (as previously mentioned in this thread it just create a new branch here called "dunfell" and I can push you the changes required to build driver on Yocto Dunfell. No need to port drivers to 6.1.x at this stage.
* reTerminal does not always boot with Yocto - the problem is, that since UART also does not work for me this is difficult to debug - What I recommend is to primarily - firstly - enable UART on reTerminal for Yocto Dunfell and afterwards we can start debugging other parts of the Yocto build, but without UART there is not much feedback I can give for Yocto if the system does not boot. It boots sometimes, but not every time.

swrpug, 2023-11-06:
I'll summarize the issues I found with the reTerminal DM:

- Yocto build fails as per [#76](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/76), but can be fixed with source patch (although I have not successfully flash the image to the device)
- Following the wiki for flashing Ubuntu 20.04 leaves the device invisible on the network making the step to compile the screen overlay impossible
- My attempts to flash the Yocto image appear to be my own lack of knowledge wrt to wic images.  I'm still trying to figure that out.

I was able to flash Sensecraft-edge-OS-V0.3.10 and restore the device, so despite my previous assertions, I no longer have a brick :-)

hecko, 2023-11-06:
> I'll summarize the issues I found with the reTerminal DM:
>
> * Yocto build fails as per [#76](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/76), but can be fixed with source patch [...]

Regarding flashing the Yocto image to reTerminal - I had the same problem that RPI Imager was telling me that the image is not the correct format - something regarding unable to extract using bz2 - so I just extracted the wic image from the bz2 package and flashed that to reTerminal. Does this sound similar to your issue @swrpug ?

swrpug, 2023-11-07:
@hecko thank you for the suggestion to unzip, that helped get me further.  I'm working out my learning on an SD card.  The issue I see is that rpi-imager's (v1.8.1) file filter only shows files with .img extension.  If I create a link, link.img, that points to the .wic file, rpi-imager happily flashes an SD card.  At this point, I have successfully built, flashed and brought up a minimal pi image using yocto.  Next, I'll try same with the reterminal yocto image.  Thanks again for your help.

bigbearishappy, 2023-11-13:
We recommand you to use the [balenaEtcher](https://etcher.balena.io/) as the flash tool for yocto image.

bigbearishappy, 2023-11-15:
For the futhur discussion of this problem. Please look into [here](https://github.com/Seeed-Studio/meta-seeed-cm4/issues/12)

closed 2023-12-04 by bigbearishappy