Seeed-Studio/seeed-linux-dtoverlays #77: The screen is not rotating in reTerminal bookworm.

mitoneko, 2023-11-03:
After applying the reTerminal.sh after the Bookworm support update (2023/11/03), after restarting, the screen is not rotating. even if I put the reTerminal in landscape mode, it remains in portrait mode.
The output of script/reTermilal.sh is as follows.
[reterminal.log](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/13246839/reterminal.log)
The dmesg log immediately after startup is as follows.
[dmesg.log](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/13246841/dmesg.log)

bigbearishappy, 2023-11-06:
Can you show me the detail of your putting reTerminal in landscape mode?
And have tried to change rotation with this command 
`DISPLAY=:0 xrandr --output DSI-1 --rotate right` ?

mitoneko, 2023-11-10:
> Can you show me the detail of your putting reTerminal in landscape mode? And have tried to change rotation with this command `DISPLAY=:0 xrandr --output DSI-1 --rotate right` ?

The results of the run are as follows.

pi@reterminal:~ $ DISPLAY=:0 xrandr --output DSI-1 --rotate right
warning: output DSI-1 not found; ignoring

mitoneko, 2023-11-10:
情報表示をすると次の通り。
DSI-1はやはりなさそう。
pi@reterminal:~ $ DISPLAY=:0 xrandr
Screen 0: minimum 16 x 16, current 720 x 1280, maximum 32767 x 32767
XWAYLAND0 connected 720x1280+0+0 (normal left inverted right x axis y axis) 62mm x 110mm
   720x1280      59.84*+
   640x480       59.38  
   320x240       59.52  
   720x480       59.71  
   640x400       59.95  
   320x200       58.96  
   720x400       59.55  
   640x350       59.77

mitoneko, 2023-11-10:
I tried the following to no avail.
pi@reterminal:~ $ DISPLAY=:0 xrandr --output XWAYLAND0 --rotate right
X Error of failed request:  BadMatch (invalid parameter attributes)
  Major opcode of failed request:  139 (RANDR)
  Minor opcode of failed request:  7 (RRSetScreenSize)
  Serial number of failed request:  22
  Current serial number in output stream:  23

bigbearishappy, 2023-11-13:
It seem xrandr can't change the LCD rotation now.
Two steps can make it work:
step one:
append the follow code to .config/wayfire.ini
```
[output:DSI-1]
mode = 720x1280@60
transform = 270
```
step two:
change the dtoverlay of reTerminal from `dtoverlay=reTerminal,tp_rotate=1` to `dtoverlay=reTerminal`

Finally reboot the system.

mitoneko, 2023-11-13:
Thanks for the advices.

As soon as step1 is saved, the screen is now in landscape mode.
When step2 is executed and rebooted, the screen remains in landscape mode.
It seems like a step forward.
However, in the past, when I rotated the device, the screen followed the rotation, but this time, the screen remains in landscape even if I turn it to portrait.

bigbearishappy, 2023-11-14:
So you want the screen to follow the rotation?
If so, I think it's better for you to close this issue and open a new one for this new problem.
Thank you.

n7ihq, 2023-11-29:
Thank you for the Bookworm display orientation fix!

n7ihq, 2023-12-02:
I made the changes as described below. The horizontal screen and touch orientation is correct. However the mouse positioning is rotated when using using TigerVNC for remote desktop access.

> It seem xrandr can't change the LCD rotation now. Two steps can make it work: step one: append the follow code to .config/wayfire.ini
>
> ```
> [output:DSI-1]
> mode = 720x1280@60
> transform = 270
> ```
>
> step [...]

n7ihq, 2023-12-07:
> I made the changes as described below. The horizontal screen and touch orientation is correct. However the mouse movement is rotated when using using TigerVNC for remote desktop access.
> Can you have a try with a USB mouse which connect to reTerminal?

Thank you for your response. I connected a USB mouse to the reTerminal. The USB mouse positioning is correct on the TigerVNC client display. The cursor positioning using a mouse in TigerVNC is rotated.