Seeed-Studio/seeed-linux-dtoverlays #78: Screen has the correct rotation, but touch behaves like the opposite

franzos, 2023-11-04:
**Describe the bug**
- Screen is rotated horizontally
- Moving the finger up and down, moves the mouse left and right

-> The mouse behaves like it's a vertical screen

**To Reproduce**

Tried with `tp_rotate=1`, `tp_rotate=0` and without.
- https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/73
- https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/58#issuecomment-1476324102

```
...

[all]
enable_uart=1
uart_2ndstage=1
dtoverlay=dwc2,dr_mode=host
dtparam=ant2
disable_splash=1
ignore_lcd=1
dtoverlay=vc4-kms-v3d-pi4
dtoverlay=i2c3,pins_4_5
gpio=13=pu
dtoverlay=reTerminal,key0=0x043,key1=0x044,key2=0x057,key3=0x058
# dtoverlay=reTerminal-bridge
```

**Expected behavior**
The touch screen to work  correctly.

**OS:**
```
6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
https://downloads.raspberrypi.com/raspios_lite_arm64/images/raspios_lite_arm64-2023-10-10/2023-10-10-raspios-bookworm-arm64-lite.img.xz
```

**Additional context**
- May be related to https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/77

franzos, 2023-11-05:
I collected some more logs, in case they are relevant.

```bash
./scripts/reTerminal.sh

### Install required tool packages
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian bookworm InRelease       
Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (3.0.10-8+deb12u1).
The following packages were automatically installed and are no longer required:
  libxcb-shape0 libxv1 libxxf86dga1
Use 'apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

### Will compile with the latest kernel...

### Uninstall previous dkms module
Module mipi_dsi-0.1 for kernel 6.1.0-rpi4-rpi-v8 (aarch64).
Before uninstall, this module version was ACTIVE on this kernel.

mipi_dsi.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.1.0-rpi4-rpi-v8/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod....
Deleting module mipi_dsi-0.1 completely from the DKMS tree.
Module ltr30x-0.1 for kernel 6.1.0-rpi4-rpi-v8 (aarch64).
Before uninstall, this module version was ACTIVE on this kernel.

als_ltr30x.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.1.0-rpi4-rpi-v8/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod....
Deleting module ltr30x-0.1 completely from the DKMS tree.
Module lis3lv02d-0.1 for kernel 6.1.0-rpi4-rpi-v8 (aarch64).
Before uninstall, this module version was ACTIVE on this kernel.

lis331dlh-i2c.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.1.0-rpi4-rpi-v8/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod....
Deleting module lis3lv02d-0.1 completely from the DKMS tree.
Module bq24179_charger-0.1 for kernel 6.1.0-rpi4-rpi-v8 (aarch64).
Before uninstall, this module version was ACTIVE on this kernel.

bq24179_charger.ko.xz:
 - Uninstallation
   - Deleting from: /lib/modules/6.1.0-rpi4-rpi-v8/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod....
Deleting module bq24179_charger-0.1 completely from the DKMS tree.
removed '/boot/firmware/overlays/reTerminal.dtbo'
removed 'overlays/rpi/.reTerminal-overlay.dtbo.d.dtc.tmp'
removed 'overlays/rpi/.reTerminal-overlay.dtbo.d.pre.tmp'
[... 129 lines cut ...]
```

It seems `dtoverlay=reTerminal,key0=0x043,key1=0x044,key2=0x057,key3=0x058,tp_rotate=0` has no effect.

I used to use `key0=0x043,key1=0x044,key2=0x057,key3=0x058` to map the buttons to F9, F10, F11, F12 but that stopped working too. Probably unrelated, but never know ... Here's how I used it in openbox:

```xml
<keybind key="F9">
  <action name="Execute">
    <command>/home/default/.config/openbox/poweroff.sh</command>
  </action>
</keybind>
```

to

```xml
<keybind key="0x26">
  <action name="Execute">
    <command>/home/default/.config/openbox/poweroff.sh</command>
  </action>
</keybind>
```

Instead of F9, `0x26` works now.

bigbearishappy, 2023-11-06:
Can you comment the `dtoverlay=reTerminal` and get the firmware version of STM32 with this command `i2ctransfer -y 1 w1@0x45 0x97 r2`?

franzos, 2023-11-06:
> Can you comment out the `dtoverlay=reTerminal` in /boot/config.txt and get the firmware version of STM32 with this command `i2ctransfer -y 1 w1@0x45 0x97 r2`?

I tried to comment out this line, but it makes no difference.

```
#dtoverlay=reTerminal,key0=0x043,key1=0x044,key2=0x057,key3=0x058,tp_rotate=0
```

and `i2ctransfer -y 1 w1@0x45 0x97 r2` would result in something like `/dev/... ` not found.

I will try this again later, and provide the exact error.

franzos, 2023-11-07:
> Can you comment out the `dtoverlay=reTerminal` in /boot/config.txt and get the firmware version of STM32 with this command `i2ctransfer -y 1 w1@0x45 0x97 r2`?

Here's what I got:

1. Comment `#dtoverlay=reTerminal,...`
2. Reboot
3. Check:

```bash
$ i2ctransfer -y 1 w1@0x45 0x97 r2
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory
```

to be sure:

```bash
$ cat /boot/config.txt | grep dtoverlay=reTerminal
# dtoverlay=reTerminal,key0=0x043,key1=0x044,key2=0x057,key3=0x058,tp_rotate=1
# dtoverlay=reTerminal-bridge
```

franzos, 2023-11-09:
I tried again with a "supported" kernel; Now nothing works anymore.

```bash
wget https://downloads.raspberrypi.com/raspios_arm64/images/raspios_arm64-2022-01-28/2022-01-28-raspios-bullseye-arm64.zip
```

After flashing, create a `ssh` file on the `boot` partition, so you can login.

Once booted, again install the latest driver for the display to work:

```bash
cd /tmp
git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays --depth 1
./scripts/reTerminal.sh
```

and the logs:

```bash
root@raspberrypi:~/seeed-linux-dtoverlays# ./scripts/reTerminal.sh

### Install required tool packages
Get:1 http://security.debian.org/debian-security bullseye-security InRelease [48.4 kB]
Get:2 http://deb.debian.org/debian bullseye InRelease [116 kB]
Get:3 http://archive.raspberrypi.org/debian bullseye InRelease [23.6 kB]
Get:4 http://deb.debian.org/debian bullseye-updates InRelease [44.1 kB]
Get:5 http://security.debian.org/debian-security bullseye-security/main armhf Packages [255 kB]
Get:6 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [313 kB]
Get:7 http://security.debian.org/debian-security bullseye-security/main arm64 Packages [253 kB]
Get:8 http://security.debian.org/debian-security bullseye-security/main Translation-en [166 kB]
Get:9 http://security.debian.org/debian-security bullseye-security/non-free Translation-en [472 B]
Get:10 http://deb.debian.org/debian bullseye/main arm64 Packages [7,951 kB]
Get:11 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages [307 kB]
Get:12 http://deb.debian.org/debian bullseye/main armhf Packages [7,838 kB]
Get:13 http://deb.debian.org/debian bullseye/main Translation-en [6,236 kB]
Get:14 http://deb.debian.org/debian bullseye/contrib armhf Packages [40.1 kB]
Get:15 http://deb.debian.org/debian bullseye/contrib arm64 Packages [40.8 kB]
Get:16 http://deb.debian.org/debian bullseye/contrib Translation-en [46.9 kB]
Get:17 http://deb.debian.org/debian bullseye/non-free armhf Packages [56.0 kB]
Get:18 http://deb.debian.org/debian bullseye/non-free arm64 Packages [72.2 kB]
Get:19 http://deb.debian.org/debian bullseye/non-free Translation-en [92.4 kB]
Get:20 http://deb.debian.org/debian bullseye-updates/main armhf Packages [14.9 kB]
Get:21 http://deb.debian.org/debian bullseye-updates/main arm64 Packages [14.9 kB]
Get:22 http://deb.debian.org/debian bullseye-updates/main Translation-en [10.1 kB]
Fetched 23.9 MB in 15s (1,553 kB/s)
Reading package lists... Done
N: Repository 'http://security.debian.org/debian-security bullseye-security InRelease' changed its 'Suite' value from 'stable-security' to 'oldstable-security'
N: Repository 'http://archive.raspberrypi.org/debian bullseye InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: Repository 'http://deb.debian.org/debian bullseye InRelease' changed its 'Version' value from '11.2' to '11.8'
N: Repository 'http://deb.debian.org/debian bullseye InRelease' changed its 'Suite' value from 'stable' to 'oldstable'
N: Repository 'http://deb.debian.org/debian bullseye-updates InRelease' changed its 'Suite' value from 'stable-updates' to 'oldstable-updates'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following package was automatically installed and is no longer required:
  libfuse2
Use 'apt autoremove' to remove it.
The following additional packages will be installed:
  dctrl-tools
Suggested packages:
  debtags menu
Recommended packages:
  linux-headers-686-pae | linux-headers-amd64 | linux-headers-generic | linux-headers
The following NEW packages will be installed:
  dctrl-tools dkms
0 upgraded, 2 newly installed, 0 to remove and 318 not upgraded.
Need to get 179 kB of archives.
After this operation, 720 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bullseye/main arm64 dctrl-tools arm64 2.24-3 [101 kB]
Get:2 http://deb.debian.org/debian bullseye/main arm64 dkms all 2.8.4-3 [78.2 kB]
[... 1727 lines cut ...]
[    6.940625] input: gpio_keys as /devices/platform/gpio_keys/input/input25
[    7.055881] 8021q: 802.1Q VLAN Support v1.8
[    7.356101] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[    7.357132] bcmgenet fd580000.ethernet eth0: Link is Down
[   11.454937] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   11.455044] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.386630] Bluetooth: Core ver 2.22
[   13.386734] NET: Registered PF_BLUETOOTH protocol family
[   13.386741] Bluetooth: HCI device and connection manager initialized
[   13.386758] Bluetooth: HCI socket layer initialized
[   13.386767] Bluetooth: L2CAP socket layer initialized
[   13.386781] Bluetooth: SCO socket layer initialized
[   13.395299] Bluetooth: HCI UART driver ver 2.3
[   13.395323] Bluetooth: HCI UART protocol H4 registered
[   13.395416] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   13.395635] Bluetooth: HCI UART protocol Broadcom registered
[   13.588771] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   13.588791] Bluetooth: BNEP filters: protocol multicast
[   13.588807] Bluetooth: BNEP socket layer initialized
[   13.595546] Bluetooth: MGMT ver 1.22
[   13.605044] NET: Registered PF_ALG protocol family
[   13.648424] Bluetooth: RFCOMM TTY layer initialized
[   13.648462] Bluetooth: RFCOMM socket layer initialized
[   13.648485] Bluetooth: RFCOMM ver 1.11
```

franzos, 2023-11-10:
I'm not sure why, but it looks like the issue is fixed.
1. Installed raspbian 64bit desktop
2. install overlays (veritcal screen after reboot)
3. install openbox; modify lightdm to use openbox (horizontal screen after reboot); not sure why the orientation changes but it's desired

There's 2 notable changes in my config; I added:

```
auto_initramfs=1
disable_fw_kms_setup=1
```

Kernel is still

```
6.1.0-rpi6-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27) aarch64
```

Closing this since it's resolved.

closed 2023-11-10 by franzos