Seeed-Studio/seeed-linux-dtoverlays #79: Creating kirkstone branch in this repository for code supported on Yocto Kirkstone

hecko, 2023-11-06:
Hello, I would like to request a creation of "kirkstone" branch in this repository so I can push changes that will make this repository to work with reTerminal with Yocto Kirkwall, Thanks.

bigbearishappy, 2023-11-10:
Hello, there is no need to do this. This repo should work fine with yocto kirkstone correctly. Maybe there is compile problem. We will fix it soon.

bigbearishappy, 2023-11-14:
Thank you for your suggestion here. 
Actually the main branch of meta-seeed-cm4 is based on yocto kirkstone.
Next time if you want to push changes of kirkstone. You can push it to the branch main of meta-seeed-cm4. If there is any furthur work for you to do.We will tell you in the PR.
And we just push the changes(https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/6de5e7f5b5dd2761ff644616c207482a31f7624f https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/8bbb823146076b9c1f81a725e95d91e240c48d9d) for reTerminal to this repo based on your code in [here](https://github.com/hecko/seeed-linux-dtoverlays/commit/6c70d13d09ed9c3697ea681c79f5b5d30fa6cdd6)

Thank you again.

closed 2023-11-14 by bigbearishappy