Seeed-Studio/seeed-linux-dtoverlays #82: reTerminal: Display does not appear if you git upgrade before driver install

matsujirushi, 2023-11-28:
Display does not appear if you git upgrade before driver install.

### Raspberry Pi OS

bookworm 64bit  2023-10-10

```
pi@raspberrypi:~ $ cat /boot/firmware/issue.txt
Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05) aarch64 GNU/Linux
pi@raspberrypi:~ $ getconf LONG_BIT
64
```

### Procedure

#### Install Raspberry Pi OS

Using Raspberry Pi Imager. 

#### apt upgrade

```
pi@raspberrypi:~ $ sudo apt update && sudo apt upgrade
...
The following packages were automatically installed and are no longer required:
  libcamera-apps libpisp0.0.1
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libpisp1 linux-headers-6.1.0-rpi6-common-rpi linux-headers-6.1.0-rpi6-rpi-2712 linux-headers-6.1.0-rpi6-rpi-v8 linux-image-6.1.0-rpi6-rpi-2712
  linux-image-6.1.0-rpi6-rpi-v8 rpicam-apps
The following packages will be upgraded:
  bluez-firmware curl ffmpeg firefox firmware-atheros firmware-brcm80211 firmware-libertas firmware-misc-nonfree firmware-realtek gldriver-test
  gstreamer1.0-plugins-bad libavcodec59 libavdevice59 libavfilter8 libavformat59 libavutil57 libcamera-apps libcamera-ipa libcamera-tools libcamera0.1
  libcurl3-gnutls libcurl4 libegl-mesa0 libfm-data libfm-extra4 libfm-gtk-data libfm-gtk4 libfm-modules libfm4 libgbm1 libgl1-mesa-dev libgl1-mesa-dri
  libglapi-mesa libglx-mesa0 libgstreamer-plugins-bad1.0-0 libjavascriptcoregtk-4.1-0 libldb2 libneatvnc0 libpam-chksshpwd libpam-modules
  libpam-modules-bin libpam-runtime libpam0g libpipewire-0.3-0 libpipewire-0.3-common libpipewire-0.3-modules libpisp-common libpostproc56
  libraspberrypi-bin libraspberrypi-dev libraspberrypi-doc libraspberrypi0 libsmbclient libspa-0.2-bluetooth libspa-0.2-modules libssl3 libswresample4
  libswscale6 libtiff6 libvlc-bin libvlc5 libvlccore9 libwbclient0 libwebkit2gtk-4.1-0 libwf-utils0 libwidevinecdm0 libwlroots11 linux-headers-rpi-2712
  linux-headers-rpi-v8 linux-image-rpi-2712 linux-image-rpi-v8 linux-kbuild-6.1 linux-libc-dev lxplug-ejecter lxplug-updater lxplug-volumepulse
  mesa-vdpau-drivers openssl pcmanfm pi-package pi-package-data pi-package-session pipanel pipewire pipewire-bin pipewire-libcamera pipewire-pulse piwiz
  pixflat-icons python3-libcamera python3-picamera2 python3-sense-hat raspberrypi-net-mods raspberrypi-ui-mods raspi-config raspi-firmware raspi-utils
  rc-gui rpi-eeprom rpi-imager samba-libs thonny vlc vlc-bin vlc-data vlc-l10n vlc-plugin-access-extra vlc-plugin-base vlc-plugin-notify vlc-plugin-qt
  vlc-plugin-samba vlc-plugin-skins2 vlc-plugin-video-output vlc-plugin-video-splitter vlc-plugin-visualization wayfire wayvnc wf-panel-pi xserver-common
  xserver-xorg-core
120 upgraded, 7 newly installed, 0 to remove and 0 not upgraded.
Need to get 306 MB of archives.
After this operation, 138 MB of additional disk space will be used.
Do you want to continue? [Y/n]
...
```

#### Install seeed driver

```
pi@raspberrypi:~ $ cd && git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
Cloning into 'seeed-linux-dtoverlays'...
remote: Enumerating objects: 859, done.
remote: Counting objects: 100% (859/859), done.
remote: Compressing objects: 100% (707/707), done.
remote: Total 859 (delta 160), reused 693 (delta 122), pack-reused 0
Receiving objects: 100% (859/859), 2.15 MiB | 3.65 MiB/s, done.
Resolving deltas: 100% (160/160), done.
pi@raspberrypi:~ $ cd seeed-linux-dtoverlays
pi@raspberrypi:~/seeed-linux-dtoverlays $ git log -n 1
commit 673d491e1c3f4d528cb7379103c5c514ea12dbb5 (grafted, HEAD -> master, origin/master, origin/HEAD)
Author: bigbearishappy <953308023@qq.com>
Date:   Tue Nov 14 08:45:11 2023 +0000

    fix kernel version compatibility issue

    Signed-off-by: bigbearishappy <953308023@qq.com>
pi@raspberrypi:~/seeed-linux-dtoverlays $ sudo ./scripts/reTerminal.sh
pi@raspberrypi:~/seeed-linux-dtoverlays $ sudo reboot
```

matsujirushi, 2023-11-28:
<details>
<summary>reTerminal.sh messages</summary>

```
### Install required tool packages
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:3 http://deb.debian.org/debian bookworm-updates InRelease
Hit:4 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libcamera-apps libpisp0.0.1
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  menu
Recommended packages:
  linux-headers-generic | linux-headers-686-pae | linux-headers-amd64 | linux-headers
The following NEW packages will be installed:
  dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 48.7 kB of archives.
After this operation, 190 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 dkms all 3.0.10-8+deb12u1 [48.7 kB]
Fetched 48.7 kB in 0s (554 kB/s)
Selecting previously unselected package dkms.
(Reading database ... 145177 files and directories currently installed.)
Preparing to unpack .../dkms_3.0.10-8+deb12u1_all.deb ...
Unpacking dkms (3.0.10-8+deb12u1) ...
Setting up dkms (3.0.10-8+deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...

### Will compile with the latest kernel...

### Uninstall previous dkms module

### Install required kernel package
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libcamera-apps libpisp0.0.1
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  raspberrypi-kernel raspberrypi-kernel-headers
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 37.3 MB of archives.
After this operation, 96.8 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.com/debian bookworm/main arm64 raspberrypi-kernel arm64 1:1.20230405-1 [27.0 MB]
Get:2 http://archive.raspberrypi.com/debian bookworm/main arm64 raspberrypi-kernel-headers arm64 1:1.20230405-1 [10.2 MB]
Fetched 37.3 MB in 6s (6,070 kB/s)
Selecting previously unselected package raspberrypi-kernel.
(Reading database ... 145201 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel_1%3a1.20230405-1_arm64.deb ...
Adding 'diversion of /boot/kernel.img to /usr/share/rpikernelhack/kernel.img by rpikernelhack'
Adding 'diversion of /boot/kernel7.img to /usr/share/rpikernelhack/kernel7.img by rpikernelhack'
Adding 'diversion of /boot/kernel7l.img to /usr/share/rpikernelhack/kernel7l.img by rpikernelhack'
Adding 'diversion of /boot/kernel8.img to /usr/share/rpikernelhack/kernel8.img by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b-plus.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-plus.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b-rev1.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b-rev1.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-b.dtb to /usr/share/rpikernelhack/bcm2708-rpi-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-cm.dtb to /usr/share/rpikernelhack/bcm2708-rpi-cm.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-zero-w.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero-w.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2708-rpi-zero.dtb to /usr/share/rpikernelhack/bcm2708-rpi-zero.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2709-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2709-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2709-rpi-cm2.dtb to /usr/share/rpikernelhack/bcm2709-rpi-cm2.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-2-b.dtb to /usr/share/rpikernelhack/bcm2710-rpi-2-b.dtb by rpikernelhack'
Adding 'diversion of /boot/bcm2710-rpi-3-b-plus.dtb to /usr/share/rpikernelhack/bcm2710-rpi-3-b-plus.dtb by rpikernelhack'
[... 704 lines cut ...]
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/' -> '/usr/share/plymouth/themes/seeed'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/seeed.plymouth' -> '/usr/share/plymouth/themes/seeed/seeed.plymouth'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/seeed.script' -> '/usr/share/plymouth/themes/seeed/seeed.script'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/splash.png' -> '/usr/share/plymouth/themes/seeed/splash.png'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/seeed/splash_v.png' -> '/usr/share/plymouth/themes/seeed/splash_v.png'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/10-disp.conf' -> '/usr/share/X11/xorg.conf.d/10-disp.conf'
'/home/pi/seeed-linux-dtoverlays/extras/reTerminal/resources/plymouth/plymouthd.conf' -> '/etc/plymouth/plymouthd.conf'
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
No state is present for card vc4hdmi0
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:0 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi0
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
No state is present for card vc4hdmi1
alsa-lib main.c:1541:(snd_use_case_mgr_open) error: failed to import hw:1 use case configuration -2
Found hardware: "vc4-hdmi" "" "" "" ""
Hardware is initialized using a generic method
No state is present for card vc4hdmi1
------------------------------------------------------
Please reboot your device to apply all settings
Enjoy!
------------------------------------------------------
```
</details>

matsujirushi, 2023-11-28:
<details>
<summary>dmesg messages</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.1.0-rpi6-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.1.58-1+rpt2 (2023-10-27)
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone DMA32: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 29 pages/cpu s79144 r8192 d31448 u118784
[    0.000000] pcpu-alloc: s79144 r8192 d31448 u118784 alloc=29*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 996912
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=E4:5F:01:91:0E:39 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=501d116d-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles logo.nologo vt.global_cursor_default=0 console=tty3 loglevel=0
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 3338984K/4050944K available (12352K kernel code, 2172K rwdata, 4164K rodata, 4224K init, 1082K bss, 187672K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40064 entries in 157 pages
[    0.000000] ftrace: allocated 157 pages with 5 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Trampoline variant of Tasks RCU enabled.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000000] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000327] Console: colour dummy device 80x25
[    0.000390] printk: console [tty1] enabled
[    0.000457] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000487] pid_max: default: 32768 minimum: 301
[    0.000620] LSM: Security Framework initializing
[    0.000836] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000900] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[... 364 lines cut ...]
[    9.646078] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    9.654200] Bluetooth: hci0: BCM4345C0 'brcm/BCM4345C0.raspberrypi,4-compute-module.hcd' Patch
[   10.046890] brcmfmac: F1 signature read @0x18000000=0x15264345
[   10.158179] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[   10.161403] usbcore: registered new interface driver brcmfmac
[... cut ...]
[   12.010472] Bluetooth: BNEP filters: protocol multicast
[   12.010490] Bluetooth: BNEP socket layer initialized
[   12.018174] Bluetooth: MGMT ver 1.22
[   12.036615] NET: Registered PF_ALG protocol family
[   13.445745] Bluetooth: RFCOMM TTY layer initialized
[   13.445778] Bluetooth: RFCOMM socket layer initialized
[   13.445799] Bluetooth: RFCOMM ver 1.11
[   15.999472] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   16.000129] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
```
</details>

bigbearishappy, 2023-12-04:
It seems that the install scripts compile the driver code for the kernel which is not running now.We will fix this.
Before that, you can just run the following command to make the LCD work:
```
make all_rpi
sudo make install_rpi
```

matsujirushi, 2024-01-11:
The problem no longer occurs on the latest Raspberry Pi OS. 😅

### Raspberry Pi OS

bookworm 64bit  2023-12-05

```
pi@raspberrypi:~ $ cat /boot/firmware/issue.txt
Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 70cd6f2a1e34d07f5cba7047aea5b92457372e05, stage4
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux
pi@raspberrypi:~ $ getconf LONG_BIT
64
```

closed 2024-01-11 by matsujirushi