Seeed-Studio/seeed-linux-dtoverlays #83: STM32G030 firmware version

matsujirushi, 2023-12-11:
Although the release name says "V1.9", the version of STM32G030F6_R2.bin is v1.8.
https://github.com/Seeed-Studio/seeed-linux-dtoverlays/releases/tag/2022-05-29-reTerminal-V1.9

Could you please write the firmware version of STM32G030F6_R2.bin in the release description?

**Links:**

* [Check the installed STM32G030 firmware version](https://wiki.seeedstudio.com/reterminal_black_screen/#check-the-installed-stm32g030-firmware-version)

bigbearishappy, 2023-12-11:
Thank you for your feedback. It's my mistake to forget update the version number in MCU souce code.
I have just fix the version number and update the release file in [here](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/releases/tag/2022-05-29-reTerminal-V1.9).
You can download and test it.

matsujirushi, 2023-12-11:
Thank you for update. 😊
I will test.

matsujirushi, 2023-12-11:
The version number has been updated.
```
$ i2ctransfer -y 1 w1@0x45 0x97 r2
0x01 0x09
```

Thank you for your quick support.

closed 2023-12-11 by matsujirushi