Seeed-Studio/seeed-linux-dtoverlays #84: reTerminal DM: ch342f chip issue details

matsujirushi, 2024-01-11:
Hi,
I connected a USB-CDC device to the USB of reTerminal DM, but Serial is not recognized.
I checked and found that the cdc_acm driver was disabled.

https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/master/scripts/reTerminal.sh#L430-L432

According to the comments, it seems that the cdc_acm driver is disabled due to an issue with the ch342f chip.
I would like to resolve the issue with the ch342f chip.

Could you please provide more details?

bigbearishappy, 2024-01-12:
It's nice of you to help us fix this problem.
Here is detail steps to reproduce the issue of ch342f:
1 remove `blacklist cdc_acm` from `/etc/modprobe.d/raspi-blacklist.conf`
2 connect you PC and reTerminal DM with a usb-to-rs232 converter
3 opent the terminal of usb-to-rs232 on your computer
4 reboot the system
5 you will get some garbled characters when the system is booting up
These garbled characters is not what we want. So we install a new driver to fix this.

If you want to fix this. I think the best way is to take look into the source code of cdc_acm and try to support the ch342f correctly in it~

matsujirushi, 2024-01-15:
Thank you for the detailed information.

closed 2024-01-15 by matsujirushi