Seeed-Studio/seeed-linux-dtoverlays #86: reTerminal: black screen issue

matsujirushi, 2024-01-13:
**I am currently working on a fix.**

Even after installing the driver, the screen does not appear in the following environments.

* Raspberry Pi OS 32-bit
  * 2023-05-03-raspios-bullseye-armhf.img.xz
    * arm_64bit=1 - default
  * 2023-10-10-raspios-bookworm-armhf.img.xz
    * arm_64bit=1 - default
    * arm_64bit=0
  * 2023-12-05-raspios-bookworm-armhf.img.xz
    * arm_64bit=1 - default
    * arm_64bit=0

I will fix reTerminal.sh script and test in these environments.

* Raspberry Pi OS 32-bit
  * 2021-05-07-raspios-buster-armhf.zip
  * 2023-02-21-raspios-bullseye-armhf.img.xz
  * 2023-05-03-raspios-bullseye-armhf.img.xz
    * arm_64bit=1 - default
    * arm_64bit=0
  * 2023-10-10-raspios-bookworm-armhf.img.xz
    * arm_64bit=1 - default
    * arm_64bit=0
  * 2023-12-05-raspios-bookworm-armhf.img.xz
    * arm_64bit=1 - default
    * arm_64bit=0
* Raspberry Pi OS 64-bit
  * 2021-05-07-raspios-buster-arm64.zip
  * 2023-05-03-raspios-bullseye-arm64.img.xz
  * 2023-10-10-raspios-bookworm-arm64.img.xz
  * 2023-12-05-raspios-bookworm-arm64.img.xz
* Ubuntu OS 64-bit
  * Ubuntu Desktop 22.04.3 LTS (2023-08-10)

matsujirushi, 2024-01-15:
`/scripts/reTerminal.sh` flow is:

```mermaid
flowchart TD
    id1(Check /boot volume space)
    id2(Install dkms package)
    id3(Uninstall modules and overlays)
    id4(Upgrade kernel)
    id5(Install kernel header)
    id6(Install modules and overlays)
    id7(Modify few files)
    
    id1 --> id2
    id2 --> id3
    id3 --> id4
    id4 --> id5
    id5 --> id6
    id6 --> id7
```

* The code to get the kernel version when installing a module is complicated.
  * **The kernel version cannot be obtained correctly with the following OS images.**
    * 2023-10-10-raspios-bookworm-armhf.img.xz + arm_64bit=0
    * 2023-12-05-raspios-bookworm-armhf.img.xz + arm_64bit=0
* By default, upgrade the Kernel.
* Not rebooted after upgrading Kernel.
  * **Unable to get kernel version with `uname -r`.**
* `--keep-kernel` can prevent kernel upgrades, but it seems that **it is not compatible with Ubuntu**.

matsujirushi, 2024-01-16:
Change Strategy:
* 1st phase
  * First, we will modify the operation for Raspberry Pi OS. Unchanged for Ubuntu.
  * For Raspberry Pi OS, `--keep-kernel` option recommended. (DON'T upgrade the kernel.)
  * Raspberry Pi OS 64-bit kernel + 32-bit userland is NOT supported. (Change to 32-bit kernel + 32-bit userland with `arm_64bit=0`)
* 2nd phase
  * Support `--keep-kernel` option for Ubuntu.
* 3rd phase
  * Enable `--keep-kernel` option by default

What do you think? @bigbearishappy

matsujirushi, 2024-01-17:
# 1st phase

## Commands

### for Raspberry Pi OS

When using a 32-bit OS, add `arm_64bit=0` to /boot/config.txt (or /boot/firmware/config.txt) and boot.

```
git clone -b issue86/step1 --depth 1 https://github.com/matsujirushi/seeed-linux-dtoverlays
cd seeed-linux-dtoverlays
sudo ./scripts/reTerminal.sh --keep-kernel
```

## Tests

- [ ] Raspberry Pi OS 32-bit
  - [ ] 2021-05-07-raspios-buster-armhf.zip
  - [ ] 2023-02-21-raspios-bullseye-armhf.img.xz
  - [ ] 2023-05-03-raspios-bullseye-armhf.img.xz
    - [ ] arm_64bit=1 - default
    - [ ] arm_64bit=0
  - [ ] 2023-10-10-raspios-bookworm-armhf.img.xz
    - [ ] arm_64bit=1 - default
    - [ ] arm_64bit=0
  - [ ] 2023-12-05-raspios-bookworm-armhf.img.xz
    - [ ] arm_64bit=1 - default
    - [ ] arm_64bit=0
- [ ] Raspberry Pi OS 64-bit
  - [ ] 2021-05-07-raspios-buster-arm64.zip
  - [ ] 2023-05-03-raspios-bullseye-arm64.img.xz
  - [ ] 2023-10-10-raspios-bookworm-arm64.img.xz
  - [ ] 2023-12-05-raspios-bookworm-arm64.img.xz
- [ ] Ubuntu OS 64-bit
  - [ ] Ubuntu Desktop 22.04.3 LTS (2023-08-10)

bigbearishappy, 2024-01-17:
> Change Strategy:
>
> * 1st phase
>
>   * First, we will modify the operation for Raspberry Pi OS. Unchanged for Ubuntu.
>   * For Raspberry Pi OS, `--keep-kernel` option recommended. (DON'T upgrade the [...]

I don't think it's a good idea to enable the --keep-kernel by default. As the raspbian OS change a lot since bookworm. The install code should seprate with two parts:
1 For old raspbian OS version(like bulleyes, busster...)
2 For new raspbian OS version(like bookworm)
At the start of the install script, we need to determine the system environment, and then deal with them with different code.

matsujirushi, 2024-01-18:
Hi @bigbearishappy ,
Thank you for response.

## enable the `--keep-kernel` by default

> I don't think it's a good idea to enable the --keep-kernel by default.

Sorry. I don't understand why.

reTerminal.sh without `--keep-kernel` compiles and installs the device driver after upgrading the kernel.
I think it is better to enable `--keep-kernel` by default for the following reasons.

* Kernel is upgraded unintentionally by the user.
* The method to obtain the kernel version is unstable. (The script is not rebooting after kernel upgrade.)

## Difference between old and new Raspberry Pi OS

> we need to determine the system environment, and then deal with them with different code.

I see.
I'm not familiar with the Raspberry Pi OS.
Could you please tell me the differences between the old and new OS that I need to deal with in the script?

bigbearishappy, 2024-01-22:
> Hi @bigbearishappy , Thank you for response.
>
> ## enable the `--keep-kernel` by default
>
> Sorry. I don't understand why.
>
> reTerminal.sh without `--keep-kernel` compiles and installs the device driver [...]

According to this [link](https://dev.blues.io/blog/easiest-upgrade-raspberry-pi-bullseye-bookworm/). You can upgrade bullseye to bookworm. It seems like the apt source link is changed between them.

matsujirushi, 2024-01-26:
FYI
* [The reTerminal driver for Raspberry Pi OS](https://github.com/matsujirushi/seeed-linux-dtoverlays/blob/main/README_reTerminal_RasPiOS.md)
* [The reTerminal driver for Ubuntu OS](https://github.com/matsujirushi/seeed-linux-dtoverlays/blob/main/README_reTerminal_UbuntuOS.md)

matsujirushi, 2024-02-19:
FYI
I create new script. `reTerminal2.sh`
https://github.com/matsujirushi/seeed-linux-dtoverlays/blob/main/scripts/reTerminal2.sh
https://github.com/matsujirushi/seeed-linux-dtoverlays/blob/main/doc/reTerminal_RasPiOS.md

bigbearishappy, 2024-02-19:
> FYI I create new script. `reTerminal2.sh` https://github.com/matsujirushi/seeed-linux-dtoverlays/blob/main/scripts/reTerminal2.sh https://github.com/matsujirushi/seeed-linux-dtoverlays/blob/main/doc/reTerminal_RasPiOS.md

The script reTerminal2.sh looks good. Can you try to merge it into the reTerminal.sh.It would be better for user to use it.

bigbearishappy, 2024-10-15:
@matsujirushi 
Is there anything new about this issue now?

closed 2024-10-30 by Lesords