Seeed-Studio/seeed-linux-dtoverlays #90: Reterminal DM how to SSD boot?

donserdal, 2024-02-11:
**Describe the bug**
The documentation on the hardware guide is missing the instructions needed to boot from the NVMe Expansion. I have followed both:

https://hagensieker.com/2022/04/17/booting-a-raspberry-pi-cm4-from-an-nvme/
https://www.jeffgeerling.com/blog/2021/raspberry-pi-can-boot-nvme-ssds-now

But the reterminal keeps booting from the eMMC. 

Hardware used:
 - Reterminal DM
 - SSD expansion board with Dual NIC (https://www.seeedstudio.com/reTerminal-DM-Dual-LAN-and-SSD-extension-p-5766.html?) 
 - Patriot 256GB NVMe M.2 SSD (P300P256GM28)

**To Reproduce**
Steps to reproduce the behavior:
Go to https://wiki.seeedstudio.com/reterminal-dm-hardware-guide/#install-ssd 

**Expected behavior**
Booting from NVME

**Additional context**
Current bootloader settings:
```
user@reterminal:~ $ rpi-eeprom-config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
BOOT_ORDER=0xf516

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
```

```
user@reterminal:~ $ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0      179:0    0  29.1G  0 disk
├─mmcblk0p1  179:1    0   256M  0 part /boot
└─mmcblk0p2  179:2    0  28.9G  0 part /
mmcblk0boot0 179:32   0     4M  1 disk
mmcblk0boot1 179:64   0     4M  1 disk
nvme0n1      259:0    0 238.5G  0 disk
├─nvme0n1p1  259:1    0   512M  0 part
└─nvme0n1p2  259:2    0   4.9G  0 part
```

bigbearishappy, 2024-02-19:
Have you check the version of your bootloader? 
Here is the detail [information](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#required-software).

bigbearishappy, 2024-02-19:
Or you can have a try to format the emmc, and try to boot from ssd again~

peterpanstechland, 2024-02-23:
Hi, I have tried to reproduce your eeprom boot order setup with the reTerminal DM and external NVMe M.2 SSD with the SSD expansion board with Dual NIC. I could boot into with SSD without any issue.
 
[image]
[image]

So I guess what you need to do is making sure the "root=PARTUUID=" in the /boot/cmdline.txt file is aligned with the UUID from the NVMe like shown below:

[image]
[image]

Please let us know how it goes.

bigbearishappy, 2024-03-19:
Closing the issue.Feel free to reopen this If you have any further problem.

closed 2024-03-19 by bigbearishappy