Seeed-Studio/seeed-linux-dtoverlays #91: Build error with kernel 6.6.20+rpt-rpi-2712

TB-archIT, 2024-03-14:
**Describe the bug**
Compiling the reTerminal display driver is failing with newest kernel on RaspiOS 64-bit.

**To Reproduce**
Steps to reproduce the behavior:
1. Flashing `2023-12-11-raspios-bookworm-arm64-lite.img.xz`
2. Connected as 'root'
3. `git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays`
4. `cd seeed-linux-dtoverlays` and `./scripts/reTerminal.sh --device reTerminal-plus`
5. Eroor as following (compete output commandline):
```
root@raspberrypi:~# git clone --depth 1 https://github.com/Seeed-Studio/seeed-linux-dtoverlays
Cloning into 'seeed-linux-dtoverlays'...
remote: Enumerating objects: 873, done.
remote: Counting objects: 100% (873/873), done.
remote: Compressing objects: 100% (715/715), done.
remote: Total 873 (delta 160), reused 707 (delta 127), pack-reused 0
Receiving objects: 100% (873/873), 2.17 MiB | 6.91 MiB/s, done.
Resolving deltas: 100% (160/160), done.
root@raspberrypi:~# cd seeed-linux-dtoverlays/
root@raspberrypi:~/seeed-linux-dtoverlays# ./scripts/reTerminal.sh --device reTerminal-plus

### Current Debian version is bookworm or later

### Install required tool packages
Get:1 http://deb.debian.org/debian bookworm InRelease [151 kB]
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:4 http://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]
Get:5 http://deb.debian.org/debian bookworm/main arm64 Packages [8,685 kB]
Get:6 http://deb.debian.org/debian bookworm/main armhf Packages [8,498 kB]
Get:7 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [372 kB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [362 kB]           
Get:9 http://deb.debian.org/debian bookworm/main Translation-en [6,109 kB]              
Get:10 http://deb.debian.org/debian-security bookworm-security/main armhf Packages [140 kB]
Get:11 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [142 kB]
Get:12 http://deb.debian.org/debian-security bookworm-security/main Translation-en [86.7 kB]
Get:13 http://deb.debian.org/debian bookworm-updates/main armhf Packages [12.1 kB]
Get:14 http://deb.debian.org/debian bookworm-updates/main arm64 Packages [12.5 kB]
Get:15 http://deb.debian.org/debian bookworm-updates/main Translation-en [13.8 kB]
Get:16 http://deb.debian.org/debian bookworm-updates/contrib arm64 Packages [768 B]
Get:17 http://deb.debian.org/debian bookworm-updates/contrib Translation-en [408 B]
Get:18 http://deb.debian.org/debian bookworm-updates/non-free armhf Packages [492 B]
Get:19 http://deb.debian.org/debian bookworm-updates/non-free arm64 Packages [12.0 kB]
Get:20 http://deb.debian.org/debian bookworm-updates/non-free Translation-en [7,744 B]
Get:21 http://deb.debian.org/debian bookworm-updates/non-free-firmware arm64 Packages [616 B]
Get:22 http://deb.debian.org/debian bookworm-updates/non-free-firmware Translation-en [384 B]
Fetched 24.7 MB in 8s (3,079 kB/s)                                                                                 
Reading package lists... Done
N: Repository 'http://deb.debian.org/debian bookworm InRelease' changed its 'Version' value from '12.4' to '12.5'
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  menu
Recommended packages:
  linux-headers-generic | linux-headers-686-pae | linux-headers-amd64 | linux-headers
The following NEW packages will be installed:
  dkms
0 upgraded, 1 newly installed, 0 to remove and 60 not upgraded.
Need to get 48.7 kB of archives.
After this operation, 190 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 dkms all 3.0.10-8+deb12u1 [48.7 kB]
Fetched 48.7 kB in 0s (1,226 kB/s)
Selecting previously unselected package dkms.
(Reading database ... 56492 files and directories currently installed.)
Preparing to unpack .../dkms_3.0.10-8+deb12u1_all.deb ...
Unpacking dkms (3.0.10-8+deb12u1) ...
Setting up dkms (3.0.10-8+deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...
[... 131 lines cut ...]
make[1]: Leaving directory '/usr/src/linux-headers-6
```

**Expected behavior**
Finishing without error

**Additional context**
Here the content of `/var/lib/dkms/ltr30x/0.1/build/make.log`:
```
DKMS make.log for ltr30x-0.1 for kernel 6.6.20+rpt-rpi-2712 (aarch64)
Thu 14 Mar 08:04:43 CET 2024
make -C /lib/modules/6.6.20+rpt-rpi-2712/build M=/var/lib/dkms/ltr30x/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-2712'
  CC [M]  /var/lib/dkms/ltr30x/0.1/build/ltr30x.o
/var/lib/dkms/ltr30x/0.1/build/ltr30x.c:1614:19: error: initialization of ‘int (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *, const struct i2c_device_id *)’ [-Werror=incompatible-pointer-types]
 1614 |         .probe  = ltr501_probe,
      |                   ^~~~~~~~~~~~
/var/lib/dkms/ltr30x/0.1/build/ltr30x.c:1614:19: note: (near initialization for ‘ltr501_driver.probe’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /var/lib/dkms/ltr30x/0.1/build/ltr30x.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /var/lib/dkms/ltr30x/0.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-2712'
make: *** [Makefile:13: all] Error 2
```

bigbearishappy, 2024-03-18:
Because of the kernel upgrade. We need to upgrade the driver code in here too.
This problem is fixed in this [commit](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/addf84413580bb9f204221eec95d906fff6f3464).
Please pull the latest code and test it .

TB-archIT, 2024-03-18:
Yes, dkms is working with it. Thx

closed 2024-03-19 by bigbearishappy