Seeed-Studio/seeed-linux-dtoverlays #92: [reTerminal DM] Build error with kernel 6.6.20-1+rpt1

matsujirushi, 2024-03-15:
### Reprocedure

1. Flash 2024-03-12-raspios-bookworm-arm64.img to reTerminal DM.
2. Install reTerminal drivers after flashing new Raspbian OS ([here](https://wiki.seeedstudio.com/reterminal-dm-flash-OS/#install-reterminal-drivers-after-flashing-new-raspbian-os))
   ```
   git clone https://github.com/Seeed-Studio/seeed-linux-dtoverlays
   cd seeed-linux-dtoverlays
   sudo ./scripts/reTerminal.sh --device reTerminal-plus
   ```

### Error message

```
make -j4 KERNELRELEASE=6.6.20+rpt-rpi-v8 all KVERSION=6.6.20+rpt-rpi-v8....(bad exit status: 2)
Error! Bad return status for module build on kernel: 6.6.20+rpt-rpi-v8 (aarch64)
Consult /var/lib/dkms/ch34x/0.1/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
DKMS make.log for ch34x-0.1 for kernel 6.6.20+rpt-rpi-v8 (aarch64)
Fri 15 Mar 13:15:26 JST 2024
make -C /lib/modules/6.6.20+rpt-rpi-v8/build M=/var/lib/dkms/ch34x/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
  CC [M]  /var/lib/dkms/ch34x/0.1/build/ch343.o
/var/lib/dkms/ch34x/0.1/build/ch343.c:1811:18: error: initialization of ‘ssize_t (*)(struct tty_struct *, const u8 *, size_t)’ {aka ‘long int (*)(struct tty_struct *, const unsigned char *, long unsigned int)’} from incompatible pointer type ‘int (*)(struct tty_struct *, const u8 *, size_t)’ {aka ‘int (*)(struct tty_struct *, const unsigned char *, long unsigned int)’} [-Werror=incompatible-pointer-types]
 1811 |         .write = ch343_tty_write,
      |                  ^~~~~~~~~~~~~~~
/var/lib/dkms/ch34x/0.1/build/ch343.c:1811:18: note: (near initialization for ‘ch343_ops.write’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/scripts/Makefile.build:248: /var/lib/dkms/ch34x/0.1/build/ch343.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:1938: /var/lib/dkms/ch34x/0.1/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.6.20+rpt-common-rpi/Makefile:246: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.6.20+rpt-rpi-v8'
make: *** [Makefile:13: all] Error 2
```

### seeed-linux-dtoverlays

```
commit 78ed7e7d2e3df8b6369a38ae655242b3fef737c8 (HEAD -> master, origin/master, origin/HEAD)
Merge: 4ff45a3 d6a2043
Author: bigbearishappy <953308023@qq.com>
Date:   Thu Mar 14 10:35:00 2024 +0000

    Merge remote-tracking branch 'upstream/master'
```

kustere, 2024-03-17:
yup have the same issue. Was trying to reinstall after latest update to 6.2.20 and after update display no longer turns on for the ReTerminal DM. Keen to know of a fix.

kustere, 2024-03-17:
It appears to be the wrong variable type on Line 731 to ch343.c

According to this: https://github.com/WCHSoftGroup/ch343ser_linux/blob/main/driver/ch343.c

the line 731 should be 
`static ssize_t ch343_tty_write(struct tty_struct *tty, const u8 *buf, size_t count);`

instead of the current:
`static int ch343_tty_write(struct tty_struct *tty, const u8 *buf, size_t count)`

therefore `ssize_t` rather than `int`

I am no c coder just doing some basic pattern matching here. Not sure if there is additional items as I notice also that the number of lines in the original linked above is different to the one in this library.

kustere, 2024-03-17:
Although I was able to get the scripts to compile and run this still does not fix the screen issue on the reTerminal DM running 6.2.20 kernel :(

bigbearishappy, 2024-03-18:
Fix commit is [here](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/25bcb462062f54f9c09a9267a29d601b0c76c382).

bigbearishappy, 2024-03-18:
> Although I was able to get the scripts to compile and run this still does not fix the screen issue on the reTerminal DM running 6.2.20 kernel :(

Can you provide the detail of your problem and open a new issue?
Thank you~

matsujirushi, 2024-03-18:
Hi @bigbearishappy ,

I have tried the latest drivers.
The build error is fixed, but screen is not displayed.
It appears that there is an error in the backlight.

Are you able to display screen there?

## dmesg

```
[    2.813589] OF: /lcd_backlight: #pwm-cells = 3 found 2
[    2.813630] of_pwm_get(): can't parse "pwms" property
[    2.813643] pwm-backlight lcd_backlight: unable to request PWM
[    2.813654] pwm-backlight: probe of lcd_backlight failed with error -22
```

[dmesg.txt](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/files/14643556/dmesg.txt)

matsujirushi, 2024-03-19:
### Info.

pwm-bcm2835 supports PWM polarity
https://github.com/raspberrypi/linux/commit/81b875892022181d1d8f9a04416403b307b66754

### Testing

With the following modifications, the LCD is now displayed.

[image]

bigbearishappy, 2024-03-19:
> ### Info.
> pwm-bcm2835 supports PWM polarity [raspberrypi/linux@81b8758](https://github.com/raspberrypi/linux/commit/81b875892022181d1d8f9a04416403b307b66754)
> 
> ### Testing
> With the following modifications, the LCD is now displayed.
> 
> ![image](https://private-user-images.githubusercontent.com/8079404/313880588-19ec5bf7-cfc2-4d2f-8115-a6b5f418d032.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA4MTM1MjYsIm5iZiI6MTcxMDgxMzIyNiwicGF0aCI6Ii84MDc5NDA0LzMxMzg4MDU4OC0xOWVjNWJmNy1jZmMyLTRkMmYtODExNS1hNmI1ZjQxOGQwMzIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDMxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAzMTlUMDE1MzQ2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODFmZTc1NDYxMzU4OGQwOTA0OTBhNjNkOGIyYTZiNmFjOTIzM2ZmMDNlMDRkNDgzNGI0N2FhZGQ0OGZlYTEzOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.veARdeVrhlanVbMRf5X2FK1xO4k7i2amY4MW1HliMjE)

Actually, I don't try it. Can you push your fix of this problem to us?

matsujirushi, 2024-03-19:
Hi @bigbearishappy ,

I think you need to determine the kernel version in the install script and switch the device tree overlay.
Since reTerminal.sh is complicated, I would like to see this part addressed there.

bigbearishappy, 2024-03-19:
> Hi @bigbearishappy ,
> 
> I think you need to determine the kernel version in the install script and switch the device tree overlay. Since reTerminal.sh is complicated, I would like to see this part addressed there.

I think we can have a try on passing a parameter to the dtoverlay of reTerminal-plus. Just like the code in [here](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/blob/8a9422d7851567dc542763851aa3723bc96e8fd8/overlays/rpi/reTerminal-overlay.dts#L247). And then config the dtoverlay  with different raspbian OS version in reTerminal.sh.

TB-archIT, 2024-03-19:
Stucking with the same error here...

```
dmesg | grep lcd
[    0.040372] OF: /lcd_backlight: #pwm-cells = 3 found 2
[    2.597359] OF: /lcd_backlight: #pwm-cells = 3 found 2
[    2.597408] pwm-backlight lcd_backlight: unable to request PWM
[    2.597419] pwm-backlight: probe of lcd_backlight failed with error -22
```
Is there something for testing already?

matsujirushi, 2024-03-19:
Hi @TB-archIT ,

> Is there something for testing already?

Try manually rewriting reTerminal-plus-overlay.dts and re-running reTerminal.sh. See [here](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/92#issuecomment-2005546304).

TB-archIT, 2024-03-20:
Thx, is working again.

kustere, 2024-03-20:
this seems to fix the ability to connect remotely to the display via RealVNC  but does not power on the backlight for me so the actual physical display still remains black. I can see that the touchscreen is still active though as I can see the mouse move through the VNC session now.

Here is the output from dmesg:

<details>

```
kustere@eggpi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.20+rpt-rpi-v8 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi Compute Module 4 Rev 1.1
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000000dc00000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000dc00000..0x000000002dbfffff (524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef642c0..0x000000003ef646bf (1 KiB) nomap non-reusable nvram@1
[    0.000000] OF: reserved mem: 0x000000003ef64700..0x000000003ef6489d (0 KiB) nomap non-reusable nvram@0
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 0, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85672 r8192 d29016 u122880
[    0.000000] pcpu-alloc: s85672 r8192 d29016 u122880 alloc=30*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v3a
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[    0.000000] CPU features: detected: Kernel page table isolation (KPTI)
[    0.000000] CPU features: detected: ARM erratum 1742098
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 snd_bcm2835.enable_hdmi=0 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=E4:5F:01:F3:B4:CB vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=3632e9ab-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles systemd.unified_cgroup_hierarchy=false apparmor=1 security=apparmor cfg80211.ieee80211_regdom=AU logo.nologo vt.global_cursor_default=0 console=tty3 loglevel=0
[    0.000000] Unknown kernel command line parameters "splash", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] mem auto-init: stack:all(zero), heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7451552K/8245248K available (13376K kernel code, 2210K rwdata, 4260K rodata, 4864K init, 1083K bss, 269408K reserved, 524288K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 43116 entries in 169 pages
[    0.000000] ftrace: allocated 169 pages with 4 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[... 573 lines cut ...]
[   83.893049] hassio: port 8(veth4ed34e7) entered forwarding state
[   84.587848] hassio: port 9(veth8e8e9d4) entered blocking state
[   84.587866] hassio: port 9(veth8e8e9d4) entered disabled state
[... cut ...]
[   87.546295] hassio: port 10(vethb0167e4) entered forwarding state
[   88.891802] hassio: port 11(veth37a4202) entered blocking state
[   88.891830] hassio: port 11(veth37a4202) entered disabled state
[   88.891870] veth37a4202: entered allmulticast mode
[   88.892099] veth37a4202: entered promiscuous mode
[   90.300835] eth0: renamed from vethdee39f5
[   90.320975] hassio: port 11(veth37a4202) entered blocking state
[   90.321015] hassio: port 11(veth37a4202) entered forwarding state
```

</details>

TB-archIT, 2024-03-20:
I have also a following problem that worked with the old kernel before and now not anymore...

My application (java running with xorg) does starting, but rotation of the screen does not work. display is in portrait but landscape is needed.

Tried both ways of
https://wiki.seeedstudio.com/Incorrect_screen_orientation_on_RPiOS_Bullseye/#raspberry-pi-os-bookworm
None of them does work. Before one of both does work (I don't know wheter ;-))

TB-archIT, 2024-03-20:
I'm quite irritated now... Whereas the link above describing a physical resolution with 1280x720 my reTerminal DM says...

```
xrandr
Screen 0: minimum 320 x 200, current 800 x 1280, maximum 7680 x 7680
HDMI-1 disconnected primary (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
DSI-1 connected 800x1280+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   800x1280      60.00*+

```

matsujirushi, 2024-03-21:
Hi @kustere ,

```
[    7.407942] mipi_dsi: Initialize kernel module
[    7.407956] mipi_dsi: (i2c_md_init) Add I2C driver
[    7.408101] mipi_dsi: Probe I2C driver
[    7.408106] mipi_dsi: (i2c_md_probe) Start
[    7.408185] mipi_dsi 1-0045: I2C read id failed: -5
[    7.408379] mipi_dsi: (i2c_md_init) Register MIPI-DSI driver
```

mipi_dsi driver is for reTerminal. (NOT reTerminal DM)
Please check your config.txt file.

matsujirushi, 2024-03-21:
Hi @TB-archIT ,
I don't know much about it, but I suspect the distribution may be affecting it.
This may be helpful.
https://github.com/matsujirushi/seeed-linux-dtoverlays/blob/82d72b11814ef59748014a889d6065a27e067193/scripts/reTerminal2.sh#L222

bigbearishappy, 2024-03-21:
> I'm quite irritated now... Whereas the link above describing a physical resolution with 1280x720 my reTerminal DM says...
>
> ```
> xrandr
> Screen 0: minimum 320 x 200, current 800 x 1280, maximum 7680 x [...]

Actually, 720x1280 is the lcd resolution for reTerminal while 800x1280 is for reTerminal DM.
You can have a try to append:
```
[output:DSI-1]
mode = 720x1280@60
transform = 270
```
to `~/.config/wayfire.ini`

kustere, 2024-03-21:
> Hi @kustere ,
>
> ```
> [    7.407942] mipi_dsi: Initialize kernel module
> [    7.407956] mipi_dsi: (i2c_md_init) Add I2C driver
> [    7.408101] mipi_dsi: Probe I2C driver
> [    7.408106] mipi_dsi: [...]

Thanks. For some reason there was multiple lines in config.txt with `dtoverlay=reTerminal,tp_rotate=0` in there so I removed the duplicates and commented out the last one leaving in the `dtoverlay=reTerminal-plus` at the end of the file and now the display is working perfectly.

Thanks for the help troubleshooting.

matsujirushi, 2024-03-22:
Create new issue and close this issue.

closed 2024-03-22 by matsujirushi