Seeed-Studio/seeed-linux-dtoverlays #95: reTerminal: Build error occurs on ubuntu23.10

mnakai3, 2024-03-25:
I installed ubuntu23.10 (kernel 6.5) on reTerminal and tried to install the driver, but an error occurs.

This commit https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/c22019add688ca6fd3f5512746696b69c6ac5521
 seems to have fixed it, but wouldn't it be better to change kernel version 6.3.0 instead of 6.6.20?

See below.
https://github.com/torvalds/linux/commit/03c835f498b540087244a6757e87dfe7ef10999b
https://github.com/torvalds/linux/commit/5ea6b17027810ffbdb5bea7d0a2b1d312dd1021c

## log
```
pi@reTerminal:~$ uname -a
Linux reTerminal 6.5.0-1005-raspi #7-Ubuntu SMP PREEMPT_DYNAMIC Sun Oct  8 08:06:18 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
pi@reTerminal:~$ dpkg -l linux-image\*
要望=(U)不明/(I)インストール/(R)削除/(P)完全削除/(H)保持
| 状態=(N)無/(I)インストール済/(C)設定/(U)展開/(F)設定失敗/(H)半インストール/(W)トリガ待ち/(T)トリガ保留
|/ エラー?=(空欄)無/(R)要再インストール (状態,エラーの大文字=異常)
||/ 名前                                  バージョン   アーキテクチ 説明
+++-=====================================-============-============-=================================================
un  linux-image                           <なし>       <なし>       (説明 (description) がありません)
ii  linux-image-6.5.0-1005-raspi          6.5.0-1005.7 arm64        Linux kernel image for version 6.5.0 on ARMv8 SMP
ii  linux-image-raspi                     6.5.0.1005.6 arm64        Raspberry Pi Linux kernel image
un  linux-image-unsigned-6.5.0-1005-raspi <なし>       <なし>       (説明 (description) がありません)

pi@reTerminal:~$ cat /etc/debian_version
trixie/sid
pi@reTerminal:~$ cat /etc/issue
Ubuntu 23.10 \n \l

pi@reTerminal:~$

pi@reTerminal:~/seeed-linux-dtoverlays$ sudo ./scripts/reTerminal.sh
./scripts/reTerminal.sh: line 564: [: trixie/sid: integer expression expected

### Current Debian version is bookworm or later

### Install required tool packages
ヒット:1 http://ports.ubuntu.com/ubuntu-ports mantic InRelease
ヒット:2 http://ports.ubuntu.com/ubuntu-ports mantic-updates InRelease
ヒット:3 http://ports.ubuntu.com/ubuntu-ports mantic-security InRelease
パッケージリストを読み込んでいます... 完了
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています... 完了
状態情報を読み取っています... 完了
dkms はすでに最新バージョン (3.0.11-1ubuntu10) です。
アップグレード: 0 個、新規インストール: 0 個、削除: 0 個、保留: 207 個。

### Will compile with the latest kernel...

### Sync kernel and userland

### Uninstall previous dkms module
Deleting module mipi_dsi-0.1 completely from the DKMS tree.

### Install required kernel package
./scripts/reTerminal.sh: line 42: [: trixie/sid: integer expression expected
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています... 完了
状態情報を読み取っています... 完了
linux-raspi はすでに最新バージョン (6.5.0.1012.13) です。
linux-headers-raspi はすでに最新バージョン (6.5.0.1012.13) です。
linux-image-raspi はすでに最新バージョン (6.5.0.1012.13) です。
アップグレード: 0 個、新規インストール: 0 個、削除: 0 個、保留: 207 個。
./scripts/reTerminal.sh: line 42: [: trixie/sid: integer expression expected
KBUILD: /lib/modules/6.5.0-1012-raspi/build
./scripts/reTerminal.sh: line 42: [: trixie/sid: integer expression expected
Sign command: /usr/bin/kmodsign
Binary update-secureboot-policy not found, modules won't be signed
Creating symlink /var/lib/dkms/mipi_dsi/0.1/source -> /usr/src/mipi_dsi-0.1

Building module:
Cleaning build area...(bad exit status: 2)
make -j4 KERNELRELEASE=6.5.0-1012-raspi all KVERSION=6.5.0-1012-raspi.....(bad exit status: 2)
ERROR (dkms apport): binary package for mipi_dsi: 0.1 not found
Error! Bad return status for module build on kernel: 6.5.0-1012-raspi (aarch64)
Consult /var/lib/dkms/mipi_dsi/0.1/build/make.log for more information.
Can't compile with this kernel, aborting
Please try to compile with the option --compat-kernel
DKMS make.log for mipi_dsi-0.1 for kernel 6.5.0-1012-raspi (aarch64)
2024年  3月 19日 火曜日 14:09:20 JST
make -C /lib/modules/6.5.0-1012-raspi/build M=/var/lib/dkms/mipi_dsi/0.1/build modules
make[1]: Entering directory '/usr/src/linux-headers-6.5.0-1012-raspi'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/touch_panel.o
  CC [M]  /var/lib/dkms/mipi_dsi/0.1/build/panel-ili9881d.o
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: In function ‘i2c_md_probe’:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:385:19: error: ‘struct drm_panel’ has no member named ‘prepare_upstream_first’; did you mean ‘prepare_prev_first’?
  385 |         md->panel.prepare_upstream_first = true;
      |                   ^~~~~~~~~~~~~~~~~~~~~~
      |                   prepare_prev_first
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c: At top level:
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:467:18: error: initialization of ‘int (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *, const struct i2c_device_id *)’ [-Werror=incompatible-pointer-types]
  467 |         .probe = i2c_md_probe,
      |                  ^~~~~~~~~~~~
/var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.c:467:18: note: (near initialization for ‘i2c_md_driver.<anonymous>.probe’)
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:251: /var/lib/dkms/mipi_dsi/0.1/build/mipi_dsi_drv.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/usr/src/linux-headers-6.5.0-1012-raspi/Makefile:2039: /var/lib/dkms/mipi_dsi/0.1/build] Error 2
make[1]: *** [Makefile:234: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.5.0-1012-raspi'
make: *** [Makefile:15: all] Error 2
pi@reTerminal:~/seeed-linux-dtoverlays$
```

## ubuntu23.10
[image]

bigbearishappy, 2024-03-26:
It looks good. We make it to 6.6.20 because the maintain work is based on the latest raspbian OS. 
It would be nice if you can create a PR which based on your idea.

Thank you in advance.

mnakai3, 2024-03-27:
If you install the driver and set tp_rotate=1 in config.txt and reboot, the touch screen will work properly.

[image]

You can also rotate the display.

[image]

bigbearishappy, 2024-03-28:
Since the issue is fixed. Closing the issue.

closed 2024-03-28 by bigbearishappy