Seeed-Studio/seeed-linux-dtoverlays #97: Stuck in rotate display from portrait to landscape

TB-archIT, 2024-03-26:
Hi,

after #92 is solved now I'm stuck in rotating the display of a "reTerminal DM".

The system:
- RaspiOS Bookworm Lite 64-Bit with all updates
- installed openjdk-jre and xorg-xserver incl. dependencies
- seeed-linux-dtoverlays for display

Starting of java application (PaperCut Release Station) on login of a special user 'papercut' via '.profile' does work with:
`xinit ~/release/pc-release-linux.sh -- vt$XDG_VTNR ;exit`

[image]

`~/.config/wayfire.ini` looks like:
```
[output:DSI-1]
mode = 800x1280@60
transform = 270
```
With `xrandr --output DSI-1 --rotate right` the display output is rotated, but cropped...

[image]

Here the output from `/var/log/Xorg.0.log`:
```
[    27.031] 
X.Org X Server 1.21.1.7
X Protocol Version 11, Revision 0
[    27.032] Current Operating System: Linux plot-pc-rs-1140 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64
[    27.032] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 snd_bcm2835.enable_hdmi=0  smsc95xx.macaddr=E4:5F:01:EC:49:81 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=886e7e07-02 rootfstype=ext4 fsck.repair=yes rootwait logo.nologo vt.global_cursor_default=0 console=tty3 loglevel=0
[    27.035] xorg-server 2:21.1.7-3+rpt2+deb12u5 (https://www.debian.org/support) 
[    27.036] Current version of pixman: 0.42.2
[    27.039] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    27.039] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    27.044] (==) Log file: "/var/log/Xorg.0.log", Time: Mon Mar 25 14:28:18 2024
[    27.046] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    27.048] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    27.050] (==) No Layout section.  Using the first Screen section.
[    27.050] (==) No screen section available. Using defaults.
[    27.050] (**) |-->Screen "Default Screen Section" (0)
[    27.050] (**) |   |-->Monitor "<default monitor>"
[    27.051] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    27.051] (**) Option "DontVTSwitch" "true"
[    27.051] (==) Automatically adding devices
[    27.051] (==) Automatically enabling devices
[    27.051] (==) Automatically adding GPU devices
[    27.051] (==) Automatically binding GPU devices
[    27.053] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    27.056] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    27.056] 	Entry deleted from font path.
[    27.056] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    27.056] 	Entry deleted from font path.
[    27.056] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    27.056] 	Entry deleted from font path.
[    27.056] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    27.056] 	Entry deleted from font path.
[    27.056] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    27.056] 	Entry deleted from font path.
[    27.056] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    27.056] 	Entry deleted from font path.
[    27.056] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    27.056] (==) ModulePath set to "/usr/lib/xorg/modules"
[    27.056] (II) The server relies on udev to provide the list of input devices.
[... 245 lines cut ...]
[    29.114] (**) Option "xkb_model" "pc105"
[    29.114] (**) Option "xkb_layout" "de"
[    29.114] (**) Option "xkb_options" "lv3:ralt_switch"
[    29.115] (**) vc4-hdmi-1: Applying InputClass "libinput keyboard catchall"
[    29.115] (II) Using input driver 'libinput' for 'vc4-hdmi-1'
[    29.115] (II) systemd-logind: returning pre-existing fd for /dev/input/event4 13:68
[    29.115] (**) vc4-hdmi-1: always reports core events
[    29.115] (**) Option "Device" "/dev/input/event4"
[    29.115] (II) libinput: vc4-hdmi-1: is a virtual subdevice
[    29.115] (**) Option "config_info" "udev:/sys/devices/platform/soc/fef05700.hdmi/rc/rc1/input5/event4"
[    29.115] (II) XINPUT: Adding extended input device "vc4-hdmi-1" (type: KEYBOARD, id 12)
[    29.115] (**) Option "xkb_model" "pc105"
[    29.115] (**) Option "xkb_layout" "de"
[    29.115] (**) Option "xkb_options" "lv3:ralt_switch"
[    36.264] (II) modeset(0): Disabling kernel dirty updates, not required.
[  7340.900] (II) modeset(0): Allocate new frame buffer 1280x800 stride
```
Last line should come from the `xrandr` command!?

I have no clue how to get the display rotated AND not cropped.

Any hints or suggestions?

Regards
Thomas

bigbearishappy, 2024-03-27:
It seems that you java application don't have the function of adaptive graphical interface. Or you should config you app to display on landscape orientation.

TB-archIT, 2024-03-27:
@bigbearishappy: thanx for your reply.

No, I don't think that the reason is to find in the java application. I would say, it has something to do with the overlay, because before the kernel upgrade to 6.6.20 it worked like a charme with the same system (RaspiOS Bookworm lite 64-bit).

bigbearishappy, 2024-03-28:
As you said below. Your app is working based on the xorg-xserver.
But according to the [link](https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/) here.
Raspbian OS bookworm had moved to using wayland rather than X11.

And the config file `~/.config/wayfire.ini` is working for wayland.

Maybe you can have a try to make the app work based on wayland?

TB-archIT, 2024-04-02:
Yes, RaspiOS Bookworm has moved to wayland. `~/.config/wayfire.ini` is configured with:
```
[output:DSI-1]
mode = 800x1280@60
transform = 270
```
You can use Xorg in parallel.

And I would still say, that a change in the overlay from kernel 6.6.20 causes this changed behavior. I repeat, before that the same java application did run in landscape mode on RaspiOS Bookworm (!) with Xorg.

bigbearishappy, 2024-04-07:
> Yes, RaspiOS Bookworm has moved to wayland. `~/.config/wayfire.ini` is configured with:
>
> ```
> [output:DSI-1]
> mode = 800x1280@60
> transform = 270
> ```
>
> You can use Xorg in parallel.
>
> And I would still [...]

We will have a try to reproduce you problem and try to fix it.

TB-archIT, 2024-04-08:
That would be great. I will test it ASAP...

franzos, 2024-04-14:
Just FYI: On our side this seems to work on

- OS: `6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07)` (Latest Raspbian)
- Auto-login with LightDM (user `default`) to openbox

By default the display comes-up like @TB-archIT, but I can rotate it with xrandr, and the output (and touch-input) appear to work as expected; Here's how I turn the screen over SSH:

```bash
$ export XAUTHORITY=/home/default/.Xauthority
$ DISPLAY=:0.0 xrandr
Screen 0: minimum 320 x 200, current 800 x 1280, maximum 7680 x 7680
HDMI-1 disconnected primary (normal left inverted right x axis y axis)
HDMI-2 disconnected (normal left inverted right x axis y axis)
DSI-1 connected 800x1280+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   800x1280      60.00*+
$ DISPLAY=:0.0 xrandr --output DSI-1 --rotate right
```

TB-archIT, 2024-04-23:
Thanks for your effort, BBear.

But I would say, that you only can use RaspberryPiReleaseStation if a running PaperCut-Server is found. Otherwise starting of this app will not work.

Meanwhile I used a ComfilePi for the ReleaseStation. On the reTerminal DM now a Node-RED-application will be installed next.

bigbearishappy, 2024-04-24:
> Thanks for your effort, BBear.
>
> But I would say, that you only can use RaspberryPiReleaseStation if a running PaperCut-Server is found. Otherwise starting of this app will not work.
>
> Meanwhile I [...]

Sorry for not solving your problem with reTerminal DM.
So is there any further work I can do for you on this issue?

TB-archIT, 2024-04-24:
I think no, therefore I'm closing this issue

closed 2024-04-24 by TB-archIT