Seeed-Studio/seeed-linux-dtoverlays #100: [reTerminal] Re-install the driver will upgrade the kernel

cyaoplus, 2024-04-07:
Hi,
When I power on the reTerminal, the buzzer will turn on automatically which is annoying (and I couldn't find /sys/class/leds/usr_buzzer to turn off it as descibed in [wiki](https://wiki.seeedstudio.com/reTerminal/) ). I decide to manually install the driver again with ./scripts/reTerminal.sh. However, this shell will automatically upgrade my kernel from 5.15.32 to 6.2.21 which I don't want. Is there any tips that I can keep my original kernel vesion?
Thanks in advance

bigbearishappy, 2024-04-07:
It should work as you wish with command `./scripts/reTerminal.sh --keep-kernel`

franzos, 2024-04-12:
@bigbearishappy why you guys don't update the docs? It must have happened to hundreds of people, and the only way to find out that there's the option to `--keep-kernel` is by reviewing the bash script.

- Your instructions don't mention `--keep-kernel` even though it's **required** in 9/10 situation
- OR at least default to KEEP the kernel (why is upgrade the default!?)

Of course you could also, simply support the latest kernel ...

bigbearishappy, 2024-04-15:
> @bigbearishappy why you guys don't update the docs? It must have happened to hundreds of people, and the only way to find out that there's the option to `--keep-kernel` is by reviewing the bash script.
> 
> * Your instructions don't mention `--keep-kernel` even though it's **required** in 9/10 situation
> * OR at least default to KEEP the kernel (why is upgrade the default!?)
> 
> EDIT: Well, exuse me; It seems the command is not actually supported on DM, which is worse:
> 
> ```shell
> ./scripts/reTerminal.sh --device reTerminal-plus --keep-kernel
> 
> ### Current Debian version is bookworm or later
> 
> ### Install required tool packages
> Hit:1 http://deb.debian.org/debian bookworm InRelease
> Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
> Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
> Get:4 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [147 kB]
> Get:5 http://deb.debian.org/debian-security bookworm-security/main armhf Packages [144 kB]
> Get:6 http://deb.debian.org/debian-security bookworm-security/main Translation-en [90.3 kB]
> Get:7 http://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]
> Get:8 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [384 kB]
> Get:9 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [374 kB]
> Fetched 1,267 kB in 3s (372 kB/s)
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> dkms is already the newest version (3.0.10-8+deb12u1).
> 0 upgraded, 0 newly installed, 0 to remove and 234 not upgraded.
> 
> ### Don't support option --keep-kernel currently.
> 
> ### Pls remove --keep-kernel in your command.
> ```
> 
> Edit: Looks like you're working on this: [#86 (comment)](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/86#issuecomment-1897845173)

Actually, For the raspbian OS before bookworm.We support option `--keep-kernel`.
We need to do some work for bookworm to support it.

bigbearishappy, 2024-04-15:
@franzos 
why we upgrade kernel by default?
Because we always want our code to work with the latest raspbian OS or some other OS like it, in which way our code can move forward with the latest OS.

franzos, 2024-04-15:
> Because we always want our code to work with the latest raspbian OS or some other OS like it,

I understand why you're doing this; I'm just saying that `--keep-kernel` should be properly documented, and probably default. I think it's fair to assume that not everyone who installs that driver, also wants their kernel upgraded - because either they are using a stable kernel, or a new kernel introduces breaking changes.

So either:
- Document `--keep-kernel` on the wiki, in the "driver" section
- OR prompt the user, if they want to update the kernel in the script
- OR default to not upgrading (and failing, if it doesn't work without)

bigbearishappy, 2024-04-16:
> I understand why you're doing this; I'm just saying that `--keep-kernel` should be properly documented, and probably default. I think it's fair to assume that not everyone who installs that driver, [...]

Thank you for your advice. We will update the document on seeed's wiki.

cyaoplus, 2024-04-18:
@bigbearishappy 
> It should work as you wish with command `./scripts/reTerminal.sh --keep-kernel`

 Thank you for your reply! I have tried the command but report error. It seems still trying to search and install the kernel? 
The searched name is kind of mismatched though... raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb v.s. raspberrypi-kernel-headers_1.20220331-1_arm64.deb on raspi website

```
pi@raspberrypi:~/seeed-linux-dtoverlays $ sudo ./scripts/reTerminal.sh --keep-kernel
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

### Current Debian version is bullseye or earlier

### Install required tool packages
Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease
Hit:3 http://security.debian.org/debian-security bullseye-security InRelease
Hit:4 http://deb.debian.org/debian bullseye-updates InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (2.8.4-3).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 325 not upgraded.

### Keep current system kernel not to change
The kernel & headers use package version: 1:1.20220331-1

### Sync kernel and userland

### Uninstall previous dkms module

### Install required kernel package

gzip: stdin: not in gzip format
debian package raspberrypi-kernel_1:1.20220331-1_arm64.deb already installed.
dpkg-query: no packages found matching raspberrypi-kernel-headers
--2024-04-18 09:50:54--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.227.39, 176.126.240.86, 176.126.240.167, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.227.39|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-04-18 09:50:55 ERROR 404: Not Found.

--2024-04-18 09:50:55--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.231.151, 46.235.227.39, 176.126.240.86, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.231.151|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-04-18 09:50:56 ERROR 404: Not Found.

--2024-04-18 09:50:56--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.231.145, 46.235.231.151, 46.235.227.39, ...
Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.231.145|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-04-18 09:50:57 ERROR 404: Not Found.

dpkg-deb: error: '/tmp/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb' is not a Debian format archive
dpkg: error processing archive /tmp/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /tmp/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
Error: Install kernel or header failed
```

bigbearishappy, 2024-04-18:
> @bigbearishappy
> 
> > It should work as you wish with command `./scripts/reTerminal.sh --keep-kernel`
> 
> Thank you for your reply! I have tried the command but returned error. It seems still trying to search and install the kernel? The searched name is kind of mismatched though... _raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb_ v.s. _raspberrypi-kernel-headers_1.20220331-1_arm64.deb_ on raspi website
> 
> ```
> pi@raspberrypi:~/seeed-linux-dtoverlays $ sudo ./scripts/reTerminal.sh --keep-kernel
> /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
> 
> ### Current Debian version is bullseye or earlier
> 
> ### Install required tool packages
> Hit:1 http://archive.raspberrypi.org/debian bullseye InRelease
> Hit:2 http://deb.debian.org/debian bullseye InRelease
> Hit:3 http://security.debian.org/debian-security bullseye-security InRelease
> Hit:4 http://deb.debian.org/debian bullseye-updates InRelease
> Reading package lists... Done
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> dkms is already the newest version (2.8.4-3).
> The following package was automatically installed and is no longer required:
>   libfuse2
> Use 'sudo apt autoremove' to remove it.
> 0 upgraded, 0 newly installed, 0 to remove and 325 not upgraded.
> 
> ### Keep current system kernel not to change
> The kernel & headers use package version: 1:1.20220331-1
> 
> ### Sync kernel and userland
> 
> ### Uninstall previous dkms module
> 
> ### Install required kernel package
> 
> gzip: stdin: not in gzip format
> debian package raspberrypi-kernel_1:1.20220331-1_arm64.deb already installed.
> dpkg-query: no packages found matching raspberrypi-kernel-headers
> --2024-04-18 09:50:54--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
> Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.227.39, 176.126.240.86, 176.126.240.167, ...
> Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.227.39|:80... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2024-04-18 09:50:55 ERROR 404: Not Found.
> 
> --2024-04-18 09:50:55--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
> Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.231.151, 46.235.227.39, 176.126.240.86, ...
> Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.231.151|:80... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2024-04-18 09:50:56 ERROR 404: Not Found.
> 
> --2024-04-18 09:50:56--  http://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
> Resolving archive.raspberrypi.org (archive.raspberrypi.org)... 46.235.231.145, 46.235.231.151, 46.235.227.39, ...
> Connecting to archive.raspberrypi.org (archive.raspberrypi.org)|46.235.231.145|:80... connected.
> HTTP request sent, awaiting response... 404 Not Found
> 2024-04-18 09:50:57 ERROR 404: Not Found.
> 
> dpkg-deb: error: '/tmp/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb' is not a Debian format archive
> dpkg: error processing archive /tmp/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb (--install):
>  dpkg-deb --control subprocess returned error exit status 2
> Errors were encountered while processing:
>  /tmp/raspberrypi-kernel-headers_1:1.20220331-1_arm64.deb
> Error: Install kernel or header failed
> ```

This problem is caused by the abnormal data acquisition of FORCE_KERNEL. I did a quick fix of it in [here](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/ef082c3eac68588331523d545f17731dfe0d1625).

Please pull the new code and test it.

Thank you~

cyaoplus, 2024-04-24:
Thanks. This issue was solved.

closed 2024-04-24 by cyaoplus