Seeed-Studio/seeed-linux-dtoverlays #101: [reTerminal DM] Support --keep-kernel in driver script

franzos, 2024-04-12:
**Is your feature request related to a problem? Please describe.**

Follow-up to [issue-100](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/issues/100#issuecomment-2051843682). I'm unable to retain the existing kernel:

```bash
./scripts/reTerminal.sh --device reTerminal-plus --keep-kernel

### Current Debian version is bookworm or later

### Install required tool packages
Hit:1 http://deb.debian.org/debian bookworm InRelease
Get:2 http://deb.debian.org/debian-security bookworm-security InRelease [48.0 kB]
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]
Get:4 http://deb.debian.org/debian-security bookworm-security/main arm64 Packages [147 kB]
Get:5 http://deb.debian.org/debian-security bookworm-security/main armhf Packages [144 kB]
Get:6 http://deb.debian.org/debian-security bookworm-security/main Translation-en [90.3 kB]
Get:7 http://archive.raspberrypi.com/debian bookworm InRelease [23.6 kB]
Get:8 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [384 kB]
Get:9 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [374 kB]
Fetched 1,267 kB in 3s (372 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
dkms is already the newest version (3.0.10-8+deb12u1).
0 upgraded, 0 newly installed, 0 to remove and 234 not upgraded.

### Don't support option --keep-kernel currently.

### Pls remove --keep-kernel in your command.
```

**Describe the solution you'd like**

Support `--keep-kernel`

bigbearishappy, 2024-04-15:
Fix commit is [here](https://github.com/Seeed-Studio/seeed-linux-dtoverlays/commit/f99e6869879b8f3766798c7dfa0cdf0af1c5752d).

You can pull the latest code the test it now.

franzos, 2024-04-15:
Thank you very much. I will test it asap.

bigbearishappy, 2024-05-14:
@franzos Anything update now?

bigbearishappy, 2024-10-15:
Closing this issue now. If there is any further problem. Please reopen it or create a new one.

closed 2024-10-15 by bigbearishappy